MarlinFirmware/Marlin #20957: [Possible BUG] Serial comunication closes/drops

rmcbc, 2021-01-31:
### Bug Description

With last bugfix-2.0.x when connected to the board with OctoPi after executing one command or wait some seconds the comunication closes.

### Configuration Files

This happens with both my printers.

[Ender 3.zip](https://github.com/MarlinFirmware/Marlin/files/5900951/Ender.3.zip)
[AM8.zip](https://github.com/MarlinFirmware/Marlin/files/5900953/AM8.zip)

### Steps to Reproduce

1. Connect using octoprint
2. After connection observe that there is no temperature messages
3. If I execute one command command or wait some seconds the connection closes/drops.

X-Ryl669, 2021-01-31:
Please test with latest bugfix, it should be resolved. If so, please close the issue. Thanks!

rmcbc, 2021-02-01:
Solved with latest bugfix. Thanks.

closed 2021-02-01 by rmcbc