MarlinFirmware/Marlin #20964: [FR] Add additional Preheat options

eblieb, 2021-02-01:
Is it possible to add additional preheat options. Not just PLA and ABS but also PLA+ and PETG?

ellensp, 2021-02-01:
Marlin already supports up to 5 Preheat options

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-02-01 by ellensp

eblieb, 2021-02-01:
Wait this was a joke to close it right? Because the documentation no where states there are 5 preheat options, in fact the documentation only ever references 2.

eblieb, 2021-02-01:
Also to claim this is only for bug related issues? There is literally a feature request option and multiple feature requests in this very location.