MarlinFirmware/Marlin #20965: Duplicate Z-Stepper Auto-Alignment feature for X and Y

tesla128, 2021-02-01:
### Description

Take the existing Z-Stepper Auto-Align feature and make it available for X and Y axis

### Feature Workflow

Build machine with multiple X and Y axis steppers, of which driving independently as-such is already a feature
Install independent end-stops with their own inputs or use stall-guard capable drivers for X and Y axis. also already supported
G34 command or configuration option to include auto-align in the homing procedure

#### Additional Information

With large-format or portable machines that have multiple motors for each axis, it would be desirable to auto-align any axis, not just the Z.  Temperature changes, changes due to moving, or manual movements of the axis during setup may mis-align one or more axis and cause skew.  All of the hardware is already in place to perform this automatically, yet the current recommendations are either manual alignment or forcing motors to skip steps.   Forcing motors to skip steps by design is bad practice and could cause damage to limit switches, cause undue belt stretch, damage flex couplers, or damage other components.

gcurtis79, 2021-02-01:
This would be a great feature on large format machines. Especially CNC and laser cutters that often have multiple motors on the y-axis. I already have some machines that could benefit from features like these.

sjasonsmith, 2021-02-08:
It this not already satisfied by `X_DUAL_ENDSTOPS` and `Y_DUAL_ENDSTOPS`? In this case you wouldn't use G34, you would just home as normal and it should take each motor to the endstops.

If you are imagining something different, please provide more detail describing the machine and how you would expect it to behavior, different than what Marlin can already do.

tesla128, 2021-02-28:
Sorry for the late reply, working on this isn't at the pace I would like and learning as I go.

I did initially misunderstand the operation of auto-align and dual end-stops. Dual end-stops does appear to do what I need in the case that I succeed in making my current end-stops adjustable to a fine enough resolution to still support the large format, especially given that I am attempting to use stall-guard.  I am undecided on the utility of skew compensation, especially if stall-guard varies any at all (reported, not experienced myself) and perhaps I'm stuck to switching to mechanical end-stops for now.

I am confident in being able to put a very accurate piece of metal in a fixed location that can be probed such as a carpenter's square where I could attach a conductive probe to the "nozzle" (router head), do a z-probe to the square that sets a known z height, move a distance in the X/Y, down some z height below the top surface of the square, and then probe into the edge of it the same way you would a conductive probe on the Z.  

This in itself may be overkill or accomplished other methods, although as I've been considering this, a carpenter square or equivalent as a probe target could also make for interesting auto-skew-compensation and also not the only embodiment.  Very likely overkill for today, but when a machine has to build a machine without human intervention (on Mars? Titan?) , it may be a necessity.

closed 2023-02-14 by thisiskeithb