MarlinFirmware/Marlin #20967: [FR]  Multi Source Temperature Sensor data per HotEnd

rflulling, 2021-02-01:
### Description

**Multi Source Temperature Sensor data per HotEnd** 

As one is better suited to lower temps and the other higher temps. 
If both are regarded to be active all of the time but marlin is smart enough to ignore The resistor at higher temps and ignore the Transistor at lower temps. rather it might be argued that at a specified range, Marlin simply shifts over reading one to reading the other rather than trying to table both? 

If both are configured properly there should be very little if any offset. In theory there should be a solid over lap in their read data. Thus the two may be tested against each other to validate calibration, and against malfunction. 

I am aware that marlin has a redundant sensor feature. But as far as I know it does not enable a function like this. 

**How can this benefit a user?**
Wider range of operational temps. 

If a user can modify a hot end to have a place for both sensor types. Then the hot end should be able to operate in a range of 90C to 500C very easily I think. 90C as reference to a rare type of material that is printed at... 90C. 

### Feature Workflow

- Extruder1:
- - Thermal Resistor Lookup Table1 (max 400C)
- - Thermal Transistor Lookup Table2 (max 850C)
- - Configuration Start and Stop Lookup Table1
- - Configuration Start and Stop Lookup Table2

#### Additional Information

Not sure what more to add here. I am not a coder. 
-Will dig for and add anything that is requested. 

Has this been requested before?

ellensp, 2021-02-01:
Most thermistors die when exposed to temperatures  higher than what they can read.
This would be a use once deal

rflulling, 2021-02-03:
> Most thermistors die when exposed to temperatures higher than what they can read.
> This would be a use once deal.

I see. I did not know this. My understanding was only that over a certain temp the value they read no longer changed. I was some how unaware that they would actually cook. 

Perhaps that they are manufactured inside glass, I assumed them to be more durable, environment wise.