MarlinFirmware/Marlin #20968: wrong motor bahaviour with dual E and dual Z on SKR 1.4 turbo

jimmi4u, 2021-02-01:
Hi folks,

I have a big issue. I'm running dual E motors with a single nozzle (cyclops clone) and dual Z motors on a SKR 1.4 turbo, with five TMC5160 steppers, with a marlin bugfix 2.0.x SKR 1.4 FW.
They are wired as within the manual described: X, Y, ZA, ZB, E0, E1. Everything is working well except the motor of E1 which acts funny: I'll try to describe the bahaviour:
On octoprint when clicking on retract it retracts the correct lenght the first time, on the second time clicking on retract it extrudes the same length, on the third time it retracts, then extracts, and so on... 
While on tool change  during printing the motor is goin back and forth (rattling), as if it can't decide which direction to go ;)
The extruders are both bondtech dual drives (left and right) for bowden.. I tried allready three different motors and two different extruders. And I have a BLTouch also working too.  
Here are my config files:
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5904232/configuration.zip)

Thanks for your help!
Paul

ellensp, 2021-02-01:
You do not have dual Z, you have 2 z motors in parallel, from the firmware perspective you have 1 z stepper motor

jimmi4u, 2021-02-02:
Hi and thanks for your comment. 
The thing with the dual Z is that if I enable this feature the fw won't compile, because of a missing pin... It took me a while 'till I figured this out: I read somewehere that wiring it this way the Z1 motor is automatically in dual  mode, which is in case the fact. With this settings I don't have troubles with compiling nor with the z1 motor, which is working as it should. This is the dilemma of having 6 motors and 5 stepper dirvers...

ellensp, 2021-02-03:
You don't have enough stepper drivers for real dual Z and dual E
X,Y,Z1,Z2,E0,E1 is six stepper drivers. The board only has 5 (ZA and ZB are on the same stepper driver) 
You have to electrically add another stepper driver and configures pins for it.

jimmi4u, 2021-02-03:
Hi @ellensp,
yeah, that's right. If I got you right, I have to buy a new board?
But I think I managed it fianlly this way. I suppose with this wiring ZA and ZB are on the same stepper...  And they are working well. But I had the probs with the E1 and checked the wiring and because I have two different motors I got diff blue and red wires, so I switched them on the E1 in order to correspond to E0, reversed direction in FW and voila, everything is working. For now at least.. ;) 
Are there other ways in order to achieve this? (Except bying a new board)
Thanks a lot!

ellensp, 2021-02-03:
BTT sell a https://www.biqu.equipment/collections/expansion-board/products/btt-rrf-wifi-v1-0-module-driver-expansion-module-for-skr-v1-4-1-4-turbo-skr-v1-3-3d-printer-part 
It can use just exp2 with a creality cr10 lcd on exp1  (step/direction/enable only, no fancy 2209 in that single cable mode)

jimmi4u, 2021-02-03:
So, if I got this right SPI Mode is possible, so my TMC5160 will work too? But goin with the SKR 1.3 Pro would be the better choice? Maybe with the TMC5160s?

NezSez, 2021-02-04:
Please read up on some of the limitations of the BTT designs regarding UART, SPI, PIN assignments, etc before taking the plunge.
See the review by [AlesO](https://www.amazon.com/gp/profile/amzn1.account.AH3P6MT5BLMJB72ZCJQHJNLHLU4A/ref=cm_cr_getr_d_gw_btm?ie=UTF8) on the SKR Pro 1.2 and using some of the BTT addons.

I have had several issues using all BTT equipment similiar to AlesO. I keep wondering if they will release the new 32bit 400mhz board they are using on their new printer as a standalone upgrade or not.

closed 2021-03-17 by github-actions[bot]