MarlinFirmware/Marlin #20969: [BUG] M25/M125 Pause behavior

qwewer0, 2021-02-01:
### Bug Description

Pausing an SD or serial print has unexpected results.

Test results:
- SD printing
	- Pause called from the gcode file or gcode sent manually through serial (USB)
		- **M25/M125**
			1. Print is paused
			2. Printhead is parked
			3. The following three gcode line after M25 is executed on the current (parked) z height
			4. Display shows "Parking...", and is unresponsive, but not frozen (temperature changed correctly)
			5. Connecting to the printer (Pronterface) and sending M24 resumes the pause procedure
			6. Display shows the resume options (Purge more, Continue)
			7. By selecting Continue, the printhead returns to the position where the pause occurred, and continues the print, but skips the already executed three gcodes 
		- **M25 P0/M125 P0**
			Same as M25/M125
		- **M25 P1/M125 P1**
			1. Print is paused
			2. Printhead is parked
			3. Display shows "Press Button to resume print"
			4. After pressing the button, the display shows the resume options (Purge more, Continue)
			5. By selecting Continue, the printhead returns to the position where the pause occurred, and continues the print

- Serial (USB) printing (via Pronterface) or gcode sent manually through serial (USB)
	- Pause called from the gcode file
		- **M25/M125**
			1. Print is paused
			2. Printhead is parked
			3. Display shows "Press Button to resume print"
			4. After pressing the button, the display shows the resume options (Purge more, Continue)
			5. By selecting Continue, the printhead returns to the position where the pause occurred, and continues the print
		- **M25 P0/M125 P0**
			Same as M25/M125
		- **M25 P1/M125 P1**
			Same as M25/M125

When the "Press Button to resume print" message is shown on the display sending M24 to the printer has no effect.

### Configuration

Printer: Ender 3
Board: SKR Mini E3 v1.2
Firmware: STM32F103RC_btt, Bugfix-2.0.x (c74f972)
Configuration files: [Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5905224/Configuration.zip)

### Requirements

- SDSUPPORT
- NOZZLE_PARK_FEATURE
- ADVANCED_PAUSE_FEATURE
- PARK_HEAD_ON_PAUSE
- Display

### Steps to Reproduce

1. Start a print through SD card or serial (USB) with `M25/M125`, `M25 P0/M125 P0`, `M25 P1/M125 P1` in it or send it manually via serial (USB) in mid print
2. Observe the pausing procedure
3. Try sending `M24` to the printer and/or use the display prompt to resume the printing
4. Observe the resuming procedure

**Expected behavior:**

- A print paused with `M25/M125` or `M25 P0/M125 P0` should not show display prompt (unless the code explicitly says so) and wait for `M24`, which should resume the print (without any display interaction?)
- A print paused with `M25 P1/M125 P1` should show display prompt (unless the code explicitly says otherwise) and be able to resume the print through the display (and with `M24` too?)

**Actual behavior:**

See "Test results" above, but in short:
- An SD print paused with `M25/M125` or `M25 P0/M125 P0` doesn't show display prompt and waits for `M24` (as it should), but after sending `M24` to the printer the display is still needed to select "Continue" in order to resume the print
- A serial print paused with `M25 P0/M125 P0` shows display prompt and the print can be resumed through the display (but not with `M24`(?))
- An SD or serial print paused with `M25 P1/M125 P1` shows display prompt and the print can be resumed through the display (as it should) (but not with `M24`(?))

#### Additional Information

This issue is connected to https://github.com/MarlinFirmware/Marlin/issues/19870

jmc5113, 2021-02-06:
+1 I'm just noticing the same issue :-/

qwewer0, 2021-02-06:
> +1 I'm just noticing the same issue :-/

Could you share your configuration files?

jmc5113, 2021-02-06:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5937673/Configuration.zip)

qwewer0, 2021-02-06:
@jmc5113 Could you try if the issue is still present on the latest bugfix-2.0.x?

jmc5113, 2021-02-06:
same thing happened on bugfix-2.0.x.

was that expected to fix it?

jmc5113, 2021-02-06:
From the dbg output it looks like extra commands are being executed before it actually pauses...

echo:M117 Place weights in slots and resume printing
echo:M125
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
X:10.00 Y:280.00 Z:5.00 E:-2.00 Count X:1600 Y:44800 Z:2000
echo:G1 X155.529 Y138.560 F12000.000
echo:G11
echo:G92 E0
X:155.53 Y:138.56 Z:5.00 E:0.00 Count X:1600 Y:44800 Z:2000

jmc5113, 2021-02-06:
and when it does pause, I can't use the display to unpause it. I have to use M24 from serial console.

jmc5113, 2021-02-06:
Also when I put in M24, it initiates the purge sequence, which I do not need as I'm only trying to pause.

qwewer0, 2021-02-06:
> same thing happened on bugfix-2.0.x.
> 
> was that expected to fix it?

It's best to test the latest bugfix-2.0.x to not work for nothing.

jmc5113, 2021-02-06:
just discovered... 

M125 P1 // will move to park and pause and show the correct prompt on the screen.

https://marlinfw.org/docs/gcode/M125.html

qwewer0, 2021-03-09:
Still here.

thinkyhead, 2021-03-09:
NOTE: `M25` is not supported as a command run from SD because it isn't a command that blocks the queue. It simply tells the machine to stop processing commands from the open SD print-job.

qwewer0, 2021-03-09:
> NOTE: `M25` is not supported as a command run from SD because it isn't a command that blocks the queue. It simply tells the machine to stop processing commands from the open SD print-job.

`M25` and `M125` same with `PARK_HEAD_ON_PAUSE` enabled, so then is it only `M25` that isn't suppose to be used or `M125` too?
https://github.com/MarlinFirmware/Marlin/blob/2f1fd4bbaab08e7fee2c405226dd7453de958b5b/Marlin/src/gcode/sd/M24_M25.cpp#L91-L97

@thinkyhead Thank you for looking into this.

thinkyhead, 2021-03-09:
`M125` is the supported method. `M24` as you point out is not guaranteed.

thinkyhead, 2021-03-09:
Was there some point in the past when these commands were behaving better? If so, how recently did this bad behavior start to appear?

The reason I mention that `M25` is non-blocking is that it only runs when it goes to the front of the queue, so there are always going to be some commands in the queue behind it. Therefore it will not behave as one expects when used in a file printing from SD.

qwewer0, 2021-03-09:
> Was there some point in the past when these commands were behaving better? If so, how recently did this bad behavior start to appear?

If I remember it correctly, it behaved like this from the start when I switched from 8bit 1.1.x to 32bit 2.0.x at around 2019. Dec.

thinkyhead, 2021-03-09:
> When the "Press Button to resume print" message is shown on the display sending M24 to the printer has no effect.

The prompt must be dismissed by a host prompt or by `M108` in this case. The `M24` command is historically still a queued command and Marlin makes no special allowance to execute it out of sequence.

qwewer0, 2021-03-09:
> The prompt must be dismissed by a host prompt or by `M108` in this case. The `M24` command is historically still a queued command and Marlin makes no special allowance to execute it out of sequence.

Is there a better way to test the pause behavior to get more information?

thinkyhead, 2021-03-09:
I don't quite understand the question.

- For testing, we can disregard how `M25` behaves when used from within a G-code file printing from SD.
- We always want `M25` to return right away and to never block the queue.
- Since `M25` can now insert a `park`, it might be prudent to call `queue.exaust()` in the `park` function.
- The `M125` command is supported as a command within a G-code stream and can block the queue. So it's ok to call it from an SD print. When called otherwise … well …
```cpp
/**
 * M125: Store current position and move to parking position.
 *       Called on pause (by M25) to prevent material leaking onto the
 *       object. On resume (M24) the head will be moved back and the
 *       print will resume.
 *
 *       When not actively SD printing, M125 simply moves to the park
 *       position and waits, resuming with a button click or M108.
 *       Without PARK_HEAD_ON_PAUSE the M125 command does nothing.
 *
 *    L<linear> = Override retract Length
 *    X<pos>    = Override park position X
 *    Y<pos>    = Override park position Y
 *    Z<linear> = Override Z raise
 *
 *  With an LCD menu:
 *    P<bool>   = Always show a prompt and await a response
 */
```

thinkyhead, 2021-03-09:
> Since `M25` can now insert a `park`, it might be prudent to call `queue.exaust()` in the `park` function.

That said, `queue.exaust()` should be considered dangerous because it could in rare and unusual cases itself invoke more pause commands or dwells…

thinkyhead, 2021-03-09:
I see that if code was added to `GCodeQueue::RingBuffer::commit_command` to check for the `M25` command and call `card.pauseSDPrint()` then it would prevent any more commands from the SD card being added to the queue until the print was resumed.

Ilya-Belyanov, 2025-03-09:
I have a problem with M25 command.
I use M24 for printing gcode from file, later send M25.
I got "echo:busy: processing" and no pause