MarlinFirmware/Marlin #20970: [BUG] TFT_COLOR_UI: Calibration dialog keeps showing up after printer power on

Miraculix200, 2021-02-01:
### Bug Description

Branch: bugfix-2.0.x
Board: MKS Robin Nano V1.2

Since commit 68abaeab19b41824bc151bc0f65e76ce8a2e4916 the TFT_COLOR_UI calibration dialog keeps showing up every time the printer is powered on.

When reverting to a previous version, there is no calibration required on power on.

Sometimes the calibration doesn't complete. After the 4th corner it asks for "top left corner" again and there's a cross in the top left and bottom right corner. Then the touch display does not react anymore. Serial connection still works.

Any idea what could be wrong here? I've tried finding the problem for hours today, as I thought it's a problem with my configuration. But then I figured out it also happens with a slightly modified Configuration.h

### Configuration Files

Configuration.h:
https://gist.github.com/Miraculix200/9b0db85f630442e83ec1864b326368c7

(no other files were modified to reproduce the bug)

### Steps to Reproduce

Flash firmware with the above Configuration.h on Robin Nano 1.2 with 3.5" FSMC display
Turn on the printer
Complete calibration (if it works)
Turn the printer off and on again
Calibration dialog appears again (and may complete or not complete)

rhapsodyv, 2021-02-01:
You should go to menu, configuration, and store settings, after a successful calibration. If you don't do it, it will keep asking for calibration, as the values are lost if you don't save it.

That commit changed the eeprom version, so everything should be saved again. That is the reason.

Try it: power on, calibrate, go to menu, configuration, and store settings.

It should work.

Let me know.

Miraculix200, 2021-02-01:
lol

Yes, that worked. 

I actually did use gcodes to save the settings to EEPROM, but it didn't work (though maybe it was the version where I forgot to #define EEPROM_SETTINGS, not sure).

Any reason why the calibration settings don't get stored automatically anymore? Or maybe they didn't. Don't remember.

So this can be closed. Or maybe it's a feature request?

rhapsodyv, 2021-02-01:
It was never saved automatically. But if you have the calibration values in your configuration.h, it will use that values when eeprom is erased.

Yes, it can be a FR asking for a auto save for the calibration screen. It’s a good idea. I will work on it ASAP.

rhapsodyv, 2021-02-02:
Autosave option was merged!

closed 2021-02-02 by rhapsodyv