MarlinFirmware/Marlin #20975: [FR] Make independent baud rate for SERIAL1

mriscoc, 2021-02-02:
The DWIN display on Ender 3v2 use the SERIAL1, when the BAUDRATE define is changed to 250000 baud the display can't communicate with the mainboard. A simple solution is making the BAUDRATE for the SERIAL1 independent from SERIAL0 defining a new constant BAUDRATE1. I propose this patch:

```diff
From 3f1604d8989496bbaf292c30d7e363e53e3bc1a6 Mon Sep 17 00:00:00 2001
From: @mriscoc
Date: Tue, 2 Feb 2021 02:42:11 -0500
Subject: [PATCH] Define BAUDRATE1 to make independent baudrate for MYSERIAL1
 Usefull, for example, for the DWIN Ender 3 v2 display)

---
 Marlin/Configuration.h    | 1 +
 Marlin/src/MarlinCore.cpp | 2 +-
 2 files changed, 2 insertions(+), 1 deletion(-)

diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index d888e62aaf..b41ec302ad 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -122,6 +122,7 @@
  * :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000]
  */
 #define BAUDRATE 250000
+#define BAUDRATE1 115200   // make independent baudrate (For example DWIN display on Ender 3 v2)
 
 // Enable the Bluetooth serial interface on AT90USB devices
 //#define BLUETOOTH
diff --git a/Marlin/src/MarlinCore.cpp b/Marlin/src/MarlinCore.cpp
index 16111936da..ada57fb65c 100644
--- a/Marlin/src/MarlinCore.cpp
+++ b/Marlin/src/MarlinCore.cpp
@@ -891,7 +891,7 @@ void setup() {
   while (!MYSERIAL0.connected() && PENDING(millis(), serial_connect_timeout)) { /*nada*/ }
 
   #if HAS_MULTI_SERIAL && !HAS_ETHERNET
-    MYSERIAL1.begin(BAUDRATE);
+    MYSERIAL1.begin(BAUDRATE1);   //make independent baudrate (For example DWIN display on Ender 3 v2)
     serial_connect_timeout = millis() + 1000UL;
     while (!MYSERIAL1.connected() && PENDING(millis(), serial_connect_timeout)) { /*nada*/ }
   #endif
-- 
2.16.2.windows.1
```

ellensp, 2021-02-02:
In bugfix and the next release LCD serial ports are now set with LCD_SERIAL_PORT with their own LCD_BAUDRATE

ellensp, 2021-02-02:
eg
```
#if HAS_DGUS_LCD
  #define LCD_SERIAL_PORT 3
  #define LCD_BAUDRATE 115200
```

ellensp, 2021-02-02:
So I'm afraid this is unnecessary going forward.

closed 2021-02-02 by ellensp

mriscoc, 2021-02-02:
Ok, but in current bugfix branch the SERIAL1 is yet using BAUDRATE

```c
  #if HAS_MULTI_SERIAL && !HAS_ETHERNET
    MYSERIAL1.begin(BAUDRATE);
    serial_connect_timeout = millis() + 1000UL;
    while (!MYSERIAL1.connected() && PENDING(millis(), serial_connect_timeout)) { /*nada*/ }
  #endif
  SERIAL_ECHOLNPGM("start");
```
How I can currently define a different baud for the SERIAL1?

Thanks.

ellensp, 2021-02-02:
A  lot of  modern controller use serial port -1   (which is baudrate less) 
so baudrate only applies to serial ports  0-7 (deepening on controller) 
If its not for communication with a host it will not be using serial_port or serial_port_1 but a new name like the LCD_SERIAL_PORT  example with its independent baud rates.