MarlinFirmware/Marlin #20977: tmc2209 resonance\vibration

rstanislav, 2021-02-02:
> Same issue, i have read all the page... and i want to add something in the investigation!
> 
> I use stealthchop and the prints start quietly, but i experience noise/vibration as soon as the stepper become hot.
> its like the stealthchop turns off!
> can anyone confirm?!
> 
> hardware: BTT v1.4 turbo, BTT TMC2209 v1.2, Cartesian, Anet A8 steppers.

@LakoSam did you ever find a solution to this problem? I have the same setup and same issue.. It starts off quiet as a mouse, but then after a few hours my stepper motors are screaming! I have dedicated 5015 turbo fans on my drivers and sufficient airflow in the case, but this issue for me is definitely something to do with time, and possibly heat.

_Originally posted by @CaptainSweatpants in https://github.com/MarlinFirmware/Marlin/issues/18286#issuecomment-734518130_

For some unknown reason original issue was closed, i have same problem - skr 1.4 turbo with tmc 2209 v1.2 btt drivers, resonance at speeds is so strong nuts on frame are getting loose.

thinkyhead, 2021-02-02:
It the prerogative of anyone to close their own issue when they are satisfied. Please don't nag them about it.

It is good that you have opened a new issue so that someone can help to resolve your specific situation. However, we need you to follow the template to help its readers investigate and resolve the issue. In the not-too-distant future a post like this will be closed automatically by a robot, because there is nothing we can do with them.

If you would like to help investigate and solve this issue, please fill out all the information requested in the bug report template, including your configuration files.

nowkillkennys, 2021-02-07:
im the same here very loud

closed 2021-03-20 by github-actions[bot]