MarlinFirmware/Marlin #20980: [BUG] Bugfix build has steppers uncommented.

ebraiman, 2021-02-03:
### Bug Description

Steppers should be commented out

### Configuration Files

[Marlin-bugfix-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/5915379/Marlin-bugfix-2.0.x.zip)

No mods and downloaded 10 minutes ago.

### Steps to Reproduce

1. Download Bugfix build
2. Open in VScode
3. Search on, A4988

Expected
//#define X_DRIVER_TYPE  DRV8825
//#define Y_DRIVER_TYPE  DRV8825
//define Z_DRIVER_TYPE  DRV8825
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
//#define Z2_DRIVER_TYPE A4988
//#define Z3_DRIVER_TYPE A4988
//#define Z4_DRIVER_TYPE A4988
//define E0_DRIVER_TYPE A4988
//#define E1_DRIVER_TYPE A4988
//#define E2_DRIVER_TYPE A4988
//#define E3_DRIVER_TYPE A4988
//#define E4_DRIVER_TYPE A4988
//#define E5_DRIVER_TYPE A4988
//#define E6_DRIVER_TYPE A4988
//#define E7_DRIVER_TYPE A4988

#define X_DRIVER_TYPE  DRV8825
#define Y_DRIVER_TYPE  DRV8825
#define Z_DRIVER_TYPE  DRV8825
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
//#define Z2_DRIVER_TYPE A4988
//#define Z3_DRIVER_TYPE A4988
//#define Z4_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE A4988
//#define E1_DRIVER_TYPE A4988
//#define E2_DRIVER_TYPE A4988
//#define E3_DRIVER_TYPE A4988
//#define E4_DRIVER_TYPE A4988
//#define E5_DRIVER_TYPE A4988
//#define E6_DRIVER_TYPE A4988
//#define E7_DRIVER_TYPE A4988

ManuelMcLure, 2021-02-03:
The current version of `Configuration.h` in the bugfix branch has:
```
#define X_DRIVER_TYPE  A4988
#define Y_DRIVER_TYPE  A4988
#define Z_DRIVER_TYPE  A4988
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
//#define Z2_DRIVER_TYPE A4988
//#define Z3_DRIVER_TYPE A4988
//#define Z4_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE A4988
//#define E1_DRIVER_TYPE A4988
//#define E2_DRIVER_TYPE A4988
//#define E3_DRIVER_TYPE A4988
//#define E4_DRIVER_TYPE A4988
//#define E5_DRIVER_TYPE A4988
//#define E6_DRIVER_TYPE A4988
//#define E7_DRIVER_TYPE A4988
```
The X, Y, Z and E0 drivers were explicitly uncommented a few months ago because they didn't behave exactly like A4988 unless they were explicitly set. I don't know how you got DRV8825 in your download, since that change was done in October and is the last change done to that file in the bugfix-2.0.x branch.

ManuelMcLure, 2021-02-03:
I just downloaded the zip file in your link, and it shows the same lines as my comment above - X, Y, Z and E0 are uncommented and set to A4988.

ellensp, 2021-02-03:
User seems to be mistaken or confused.

closed 2021-02-03 by ellensp