MarlinFirmware/Marlin #20982: [BUG] Neopixels will not consistently change colors

lightmaster, 2021-02-03:
### Bug Description

Neopixels do not change color consistently nor repeatably. For instance, can click the LCD button for the white default, and each time it's clicked, the Neopixel will change. Can Tell it to change to "Blue" and some of the LEDs will change to blue, some to red. This occurs both by gcode and by LCD screen buttons. Also seems to happen irregardless of whether I have 1 Neopixel setup or 2 strips of them setup.

### Configuration Files

[Marlin config](https://drive.google.com/file/d/1LxC_llp7TaaGjvmDxNiEEWH1oUTUm_zM/view?usp=sharing)

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. Configure Neopixel as they are in the attached config files.
2. Try to change Neopixel to one color/preset.
3. Change them to the same preset again, with or without changing them to a different preset first.

**Expected behavior:**

A Neopixel strip should all change to the same color and brightness based on which preset or RGB setting is picked. Setting colors should also be consistent, in that if I pick Preset "A", and then immediately pick Preset "A" again, I shouldn't notice any change on the second click of Preset "A" since they are already on Preset "A".

**Actual behavior:**

Can tell them to go to the same preset repeatedly and each time the Neopixels will light up differently. 

#### Additional Information

Video showing me sending this command over and over again: `M150 R255 U255 B255 P255`
[image]

lightmaster, 2021-02-03:
Also, I am on the now current build of bugfix-2.0.x

ellensp, 2021-02-03:
config files 
a) wont even compile on 2.0.7.2 
b) don't have neopixels enabled, 
c) aren't for bugfix. 

Any why did you ignore the huge warning "#error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. "

neopixels will not work reliably on this platform using Adafruit library, (or derivatives) ever. 
The more neopixels you have the worse it becomes.

neopixels require a dedicated controller so the precise timing can be met without interfering with the precise timing Marlin requires.  

It may be possible in the future if someone writes a DMA driven system 

Further this is a duplication of https://github.com/MarlinFirmware/Marlin/issues/19800

lightmaster, 2021-02-03:
oops, accidentally uploaded the wrong config. I've changed the link above to: [Marlin Config (Correct)](https://drive.google.com/file/d/1Q7ELko1hUF2a-FK2B-x6u7lLW7f-H8F8/view?usp=sharing).

This does compile fine under the bugfix-2.0.x branch using `mks_robin_nano35_stm32` env

lightmaster, 2021-02-03:
> And why did you ignore the huge warning "#error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. "

I didnt see any warning like this. And I just scrolled back through the compile logs, and there is no such warning when I compile.

ellensp, 2021-02-03:
In https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/HAL/STM32F1/inc/SanityCheck.h#L51
```
#if ENABLED(NEOPIXEL_LED)
  #error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!"
#endif
```
but perhaps your using 
#default_envs = mks_robin_stm32 
instead of the older 
#default_envs = mks_robin_nano35

lightmaster, 2021-02-03:
yup, using `mks_robin_nano35_stm32` since the older one takes up too much flash space to be able to use UBL Leveling. Neopixel support compiles without issue for the stm32 version and functions at least somewhat, so figured there was a bug within the code that set the colors of the Neopixels.

ellensp, 2021-02-03:
with new information, its not that same as 19800, using different build environment, reopening.

X-Ryl669, 2021-02-05:
Can you try to apply #20901 to see if it fixes it ?
My hypothesis being that the STM32 is outputting wrong timing for the led strip. With my patch it should be fixed, so please confirm.

lightmaster, 2021-02-05:
@X-Ryl669 Soon as this print is done in a couple hours, sure. Will that PR affect `mks_robin_nano35_stm32` as well or do I need to change anything first?

X-Ryl669, 2021-02-05:
If you use @ellensp fork of Neopixel, there's nothing to change. Else, you must make sure that the include path to "Delay.h" is correct in NeoPixels.cpp. It should point to Marlin/src/core/Delay.h. There is no other change in this PR.

lightmaster, 2021-02-06:
How would I go about switching it to use his fork?

lightmaster, 2021-02-06:
Not really sure what I did wrong, but Neopixels will not react at all with the PR. First I cloned your PR and copied over my configs, then changed `NEOPIXEL_LED` as instructed here https://github.com/MarlinFirmware/Marlin/issues/19800#issuecomment-716145359. Then I tried created a new branch of my own fork of Marlin and merging in your PR, but still no reaction from the Neopixels.

X-Ryl669, 2021-02-06:
The PR has been merged in bugfix branch. You can now do as usual, pull bugfix, apply your config. You can apply the comment from @ellensp for using his fork (it's in platformio.ini). You might need to disable the `#error Not supported in STM32`in sanity_check (you've already done this earlier).

lightmaster, 2021-02-06:
Alright, about to try it out again. I did notice yesterday that when I applied @ellensp's fork, I had to modify the location of Delay.h because platformio put his fork deep into the .pio directory

lightmaster, 2021-02-06:
After adjusting the path to `Delay.h` it does build successfully, but I get no control over Neopixel at all. I also see a lot of warnings in the output from building: [pastebin log](https://pastebin.com/3mtDYc6n)

X-Ryl669, 2021-02-06:
Strange. The reported warning are harmless. It seems you did everything as expected.
Please double check the Neopixel configuration in `Configuration.h`
 
Do you have an oscilloscope to plug on the Neopixel's output pin ? 

Else, at least, can you enable `MARLIN_DEV_MODE` in `Configuration_adv.h`, build again, install the firmware, and run in the serial terminal `D6` and post the output here ?

I just want to make sure the delay code is working correctly on your system.

lightmaster, 2021-02-06:
It's kid's birthday party here today, so I'll work on that later tonight. I'll post the logs afterwards.

X-Ryl669, 2021-02-06:
Happy birthday, then!

lightmaster, 2021-02-06:
I'll tell my kid an internet stranger said Happy Birthday then. 🤣🤣

lightmaster, 2021-02-08:
No oscilloscope sadly.

Neopixel config looks fine to me: 
```
// Support for Adafruit NeoPixel LED driver
#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRB // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  #define NEOPIXEL_PIN     PA6       // LED driving pin
  // #define NEOPIXEL2_TYPE  NEO_GRB
  // #define NEOPIXEL2_PIN    PC7
  #define NEOPIXEL_PIXELS 48       // Number of LEDs in the strip. (Longest strip when NEOPIXEL2_SEPARATE is disabled.)
  #define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS   7  // Initial brightness (0-255)
  #define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup

  // Support for second Adafruit NeoPixel LED driver controlled with M150 S1 ...
  // #define NEOPIXEL2_SEPARATE
  #if ENABLED(NEOPIXEL2_SEPARATE)
    #define NEOPIXEL2_PIXELS      48  // Number of LEDs in the second strip
    #define NEOPIXEL2_BRIGHTNESS   5  // Initial brightness (0-255)
    // #define NEOPIXEL2_STARTUP_TEST    // Cycle through colors at startup
  #else
    //#define NEOPIXEL2_INSERIES      // Default behavior is NeoPixel 2 in parallel
  #endif
```

Building with `MARLIN_DEV_MODE` enabled now, will update with the output of that command when that's finished.

lightmaster, 2021-02-08:
Output of D6:
```
Send: D6
Recv: Computed delay calibration value: 33
Recv: Calling delay for 1cycles took: 3cycles
Recv: Calling delay for 5cycles took: 6cycles
Recv: Calling delay for 10cycles took: 11cycles
Recv: Calling delay for 20cycles took: 21cycles
Recv: Calling delay for 50cycles took: 51cycles
Recv: Calling delay for 100cycles took: 101cycles
Recv: Calling delay for 150cycles took: 151cycles
Recv: Calling delay for 200cycles took: 201cycles
Recv: Calling delay for 350cycles took: 351cycles
Recv: Calling delay for 500cycles took: 501cycles
Recv: Calling delay for 750cycles took: 751cycles
Recv: Calling delay for 1000cycles took: 1001cycles
Recv: Calling delay for 1cycles took: 36cycles
Recv: Calling delay for 5cycles took: 36cycles
Recv: Calling delay for 10cycles took: 36cycles
Recv: Calling delay for 20cycles took: 36cycles
Recv: Calling delay for 50cycles took: 55cycles
Recv: Calling delay for 100cycles took: 109cycles
Recv: Calling delay for 150cycles took: 154cycles
Recv: Calling delay for 200cycles took: 208cycles
Recv: Calling delay for 350cycles took: 352cycles
Recv: Calling delay for 500cycles took: 505cycles
Recv: Calling delay for 750cycles took: 757cycles
Recv: Calling delay for 1000cycles took: 1009cycles
Recv: Calling delay_dwt for 1cycles took: 38cycles
Recv: Calling DELAY_CYCLES directly  for 1cycles took: 1cycles
Recv: Calling DELAY_CYCLES directly  for 5cycles took: 6cycles
Recv: Calling DELAY_CYCLES directly  for 10cycles took: 11cycles
Recv: Calling DELAY_CYCLES directly  for 20cycles took: 22cycles
Recv: Calling DELAY_CYCLES directly  for 50cycles took: 55cycles
Recv: Calling DELAY_CYCLES directly  for 100cycles took: 100cycles
Recv: Calling DELAY_CYCLES directly  for 200cycles took: 208cycles
Recv: ok
```

lightmaster, 2021-02-08:
I tried to change the default Adafruit Neopixel library to use the new `Delay.h`, however in the file `Marlin/src/features/leds/neopixel.cpp`, there is no reference to a `Delay.h` file. Tried just putting the `#include...` for it in the file anyways, and it compiled without any errors but didn't seem to change how the Neopixel behaves, as it it light up and changed inconsistently like its been doing

X-Ryl669, 2021-02-08:
D6 output is perfect, it's working as expected. You've less than 2 cycles of error for small delays, so it should work. 
If the NeoPixels CPP file you have does not include Delay.h, it means it has no way to use the new Delay function (so it'll fallback to the old/imprecise delay loop). 
Adding the line is not working because the code in this particular file does not require it, so if you add a dependency that's not required, it'll just ignore it.

Can you delete the complete NeoPixel folder and post the output of platformio.ini you are using ? 
If it's good (I'll tell you that once you've posted it), it should re-download the library and, as such, find the right file. You don't need to modify any file in the NeoPixel library if it's good.

lightmaster, 2021-02-08:
contents of `platformio.ini`: https://pastebin.com/2Yse37gH

X-Ryl669, 2021-02-08:
You need to change this: 
```
NEOPIXEL_LED            = Adafruit NeoPixel@1.5.0
                          src_filter=+<src/feature/leds/neopixel.cpp>
```
by this:
```
NEOPIXEL_LED = Adafruit NeoPixel=https://github.com/ellensp/Adafruit_NeoPixel
```

Then run `platformio run -e <your env> -t clean` so you start from a good base.

lightmaster, 2021-02-08:
I added it to the `mks_robin_nano35_stm32` env specifically, like what was done on line `796` for the `STM32F103RC_meeb` env. That wouldn't work? Figured if this gets integrated into the `platformio.ini` for the main repo, shouldn't be in a way that effects other envs that don't need this.

X-Ryl669, 2021-02-08:
I'm not sure. Never tried your way.

lightmaster, 2021-02-08:
I'll try the other way when I get a chance (probably tomorrow morning) and see if it makes a difference.

Assuming it doesn't make a difference, since the delays seemed correct anyways, what next?

X-Ryl669, 2021-02-08:
You'll need to capture the build log just to make sure the right module is built. It should look like this:
```
Processing STM32F103RC_btt_512K (platform: ststm32@~10.0; board: genericSTM32F103RC; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 (10.0.1) > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32-maple 3.10000.201129 (1.0.0)
 - tool-stm32duino 1.0.1
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SoftwareSerialM> 1.0.0
|-- <TMCStepper> 0.7.1
|   |-- <SoftwareSerialM> 1.0.0
|-- <U8glib-HAL> 0.4.3
|   |-- <Wire> 1.0
|-- <STM32ADC> 1.0
|-- <EEPROM>
|-- <USBComposite for STM32F1> 0.99
|-- <Wire> 1.0
|-- <Adafruit Neopixel> @ 093cd8b
|-- <Servo(STM32F1)> 1.1.2
```

I've completely made up the text above from thin air, but the important part is the Adafruit Neopixel @ some hash code and not a version. If you have a version and not the hash code I've copied above, you end not using the right library.
If you do happen to have the right library, then you'll need to send me your binary (firmware.elf) file so I can analyze it. 
Thanks!

lightmaster, 2021-02-08:
Looks like its using the right library. i think i said this earlier, but i do have to modify the path to `Delay.h` cuz the relative path in that fork doesn't point to quite the right spot.
[firmware.elf.zip](https://github.com/MarlinFirmware/Marlin/files/5947417/firmware.elf.zip)

```
> Executing task: C:\Users\light\.platformio\penv\Scripts\pio.exe run <

Processing mks_robin_nano35_stm32 (platform: ststm32@~10.0; board: genericSTM32F103VE; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103VE.html
PLATFORM: ST STM32 (10.0.1) > STM32F103VE (64k RAM. 512k Flash)       
HARDWARE: STM32F103VET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 4.10900.200819 (1.9.0)
 - framework-cmsis 2.50501.200527 (5.5.1)
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 13 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TMCStepper> 0.7.1
|   |-- <SoftwareSerial> 1.0
|   |-- <SPI> 1.0
|-- <Adafruit NeoPixel> 1.2.4+sha.093cd8b
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <SoftwareSerial> 1.0
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
|-- <EEPROM> 2.0.1
Building in release mode
```

ellensp, 2021-02-08:
Delay.h path  was pre mks_robin_nano35_stm32 only works on older mks_robin_nano35

X-Ryl669, 2021-02-09:
Thanks @ellensp. So, if I follow your idea, @lightmaster, please add: 
```
#define TARGET_STM32F1
```
just after the `#include <...... delay.h>` in NeoPixels.cpp file. To make sure it's being accounted for, add a temporary `#error Here` after line 1903 (the line that read `#elif defined(TARGET_STM32F1)`) and build it. It should break compilation with an "error: Here" meaning it's compiling the right code. If it does, remove the `#error Here` line and build once again, it should work.

X-Ryl669, 2021-02-09:
Honestly, the Neopixel code is horrible. Depending on such code is very fragile IMHO. If you define TARGET_STM32F1 it will probably break on GPIO_SET and GPIO_CLEAR macro...

ellensp, 2021-02-09:
The whole concept of neopixels is horrible. it is not comparable with anything but a dedicated controller. Anything else is at best a kludge.

X-Ryl669, 2021-02-09:
Please apply this patch to the Adafruit_Neopixel.cpp file:
```
--- Adafruit_NeoPixel.cpp	2021-02-09 19:24:16.995616200 +0100
+++ ".pio/libdeps/mks_robin_nano35_stm32/Adafruit NeoPixel/Adafruit_NeoPixel.cpp"	2021-02-09 19:19:27.692274869 +0100
@@ -49,7 +49,9 @@
 //and enter full path to this file below

 //#include <replace with path to Delay.h file>
-#include "../../../src/HAL/shared/Delay.h"
+#include "../../../../Marlin/src/HAL/shared/Delay.h"
+#include "../../../../Marlin/src/HAL/STM32/fastio.h"
+#define TARGET_STM32F1

 #ifdef TARGET_LPC1768
@@ -1902,16 +1904,16 @@
 #endif
 #elif defined(TARGET_STM32F1)
   volatile uint8_t  *ptr, *end, p, bitMask;
-  volatile uint32_t  pinMask;
+  // volatile uint32_t  pinMask;

-  pinMask =  BIT(PIN_MAP[pin].gpio_bit);
+  // pinMask =  BIT(PIN_MAP[pin].gpio_bit);
   ptr     =  pixels;
   end     =  ptr + numBytes;
   p       = *ptr++;
   bitMask =  0x80;

-  #define GPIO_SET(IO)   (PIN_MAP[IO].gpio_device->regs->BSRR = pinMask)
-  #define GPIO_CLEAR(IO) (PIN_MAP[IO].gpio_device->regs->BRR =  pinMask)
+  #define GPIO_SET(IO)   WRITE(IO, 1) //(PIN_MAP[IO].gpio_device->regs->BSRR = pinMask)
+  #define GPIO_CLEAR(IO) WRITE(IO, 0) //(PIN_MAP[IO].gpio_device->regs->BRR =  pinMask)
 #ifdef NEO_KHZ400 // 800 KHz check needed only if 400 KHz support enabled
   if(is800KHz) {
 #endif
```
And report if it works for you. Thanks!

lightmaster, 2021-02-10:
Apply that to the Adafruit_NeoPixel from his fork, or the stock one that STM32 normally uses?

X-Ryl669, 2021-02-10:
From the fork. You might need to do it by hand, since the path is wrong in the diff.

lightmaster, 2021-02-10:
Won't be able to test it until I get home in a couple hours, but it does build without any major error.

lightmaster, 2021-02-10:
So that makes it stuck at full brightness for all 3 sets of LEDs, equivalent to `M150 R255 U255 B255 P255`

lightmaster, 2021-02-10:
Here's the `D6` output of the standard Neopixel stuff for `mks_robin_nano35_stm32`

```
Send: D6
Recv: Computed delay calibration value: 33

Recv: Calling delay for 1cycles took: 2cycles
Recv: Calling delay for 5cycles took: 6cycles
Recv: Calling delay for 10cycles took: 11cycles
Recv: Calling delay for 20cycles took: 22cycles
Recv: Calling delay for 50cycles took: 51cycles
Recv: Calling delay for 100cycles took: 101cycles
Recv: Calling delay for 150cycles took: 152cycles
Recv: Calling delay for 200cycles took: 201cycles
Recv: Calling delay for 350cycles took: 351cycles
Recv: Calling delay for 500cycles took: 502cycles
Recv: Calling delay for 750cycles took: 751cycles
Recv: Calling delay for 1000cycles took: 1001cycles

Recv: Calling delay for 1cycles took: 36cycles
Recv: Calling delay for 5cycles took: 36cycles
Recv: Calling delay for 10cycles took: 36cycles
Recv: Calling delay for 20cycles took: 36cycles
Recv: Calling delay for 50cycles took: 55cycles
Recv: Calling delay for 100cycles took: 109cycles
Recv: Calling delay for 150cycles took: 154cycles
Recv: Calling delay for 200cycles took: 208cycles
Recv: Calling delay for 350cycles took: 352cycles
Recv: Calling delay for 500cycles took: 505cycles
Recv: Calling delay for 750cycles took: 757cycles
Recv: Calling delay for 1000cycles took: 1009cycles

Recv: Calling delay_dwt for 1cycles took: 39cycles

Recv: Calling DELAY_CYCLES directly  for 1cycles took: 1cycles
Recv: Calling DELAY_CYCLES directly  for 5cycles took: 5cycles
Recv: Calling DELAY_CYCLES directly  for 10cycles took: 10cycles
Recv: Calling DELAY_CYCLES directly  for 20cycles took: 20cycles
Recv: Calling DELAY_CYCLES directly  for 50cycles took: 55cycles
Recv: Calling DELAY_CYCLES directly  for 100cycles took: 100cycles
Recv: Calling DELAY_CYCLES directly  for 200cycles took: 208cycles
```

X-Ryl669, 2021-02-10:
It's not clear. What you meant was it worked, but you can't control the pixel value, right ?
Can you send me your firmware.elf please ?

X-Ryl669, 2021-02-10:
Maybe I've inverted 0 and 1 in the WRITE() code above, so you can try to send a `M150 R255 U255 B255 P255` to get black ?
If the sequence of data to the led is correct that the WS2812B understand it as one, it's a good news. It can't happen by luck, the timing is too strict.

lightmaster, 2021-02-10:
So, it turns on at boot to full brightness, and stays that way with no control over it. On LCD, clicking on/off, color presents, and manual sliders does nothing, even when setting to 0.

I don't think I actually sent any commands directly via terminal, so I'll try modifying that file again from scratch and see.

lightmaster, 2021-02-10:
[firmware.elf.zip](https://github.com/MarlinFirmware/Marlin/files/5961048/firmware.elf.zip)

I modified the file as instructed again and same result: Neopixel stuck at 100% brightness, will not react to M150 and setting to full brightness does not turn it off either.

X-Ryl669, 2021-02-11:
Ok, it seems that there is something clunky with your build option:
The code:
```
DELAY_NS(700);
``` 
which, should expand to `#define DELAY_NS(x) DELAY_CYCLES((x) * ((F_CPU) / 1000000UL) / 1000UL)` that is: 
a compile time constant equal to 50 if you CPU is running at 72MHz

is compiled to this:
```
 803bb50:       486a            ldr     r0, [pc, #424]  ; (803bcfc <Adafruit_NeoPixel::show()+0x21c>)
 803bb52:       611a            str     r2, [r3, #16]
 803bb54:       f8d8 2000       ldr.w   r2, [r8]
 803bb58:       f8d9 3000       ldr.w   r3, [r9]
 803bb5c:       fbb2 f0f0       udiv    r0, r2, r0
 803bb60:       f44f 722f       mov.w   r2, #700        ; 0x2bc
 803bb64:       4342            muls    r2, r0
 803bb66:       f44f 707a       mov.w   r0, #1000       ; 0x3e8
 803bb6a:       fbb2 f0f0       udiv    r0, r2, r0
 803bb6e:       4798            blx     r3
 803bb70:       f9b4 3006       ldrsh.w r3, [r4, #6]
 803bb74:       2b4f            cmp     r3, #79 ; 0x4f
 803bb76:       d838            bhi.n   803bbea <Adafruit_NeoPixel::show()+0x10a>
```
Which is a runtime compilation of the equation above (with 2 very long divisions that are breaking the timing completely).
So, step 1, resave the Neopixel.cpp file so it'll be rebuild and run `platformio run -v -e mks_robin_nano_stm32`. It'll be verbose but it should rebuild the NeoPixels file and you'll see the actual compile command used. Please post it here so I can ensure it's compiled with all optimization turned on.

Step 2:
======

As a temporary workaround, can you change the lines "DELAY_NS(700)" (line 1926) with `DELAY_CYCLES(50)`, the line `DELAY_NS(150)` with `DELAY_CYCLES(11)`, the line DELAY_NS(100)` with `DELAY_CYCLES(7)` and `DELAY_NS(750)` with `DELAY_CYCLES(54)`

X-Ryl669, 2021-02-11:
Indeed, on `arduino_ststm32` package (not libmaple), F_CPU is defined to a runtime variable `uint32_t SystemCoreClock`, so the compiler can not optimize the ugly computation above. I don't know why they made it this way, but it completely breaks low level computation we need. 

I'm not sure I'll be able to solve this, since the minimum number of cycles for waiting on a run-time delay is around 35 cycles (see #20901 measurement or your own D6 above) so we can't reach the value 7 and 11 that's required for the WS2812B LED timing.

lightmaster, 2021-02-11:
> # So, step 1:
> 
> Re-save the Neopixel.cpp file so it'll be rebuild and run `platformio run -v -e mks_robin_nano_stm32`. It'll be verbose but it should rebuild the NeoPixels file and you'll see the actual compile command used. Please post it here so I can ensure it's compiled with all optimizations turned on.

This what you were looking for?
```
PS C:\Users\light\OneDrive\Documents\GitHub\Marlin_Q5> platformio run -v -e mks_robin_nano35_stm32
Processing mks_robin_nano35_stm32 (platform: ststm32@~10.0; extends: common_stm32; build_flags: -fmax-errors=5 -g -D__MARLIN_FIRMWARE__ -fmerge-constants, -std=gnu++14, -DUSBCON -DUSBD_USE_CDC, -DTIM_IRQ_PRIO=13, -DADC_RESOLUTION=12 -DMCU_STM32F103VE -DSS_TIMER=4 -DENABLE_HWSERIAL3; board: genericSTM32F103VE; board_build.core: stm32; board_build.variant: MARLIN_F103Vx; board_build.ldscript: 
ldscript.ld; board_build.offset: 0x7000; board_build.encrypt: Yes; board_build.firmware: Robin_nano.bin; board_upload.offset_address: 0x08007000; build_unflags: -std=gnu++11 -DUSBCON -DUSBD_USE_CDC; debug_tool: jlink; upload_protocol: jlink; custom_marlin.neopixel_led: Adafruit NeoPixel=https://github.com/ellensp/Adafruit_NeoPixel
src_filter=+<src/feature/leds/neopixel.cpp>; extra_scripts: pre:buildroot/share/PlatformIO/scripts/common-dependencies.py, pre:buildroot/share/PlatformIO/scripts/common-cxxflags.py, post:buildroot/share/PlatformIO/scripts/common-dependencies-post.py, pre:buildroot/share/PlatformIO/scripts/generic_create_variant.py, buildroot/share/PlatformIO/scripts/stm32_bootloader.py, buildroot/share/PlatformIO/scripts/mks_encrypt.py; src_filter: +<src/*> -<src/config> -<src/HAL> +<src/HAL/shared>, -<src/lcd/HD44780> -<src/lcd/TFTGLCD> -<src/lcd/dwin> -<src/lcd/dogm> -<src/lcd/tft> -<src/lcd/tft_io>, -<src/HAL/STM32/tft> -<src/HAL/STM32F1/tft>, -<src/lcd/menu>, -<src/lcd/menu/game/game.cpp> -<src/lcd/menu/game/brickout.cpp> -<src/lcd/menu/game/invaders.cpp>, -<src/lcd/menu/game/maze.cpp> -<src/lcd/menu/game/snake.cpp>, -<src/lcd/menu/menu_backlash.cpp>, -<src/lcd/menu/menu_bed_corners.cpp>, -<src/lcd/menu/menu_bed_leveling.cpp>, -<src/lcd/menu/menu_cancelobject.cpp>, -<src/lcd/menu/menu_delta_calibrate.cpp>, -<src/lcd/menu/menu_filament.cpp>, -<src/lcd/menu/menu_info.cpp>, -<src/lcd/menu/menu_job_recovery.cpp>, -<src/lcd/menu/menu_language.cpp>, -<src/lcd/menu/menu_led.cpp>, -<src/lcd/menu/menu_media.cpp>, -<src/lcd/menu/menu_mmu2.cpp>, -<src/lcd/menu/menu_password.cpp>, -<src/lcd/menu/menu_power_monitor.cpp>, -<src/lcd/menu/menu_spindle_laser.cpp>, -<src/lcd/menu/menu_temperature.cpp>, -<src/lcd/menu/menu_tmc.cpp>, -<src/lcd/menu/menu_touch_screen.cpp>, -<src/lcd/menu/menu_tramming.cpp>, -<src/lcd/menu/menu_ubl.cpp>, -<src/lcd/extui/lib/mks_ui>, -<src/lcd/extui/lib/dgus> -<src/lcd/extui/dgus_lcd.cpp>, -<src/lcd/extui/example.cpp>, -<src/lcd/extui/malyan_lcd.cpp>, -<src/lcd/extui/lib/ftdi_eve_touch_ui>, -<src/lcd/extui/anycubic_chiron_lcd.cpp> -<src/lcd/extui/lib/anycubic_chiron>, -<src/lcd/extui/anycubic_i3mega_lcd.cpp> -<src/lcd/extui/lib/anycubic_i3mega>, -<src/lcd/lcdprint.cpp>, -<src/lcd/touch/touch_buttons.cpp>, -<src/sd/usb_flashdrive/lib-uhs2> -<src/sd/usb_flashdrive/lib-uhs3>, -<src/sd/usb_flashdrive/Sd2Card_FlashDrive.cpp>, -<src/sd/cardreader.cpp> -<src/sd/Sd2Card.cpp> -<src/sd/SdBaseFile.cpp> -<src/sd/SdFatUtil.cpp> -<src/sd/SdFile.cpp> -<src/sd/SdVolume.cpp> -<src/gcode/sd>, -<src/HAL/shared/backtrace>, -<src/HAL/shared/eeprom_if_i2c.cpp>, -<src/HAL/shared/eeprom_if_spi.cpp>, -<src/feature/babystep.cpp>, -<src/feature/backlash.cpp>, -<src/feature/baricuda.cpp> -<src/gcode/feature/baricuda>, -<src/feature/bedlevel/abl> -<src/gcode/bedlevel/abl>, -<src/feature/bedlevel/mbl> -<src/gcode/bedlevel/mbl>, -<src/feature/bedlevel/ubl> -<src/gcode/bedlevel/ubl>, -<src/feature/binary_stream.cpp> -<src/libs/heatshrink>, -<src/feature/bltouch.cpp>, -<src/feature/cancel_object.cpp> -<src/gcode/feature/cancel>, -<src/feature/caselight.cpp> -<src/gcode/feature/caselight>, -<src/feature/closedloop.cpp>, -<src/feature/controllerfan.cpp> -<src/gcode/feature/controllerfan>, -<src/feature/dac> -<src/feature/digipot>, -<src/feature/direct_stepping.cpp> -<src/gcode/motion/G6.cpp>, -<src/feature/e_parser.cpp>, -<src/feature/encoder_i2c.cpp>, -<src/feature/ethernet.cpp> -<src/gcode/feature/network/M552-M554.cpp>, -<src/feature/fanmux.cpp>, -<src/feature/filwidth.cpp> -<src/gcode/feature/filwidth>, -<src/feature/fwretract.cpp> -<src/gcode/feature/fwretract>, -<src/feature/host_actions.cpp>, -<src/feature/hotend_idle.cpp>, -<src/feature/joystick.cpp>, -<src/feature/leds/blinkm.cpp>, -<src/feature/leds/leds.cpp>, -<src/feature/leds/neopixel.cpp>, -<src/feature/leds/pca9533.cpp>, -<src/feature/leds/pca9632.cpp>, -<src/feature/leds/printer_event_leds.cpp>, -<src/feature/leds/tempstat.cpp>, -<src/feature/max7219.cpp>, -<src/feature/meatpack.cpp>, -<src/feature/mixing.cpp>, -<src/feature/mmu/mmu.cpp>, -<src/feature/mmu/mmu2.cpp> -<src/gcode/feature/prusa_MMU2>, -<src/feature/password> -<src/gcode/feature/password>, -<src/feature/pause.cpp>, -<src/feature/power.cpp>, -<src/feature/power_monitor.cpp> -<src/gcode/feature/power_monitor>, -<src/feature/powerloss.cpp> -<src/gcode/feature/powerloss>, -<src/feature/probe_temp_comp.cpp>, -<src/feature/repeat.cpp>, -<src/feature/runout.cpp> -<src/gcode/feature/runout>, -<src/feature/solenoid.cpp> -<src/gcode/control/M380_M381.cpp>, -<src/feature/spindle_laser.cpp> -<src/gcode/control/M3-M5.cpp>, -<src/feature/tmc_util.cpp> -<src/module [... cut ...]
FYSETC_E4                              IGNORED
teensy31                               IGNORED
teensy35                               IGNORED
teensy36                               IGNORED
teensy41                               IGNORED
linux_native                           IGNORED
include_tree                           IGNORED
==================================================================================== 1 succeeded in 00:00:23.049 ====================================================================================
```

X-Ryl669, 2021-02-11:
Yes thanks. Please try step 2 and tell me the result.

lightmaster, 2021-02-11:
I'll be a couple hours before I'm home to reflash the firmware, but it's compiled and waiting now.

X-Ryl669, 2021-02-11:
Great!

ellensp, 2021-02-11:
Perhaps it would be an idea to  have a #define OVERIDE_F_CPU {frequency in question}  in the pins file for this board
And in the code a #if defined  OVERIDE_F_CPU  #define DELAY_NS(x) DELAY_CYCLES((x) * ((OVERIDE_F_CPU) / 1000000UL) / 1000UL) else use the normal #define.

X-Ryl669, 2021-02-11:
Yep. In fact, we could undef F_CPU and define it to the right value so there's less spaghetti code where it's being used. But that means knowing the CPU frequency of all the board that Marlin support. Do you know where I could find this information ?

ellensp, 2021-02-11:
Add printing F_CPU it to D6 dump.

X-Ryl669, 2021-02-11:
I'll do that. Meanwhile, I've also created #21048 to collect them. Let's see if it works.

lightmaster, 2021-02-11:
Alright, so with the change to using `DELAY_CYCLES`, I have control of the Neopixels back, but it's not really any better than the default library. Interestingly, if I send a command like `M150 R255 U0 B0 P70`, first time some pixels will be all red, some all green, some all blue. Send that same command again and they will all be red but too bright. Send it a 3rd time and it'll be closer to what I'd expect from the `P70` brightness. If I then send it different colors within a minute or so, they will also look like I'd expect. After waiting a couple minutes, the next command will act like the very first command and mix up the colors.

So sending multiple commands per minute will get it to change like I want, but a few minutes between commands reverts it back to being erratic.

X-Ryl669, 2021-02-11:
Can you send me again your firmware.elf ?

lightmaster, 2021-02-11:
Looking at the default neopixel library, it references `F_CPU`, which is defined in `stm32_def.h` as `SystemCoreClock`, which in turn is defined in `system_stm32f1xx.c as `16000000`. I assume that `16000000` is in Hz, so 16MHz. If this board actually runs at 72MHz, can I try redefining `F_CPU` in the neopixel library to `72000000` to see if it works better? Is there any risk of frying anything by modifying that value only for the neopixel library?

X-Ryl669, 2021-02-11:
Nope, I've already computed the number of cycles based on 72MHz so if you made the change DELAY_NS to DELAY_CYCLES, it'd work as-is.

lightmaster, 2021-02-11:
[firmware.elf.zip](https://github.com/MarlinFirmware/Marlin/files/5964975/firmware.elf.zip)

X-Ryl669, 2021-02-11:
The code is now correct. I can't do anything more without an oscilloscope now. From the behavior you're describing, it looks like that either:
1. The code is not executed with the same latency every time (since you have a STM32, there is an instruction cache but it should be more or less transparent when run twice in a row). Yet you need to run it 3x to get what you expect.
2. Something interrupts the NeoPixel::show method. It shouldn't happen since the function starts with a `noInterrupt` call, but only an oscilloscope could assert this.
3. The timing are not that good. Again, an oscilloscope would be required to ensure the timings on the pin. You can try to adjust the timing here:

If I use the WS2811 datasheet as a reference, the delay should be:

|  Bit value  | High time (min) | High time (avg) | High time (max) | Number of cycles at 72MHz |
|-|-|-|-|-|
| 1 (high) | 1050 ns | 1200ns | 1350ns | 86 |
| 1 (low)  |  1150 ns | 1300ns | 1450ns |  94 |
| 0 (high) |    350ns |  500ns | 650ns | 36 |
| 0 (low)  |  1850ns |  2000ns| 2150ns | 144 |

So you'll have to hack the delay cycles in the code accordingly:
```
      if(p & bitMask) {
        // data ONE
        // min: 650 typ: 800 max: 950
        GPIO_SET(pin);
        DELAY_CYCLES(86);  //high
        // min: 300 typ: 450 max: 600
        GPIO_CLEAR(pin);
        DELAY_CYCLES(94);  //low
      } else {
        // data ZERO
        // min: 250  typ: 400 max: 550
        GPIO_SET(pin);
        DELAY_CYCLES(36);  //high
        // data low
        // min: 700 typ: 850 max: 1000
        DELAY_CYCLES(144);
      }
```
If you have a WS2812b, then the values should be (in the order above): One: 58, 32, Zero: 29, 61 

Please try the one of the combination and report.

[19 more replies]

closed 2021-06-11 by github-actions[bot]