MarlinFirmware/Marlin #20984: [BUG] ("more" menu in LVGL, no description)

Malderin, 2021-02-03:
### Bug Description

When activating CUSTOM_USER_MENUS in configuration_adv.n, captions to the "more" menu buttons are not shown (except for the G-code button). Instead, it says "Text" everywhere.
[image]

### Configuration Files

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/5916922/Configuration_adv.zip)

Malderin, 2021-02-05:
I fixed it like this:
https://github.com/Malderin/Marlin_2.0.x_official/commit/bf987eafa3377ca9a8757e11eee21a0a03ae2377

sjasonsmith, 2021-02-07:
I am presuming that @Malderin's merged PR fixes this issue, since they both reported the bug and contributed the fix.

closed 2021-02-07 by sjasonsmith