MarlinFirmware/Marlin #20989: [BUG] No longer compiling for SAM/DUE

marciot, 2021-02-03:
When compiling for SAM/DUE, I get the following error:

```
In file included from src/HAL/DUE/../../inc/../core/boards.h:24:0,
                 from src/HAL/DUE/../../inc/MarlinConfigPre.h:37,
                 from src/HAL/DUE/../../inc/MarlinConfig.h:28,
                 from src/HAL/DUE/MarlinSerialUSB.cpp:30:
src/HAL/DUE/MarlinSerialUSB.cpp: In static member function 'static int MarlinSerialUSB::peek()':
src/HAL/DUE/MarlinSerialUSB.cpp:72:51: error: 'emergency_state' was not declared in this scope
   TERN_(EMERGENCY_PARSER, emergency_parser.update(emergency_state, (char)pending_char));
                                                   ^
src/HAL/DUE/../../inc/../core/macros.h:443:26: note: in definition of macro 'THIRD'
 #define THIRD(a,b,c,...) c
                          ^
src/HAL/DUE/../../inc/../core/macros.h:192:29: note: in expansion of macro '___TERN'
 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
                             ^~~~~~~
src/HAL/DUE/../../inc/../core/macros.h:191:29: note: in expansion of macro '__TERN'
 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
                             ^~~~~~
src/HAL/DUE/../../inc/../core/macros.h:190:29: note: in expansion of macro '_TERN'
 #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION converted to A or '<nul>'
                             ^~~~~
src/HAL/DUE/MarlinSerialUSB.cpp:72:3: note: in expansion of macro 'TERN_'
   TERN_(EMERGENCY_PARSER, emergency_parser.update(emergency_state, (char)pending_char));
   ^~~~~
src/HAL/DUE/MarlinSerialUSB.cpp: In static member function 'static int MarlinSerialUSB::read()':
src/HAL/DUE/MarlinSerialUSB.cpp:94:51: error: 'emergency_state' was not declared in this scope
   TERN_(EMERGENCY_PARSER, emergency_parser.update(emergency_state, (char)c));
                                                   ^
src/HAL/DUE/../../inc/../core/macros.h:443:26: note: in definition of macro 'THIRD'
 #define THIRD(a,b,c,...) c
                          ^
src/HAL/DUE/../../inc/../core/macros.h:192:29: note: in expansion of macro '___TERN'
 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
                             ^~~~~~~
src/HAL/DUE/../../inc/../core/macros.h:191:29: note: in expansion of macro '__TERN'
 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
                             ^~~~~~
src/HAL/DUE/../../inc/../core/macros.h:190:29: note: in expansion of macro '_TERN'
 #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION converted to A or '<nul>'
                             ^~~~~
src/HAL/DUE/MarlinSerialUSB.cpp:94:3: note: in expansion of macro 'TERN_'
   TERN_(EMERGENCY_PARSER, emergency_parser.update(emergency_state, (char)c));
   ^~~~~
Makefile:969: recipe for target 'applet/src/HAL/DUE/MarlinSerialUSB.o' failed
make: *** [applet/src/HAL/DUE/MarlinSerialUSB.o] Error 1
```

rhapsodyv, 2021-02-03:
@X-Ryl669 related?

ellensp, 2021-02-04:
Please attach your configuration files.

X-Ryl669, 2021-02-04:
@marciot: Can you apply the following patch and report if it's ok for you?
```
diff --git a/Marlin/src/HAL/DUE/MarlinSerialUSB.cpp b/Marlin/src/HAL/DUE/MarlinSerialUSB.cpp
index d85aaf14b0..3d73261c2c 100644
--- a/Marlin/src/HAL/DUE/MarlinSerialUSB.cpp
+++ b/Marlin/src/HAL/DUE/MarlinSerialUSB.cpp
@@ -33,10 +33,6 @@

 #include "MarlinSerialUSB.h"

-#if ENABLED(EMERGENCY_PARSER)
-  #include "../../feature/e_parser.h"
-#endif
-
 // Imports from Atmel USB Stack/CDC implementation
 extern "C" {
   bool usb_task_cdc_isenabled();
@@ -69,7 +65,7 @@ int MarlinSerialUSB::peek() {

   pending_char = udi_cdc_getc();

-  TERN_(EMERGENCY_PARSER, emergency_parser.update(emergency_state, (char)pending_char));
+  TERN_(EMERGENCY_PARSER, emergency_parser.update(static_cast<MSerialT*>(this)->emergency_state, (char)pending_char));

   return pending_char;
 }
@@ -91,7 +87,7 @@ int MarlinSerialUSB::read() {

   int c = udi_cdc_getc();

-  TERN_(EMERGENCY_PARSER, emergency_parser.update(emergency_state, (char)c));
+  TERN_(EMERGENCY_PARSER, emergency_parser.update(static_cast<MSerialT*>(this)->emergency_state, (char)c));

   return c;
 }
```

X-Ryl669, 2021-02-04:
@rhapsodyv Since the serial base is already implementing emergency parser, hence, all child class have it. I'm wondering if it's still worth it to test for EMERGENCY_PARSER's macro and instead apply it unconditionally everywhere. That would prevent such bug that are hard to sport due to the numerous possible combination.

marciot, 2021-02-04:
@X-Ryl669: I get a different error now:

```
In file included from src/HAL/DUE/../../inc/../core/boards.h:24:0,
                 from src/HAL/DUE/../../inc/MarlinConfigPre.h:37,
                 from src/HAL/DUE/../../inc/MarlinConfig.h:28,
                 from src/HAL/DUE/MarlinSerialUSB.cpp:30:
src/HAL/DUE/MarlinSerialUSB.cpp: In static member function 'static int MarlinSerialUSB::peek()':
src/HAL/DUE/MarlinSerialUSB.cpp:68:74: error: 'this' is unavailable for static member functions
   TERN_(EMERGENCY_PARSER, emergency_parser.update(static_cast<MSerialT*>(this)->emergency_state, (char)pending_char));
                                                                          ^
src/HAL/DUE/../../inc/../core/macros.h:443:26: note: in definition of macro 'THIRD'
 #define THIRD(a,b,c,...) c
                          ^
src/HAL/DUE/../../inc/../core/macros.h:192:29: note: in expansion of macro '___TERN'
 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
                             ^~~~~~~
src/HAL/DUE/../../inc/../core/macros.h:191:29: note: in expansion of macro '__TERN'
 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
                             ^~~~~~
src/HAL/DUE/../../inc/../core/macros.h:190:29: note: in expansion of macro '_TERN'
 #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION converted to A or '<nul>'
                             ^~~~~
src/HAL/DUE/MarlinSerialUSB.cpp:68:3: note: in expansion of macro 'TERN_'
   TERN_(EMERGENCY_PARSER, emergency_parser.update(static_cast<MSerialT*>(this)->emergency_state, (char)pending_char));
   ^~~~~
src/HAL/DUE/MarlinSerialUSB.cpp: In static member function 'static int MarlinSerialUSB::read()':
src/HAL/DUE/MarlinSerialUSB.cpp:90:74: error: 'this' is unavailable for static member functions
   TERN_(EMERGENCY_PARSER, emergency_parser.update(static_cast<MSerialT*>(this)->emergency_state, (char)c));
                                                                          ^
src/HAL/DUE/../../inc/../core/macros.h:443:26: note: in definition of macro 'THIRD'
 #define THIRD(a,b,c,...) c
                          ^
src/HAL/DUE/../../inc/../core/macros.h:192:29: note: in expansion of macro '___TERN'
 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
                             ^~~~~~~
src/HAL/DUE/../../inc/../core/macros.h:191:29: note: in expansion of macro '__TERN'
 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
                             ^~~~~~
src/HAL/DUE/../../inc/../core/macros.h:190:29: note: in expansion of macro '_TERN'
 #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION converted to A or '<nul>'
                             ^~~~~
src/HAL/DUE/MarlinSerialUSB.cpp:90:3: note: in expansion of macro 'TERN_'
   TERN_(EMERGENCY_PARSER, emergency_parser.update(static_cast<MSerialT*>(this)->emergency_state, (char)c));
   ^~~~~
Makefile:969: recipe for target 'applet/src/HAL/DUE/MarlinSerialUSB.o' failed
make: *** [applet/src/HAL/DUE/MarlinSerialUSB.o] Error 1
make: *** Waiting for unfinished jobs....
```

It looks like `MarlinSerialUSB::peek()` is static.

X-Ryl669, 2021-02-04:
I'm sorry, you haven't attached your configuration files, so I can't build it myself.
Meanwhile, please replace MarlinSerialUSB.h class by this (remove all static declaration here):
```
struct MarlinSerialUSB {
  void begin(const long);
  void end();
  int peek();
  int read();
  void flush();
  bool available();
  size_t write(const uint8_t c);

  #if ENABLED(SERIAL_STATS_DROPPED_RX)
    FORCE_INLINE uint32_t dropped() { return 0; }
  #endif

  #if ENABLED(SERIAL_STATS_MAX_RX_QUEUED)
    FORCE_INLINE int rxMaxEnqueued() { return 0; }
  #endif
};
```

marciot, 2021-02-04:
@X-Ryl669: I get a different error now:

```
src/HAL/DUE/MarlinSerialUSB.cpp:104:31: error: no 'void MarlinSerialUSB::flushTX()' member function declared in class 'MarlinSerialUSB'
 void MarlinSerialUSB::flushTX() { }
```

I am attaching my config files so you can compile it yourself.

[config.zip](https://github.com/MarlinFirmware/Marlin/files/5927094/config.zip)

X-Ryl669, 2021-02-04:
You can remove the implementation of flushTX in the CPP file. Since it's empty, there is no point keeping it. The base class will detect it and use its own empty method instead.

marciot, 2021-02-04:
@X-Ryl669: That did the trick. It now compiles! Thank you!

marciot, 2021-02-04:
> You can remove the implementation of flushTX in the CPP file. Since it's empty, there is no point keeping it. The base class will detect it and use its own empty method instead.
> 
> BTW, your configuration does not build. It's missing some pin definition.

Ummm, sorry about that. I am using a Marlin fork which isn't exactly the same as upstream Marlin.

marciot, 2021-02-04:
@X-Ryl669: Thank you for your help! I don't know whether you want to create a PR for it or whether I should do it, but here is a patch with all the changes I made with your help:

[0001-Fix-for-compilation-error-on-SAM-DUE.zip](https://github.com/MarlinFirmware/Marlin/files/5927450/0001-Fix-for-compilation-error-on-SAM-DUE.zip)

ellensp, 2021-02-05:
Created a PR based on the patch. Confirmed it compiles

X-Ryl669, 2021-02-05:
Sorry @ellensp, I wasn't clear. I've already included the fix in the #20985. I'm fixing serial bugs in there. So there was no need for a new PR, you can probably close yours. Thank for you work, and again sorry for wasting your time.

ellensp, 2021-02-10:
@X-Ryl669 since #20985 has been merged. Is this sorted?

marciot, 2021-02-10:
Yeah, all good!

closed 2021-02-10 by marciot