MarlinFirmware/Marlin #20990: RESTORE_LEVELING_AFTER_G28 not works on AUTO_BED_LEVELING_LINEAR

mirecta, 2021-02-03:
I have 2 z endstops endstops is under bed level and bltouch 
###Reproduce
G28 
G29
M500
G1 F500 X0 Y0 Z0 ; goto zero nicely 
if i do G28
and then G1 F500 X0 Y0 Z0; hotend crash to bed 

[conf.zip](https://github.com/MarlinFirmware/Marlin/files/5921836/conf.zip)

ellensp, 2021-02-04:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

mirecta, 2021-02-07:
ok i will try it and will rite about it

mirecta, 2021-02-07:
So with branch bugfix-2.0.x it doesn't works too

Msaza1, 2021-02-19:
> I have 2 z endstops endstops is under bed level and bltouch
> ###Reproduce
> G28
> G29
> M500
> G1 F500 X0 Y0 Z0 ; goto zero nicely
> if i do G28
> and then G1 F500 X0 Y0 Z0; hotend crash to bed
> 
> [conf.zip](https://github.com/MarlinFirmware/Marlin/files/5921836/conf.zip)
(https://www.msaza.com)

Msaza1, 2021-02-19:
``gcloud sql connect InstanceIdHere \
    --user=root

thisiskeithb, 2021-04-09:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

closed 2021-06-20 by github-actions[bot]

mirecta, 2021-08-04:
ok i will do it