MarlinFirmware/Marlin #20991: [BUG/FR] SPI MODE sanity check ask to define Step/Dir pins , but UNUSED. (Don't compile when using these GPIOS elsewhere)

studiodyne, 2021-02-04:
SPI MODE  don't need Step/Dir pins , and these pins can be used  for any other functionnality , but impossible to compile if Step/Dir are not defined for steppers

Please modifiy sanity check 

Thks

ManuelMcLure, 2021-02-04:
SPI is only used to configure the drivers. Step/Dir are still used for motion.

ellensp, 2021-02-04:
ManuelMcLure is correct. All marlin stepper drivers use STEP and DIRECTION pins, other interfaces are for setup and configurations

closed 2021-02-04 by ellensp

studiodyne, 2021-02-04:
Wrong ! UART is just a setup interface
SPI is a full motion and setup bus , and if the drivers is in SPI mode , step/dir not functionnal 
Please reopen

ellensp, 2021-02-04:
not in marlin

studiodyne, 2021-02-04:
I want to cry :cry 
I need GPIOS ......
Thks