MarlinFirmware/Marlin #20993: [BUG] No Homing after level the corners

Seelenkind, 2021-02-04:
### Bug Description

No homing after leveling the corners

### Configuration Files

`Configuration.h` 

#if ENABLED(LEVEL_BED_CORNERS)
  #define LEVEL_CORNERS_INSET_LFRB { 60, 60, 60, 60 } // (mm) Left, Front, Right, Back insets
  #define LEVEL_CORNERS_HEIGHT      -0.3  // (mm) Z height of nozzle at leveling points
  #define LEVEL_CORNERS_Z_HOP       4.0   // (mm) Z height of nozzle between leveling points
  #define LEVEL_CENTER_TOO              // Move to the center after the last corner
  //#define LEVEL_CORNERS_USE_PROBE
  #if ENABLED(LEVEL_CORNERS_USE_PROBE)
    #define LEVEL_CORNERS_PROBE_TOLERANCE 0.1
    #define LEVEL_CORNERS_VERIFY_RAISED   // After adjustment triggers the probe, re-probe to verify
    //#define LEVEL_CORNERS_AUDIO_FEEDBACK
  #endif

### Steps to Reproduce
1. level corner at printer display
2. 
**Expected behavior:**

Homing should be carried out after leveling the corners (optionally the middle)

**Actual behavior:**

After bedleveling via the printer menu, the nozzle remains on the bed and does not move to the home position. When the bed and nozzle are heated up, holes appear very quickly on the pressure plate.

#### Additional Information

NezSez, 2021-02-04:
Just to clarify initial conditions for troubleshooting, have you checked:

  Z_AFTER_PROBING
  Z_AFTER_HOMING

in Configuration.h ?

ellensp, 2021-02-05:
Please attach your configuration files.

Seelenkind, 2021-02-05:
The point is not that the nozzle is raised, of course that is a problem because the nozzle remains on the print bed, but that no homing is carried out after the corners have been leveled.

Link to config files.
[https://github.com/Seelenkind/Marlin-2.0.x-BTT-SKR-E3-MINI-V2.0-DE](url)

qwewer0, 2021-02-05:
Tested it on the latest bugfix-2.0.x, and after the leveling is done, the nozzle moves to `LEVEL_CORNERS_Z_HOP` Z height.
https://github.com/MarlinFirmware/Marlin/blob/708ea3d0bb82f067424c75636c28a549aa3d3b06/Marlin/src/lcd/menu/menu_bed_corners.cpp#L332
So for me it works as intended.

sarvenn, 2021-02-08:
After selecting "Done" nozzle raises off the bed. I just tried it with the latest bugfix firmware. I also use skr mini e3 v2 on my ender3 pro as well as tft 35 e3.

Seelenkind, 2021-02-09:
OK, apparently the problem is with 2.0.7.2 when it works with BugFix 2.0.x. I have to see what the difference is in the firmware.

sarvenn, 2021-02-16:
Have you checked it with the latest bugfix firmware? @Seelenkind
2.07.2'de de homing yapıyor muydu, hatırlamıyorum, ama yapıyordu sanırım.

Seelenkind, 2021-02-18:
> Have you checked it with the latest bugfix firmware? @Seelenkind
> 2.07.2'de de homing yapıyor muydu, hatırlamıyorum, ama yapıyordu sanırım.

Haven't tried it yet, but it really is because I have to use the BugFix version. 

sanırım 2.0.x BugFix kullanılmalı . Her neyse 2.0.7.2 sorunu olmalı . BL Touch taktım şimdi , zaten G-Code kullanıyorum elden Level yapmak için

thisiskeithb, 2021-04-10:
> OK, apparently the problem is with 2.0.7.2 when it works with BugFix 2.0.x.

Excellent!

closed 2021-04-10 by thisiskeithb