MarlinFirmware/Marlin #20995: anet FEEDRATE

MacSimov, 2021-02-04:


MacSimov, 2021-02-04:
1.Download marlinfirmware (https://github.com/MarlinFirmware/Marlin)
2.Download Configurations  Marlin 2.0 with bug fixes (https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x)
3.Use VSCode+platformio+AutoBuildMarlin
  Start build. Result: error in Marlin\src\motion.cpp file:

VScode print:
In file included from Marlin\src\module\../inc/MarlinConfig.h:47:0,
                 from Marlin\src\module\motion.h:31,
                 from Marlin\src\module\motion.cpp:27:
Marlin\src\module\motion.cpp:154:16: error: 'HOMING_FEEDRATE_XY' was not declared in this scope
     MMM_TO_MMS(HOMING_FEEDRATE_XY), MMM_TO_MMS(HOMING_FEEDRATE_XY),
                ^
Marlin\src\module\../inc/../core/types.h:76:43: note: in definition of macro 'MMM_TO_MMS'
 #define MMM_TO_MMS(MM_M) feedRate_t(float(MM_M) / 60.0f)
                                           ^
Marlin\src\module\motion.cpp:154:48: error: 'HOMING_FEEDRATE_XY' was not declared in this scope
     MMM_TO_MMS(HOMING_FEEDRATE_XY), MMM_TO_MMS(HOMING_FEEDRATE_XY),
                                                ^
Marlin\src\module\../inc/../core/types.h:76:43: note: in definition of macro 'MMM_TO_MMS'
 #define MMM_TO_MMS(MM_M) feedRate_t(float(MM_M) / 60.0f)
                                           ^
Marlin\src\module\motion.cpp:156:14: error: 'HOMING_FEEDRATE_Z' was not declared in this scope
   MMM_TO_MMS(HOMING_FEEDRATE_Z)
              ^
Marlin\src\module\../inc/../core/types.h:76:43: note: in definition of macro 'MMM_TO_MMS'
 #define MMM_TO_MMS(MM_M) feedRate_t(float(MM_M) / 60.0f)
                                           ^
Marlin\src\module\motion.cpp: In function 'void do_z_clearance(const float&, bool, bool, bool)':
Marlin\src\module\motion.cpp:516:100: error: 'HOMING_FEEDRATE_Z' was not declared in this scope
   do_blocking_move_to_z(_MIN(zdest, Z_MAX_POS), MMM_TO_MMS(TERN(HAS_BED_PROBE, Z_PROBE_SPEED_FAST, HOMING_FEEDRATE_Z)));
                                                                                                    ^
Marlin\src\module\../inc/../core/types.h:76:43: note: in definition of macro 'MMM_TO_MMS'
 #define MMM_TO_MMS(MM_M) feedRate_t(float(MM_M) / 60.0f)
                                           ^
Marlin\src\module\../inc/../core/macros.h:196:29: note: in expansion of macro 'THIRD'
 #define ___TERN(P,V...)     THIRD(P,V)              // If first argument has a comma, A. Else B.
                             ^
Marlin\src\module\../inc/../core/macros.h:195:29: note: in expansion of macro '___TERN'
 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
                             ^
Marlin\src\module\../inc/../core/macros.h:194:29: note: in expansion of macro '__TERN'
 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
                             ^
Marlin\src\module\../inc/../core/macros.h:190:29: note: in expansion of macro '_TERN'
 #define TERN(O,A,B)         _TERN(_ENA_1(O),B,A)    // OPTION converted to '0' or '1'
                             ^
Marlin\src\module\motion.cpp:516:60: note: in expansion of macro 'TERN'
                                                            ^
At global scope:
cc1plus.exe: warning: unrecognized command line option '-Wno-register'
*** [.pio\build\sanguino1284p\src\src\module\motion.cpp.o] Error 1

I add this:

#define HOMING_FEEDRATE_XY 500
#define HOMING_FEEDRATE_Z 500

in Configuration.h file.
And then rebuilt it. It build with success status.
I'm not yet upload it in printers board, but thinck it will be work.

Thansk a lot for big and good job. 

Aleksandr Maksimov from Russia.
maksimovaleksandr@inbox.ru

thisiskeithb, 2021-02-04:
You can’t mix and match versions of Marlin. If you’re using `bugfix-2.0.x` configs, you have to use the `bugfix-2.0.x` version of Marlin.
<hr>
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-02-04 by thisiskeithb

MacSimov, 2021-02-06:
Realy? May be do you know where  bugfix-2.0.x version of Marlin. I didn't see it on officiale site https://github.com/MarlinFirmware

ellensp, 2021-02-06:
https://github.com/MarlinFirmware/Marlin   then select the version you want with the switch branch pull down. default is 2.0.x select bugfix-2.0.x for bugfix then download.
Same with Config examples https://github.com/MarlinFirmware/Configurations. Default is Import-2.0.x DO NOT USE THIS!, import is for configuration developer use only. Use switch branch pull down to select bugfix-2.0.x then download.

MacSimov, 2021-02-07:
Oh, Thanks. I find it. I will try to use it.