MarlinFirmware/Marlin #20999: [BUG] [A FIX INCLUDED] Spindle PWM is very wrong (overflow on 8 bit boards ?)

nessotrin, 2021-02-05:
### Bug Description

When using spindle control in RPM mode, speed is completely wrong. It works fine when using 0-255 mode.
PWM values measured on the board are wildly inconsistent to what they should be. Values go up and down when increasing speed linearly, sometimes going back to zero and always much too low.
**But only when configured max RPM is above maximum 8 bit values. Target is 8 bit, Arduino mega based.**

I traced it to spindle_laser.h (line 56):
`return unitPower ? round(100 * (cpwr - SPEED_POWER_FLOOR) / (SPEED_POWER_MAX - SPEED_POWER_FLOOR)) : 0;`

**It's supposed to convert the speed to percentage of max speed, but 1500 out of 1500 returns 12%**, so it's broken.

=> It seems to work properly when replaced with:
`return unitPower ? round(100.0f * (cpwr - SPEED_POWER_FLOOR) / (SPEED_POWER_MAX - SPEED_POWER_FLOOR)) : 0;
`

I don't have time to check this properly. Hopefully someone can look into it.
Values seem to overflow as speed goes back to 0 at regular RPM intervals.

### Configuration Files

I don't think it's needed, but there : 
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5930118/Marlin.zip)

### Steps to Reproduce

1. Configure a board with SPINDLE_FEATURE, CUTTER_POWER_UNIT as RPM and CUTTER_POWER_RELATIVE
2. Set SPEED_POWER_MAX to 1500 (max RPM of motor)
3. Set spindle to max speed using : 'M3 S1500'
4. Measure PWM pin with multimeter

**Expected behavior:**

Measured PWM output to be 5V (100% PWM).
Debug log shows cpwr_to_pct(1500) returns 100.

**Actual behavior:**

Measured PWM output is 0.48V.
Debug log shows cpwr_to_pct(1500) returns 12.

#### Additional Information

Tested on 8 bit target only (Megatronics V3.1, DIY CNC).
Tested on latest 'bugfix-2.0.x'. I never used that feature before, so I don't know if it's new.
Compiled on Linux with latest PlatformIO (Deviot under Sumblime Text 3).

thisiskeithb, 2021-02-05:
Since you've provided a fix, it'd be faster if you submitted a [Pull Request](https://marlinfw.org/docs/development/getting_started_pull_requests.html).

ellensp, 2021-02-05:
Config files are important as it shows your not running bugfix.

this issue was found and patched in bugfix in https://github.com/MarlinFirmware/Marlin/pull/19970

closed 2021-02-05 by ellensp