MarlinFirmware/Marlin #21000: [BUG] (Einsy Rambo 1.2b SD card init fail)

dzario, 2021-02-05:
### Instructions
Eins Rambo 1.2b 
LCD - RepRapDiscount Smart Controller
marlin 2.0.7.2
Platformio : Core 5.1.1a1 Home 3.3.3

### Bug Description
SD init fail
LCD not working

I have new original Prusa Einsy Rambo 1.2b main board , with original Prusa LCD .
I test Marlin 2.0.7.2 , Platformio , all are latest version, update , if compile no error.

### Configuration Files
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5931811/configuration.zip)
### Steps to Reproduce
1. After flash firmware on LCD bootloader show : Original Prusa i3 , Prusa Research ... 
2. Then show mishmash (unreadable character) 
3. Connect USB and from any Terminal see Marlin Gcode respond.
4. Remove SD card from slot , see in terminal Card remove
5. Insert card to slot . terminal answer is SD Init Fail
I don't want solve problem with LCD , I need solve problem with SD card.
[https://photos.app.goo.gl/wWMQvR41GegwsCqu9](url)

I want configure with I2C OLED LCD (working) , but SD card not working, Prusa LCD use just for SD card slot and encoder , in configuration file not define Prusa LCD , only OLED . 
Prusa LCD connect only for testing if hardware is OK. Hardware is OK , tested with original Prusa firmware.

Einsy Rambo 1.2b have TMC2130 in SPI mode , on same SPI is connected SD card slot.
All other function working, stepp motor , sensor less homing , etc .. only SD card not working.

ellensp, 2021-02-05:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

dzario, 2021-02-05:
Hi. Still same problem.
On first .
Test with original Prusa LCD , this show on LCD : [https://photos.app.goo.gl/LqfNmtCBBGjuLbAA7](url)
Let to focus to start working Marlin 2.0.x with Prusa MK3/S printer.

ellensp, 2021-02-06:
When you built your firmware did you use set `default_envs = rambo` in platformio? I suspect not
If you just use the default RAMPS environment you get LCD issues and SD issue as they both use as pins 70-85 which are not defined in RAMPS.

dzario, 2021-02-06:
this is in my platformio.ini file .. 
`src_dir      = Marlin`
`boards_dir   = buildroot/share/PlatformIO/boards`
`default_envs = mega2560`
`include_dir  = Marlin`
I change to :
`src_dir      = Marlin`
`boards_dir   = buildroot/share/PlatformIO/boards`
`default_envs = rambo`
`include_dir  = Marlin`
Compile ... flash .. now working ..
**Problem solving ...** 
I apologize to all for this stupid question .. but I don't know, so I need edit more parameter in platformio.ini
I just write mega2560 , Einsy Rambo board have this MCU .
in Configuration.h just define `#define MOTHERBOARD BOARD_EINSY_RAMBO`
Than why rambo not einsy rambo ?? where I can find list of board that I can define for platformio.ini??

Perfect .. thanks ...

ellensp, 2021-02-06:
List of all boards see Marlin/src/core/boards.h
List of environment you need to use for each board Marlin/src/pins/pins.h

eg BOARD_EINSY_RAMBO
In boards.h   (note: you need to drop the "BOARD_" prefix 
```
#elif MB(EINSY_RAMBO)
  #include "rambo/pins_EINSY_RAMBO.h"           // ATmega2560                             env:rambo
```

ellensp, 2021-02-06:
no you don't need to edit anything else in platformio, unless you adding a new unsupported board.

closed 2021-02-06 by ellensp

Milan-prdp, 2021-02-24:
> ### Instructions
> Eins Rambo 1.2b
> LCD - RepRapDiscount Smart Controller
> marlin 2.0.7.2
> Platformio : Core 5.1.1a1 Home 3.3.3
>
> ### Bug Description
> SD init fail
> LCD not working
>
> I have new original [...]

Milan-prdp, 2021-02-24:
I too facing same issue. Anyone help

ellensp, 2021-02-24:
use build environment rambo not mega2560

Milan-prdp, 2021-02-24:
> use build environment rambo not mega2560
> ie in platformio.ini set `default_envs = rambo`

I've 3 new einsy controllers , one board is working fine with Marlin 2.0 and other 2 shows init fail message. I used same program for all the boards. This make me more confusing.