MarlinFirmware/Marlin #21003: [RFC] Rename the bugfix branch

X-Ryl669, 2021-02-05:
Just a small question to ask to developers. 
I think the bugfix branch should be called develop instead, since it's not about bug fixing only (or I'm completely wrong here) but more about improving Marlin. It's a semantic issue, not a logical issue.

IMHO, the pros for a develop branch:
* It's clear the branch is unstable (while bugfix gives the idea it's more stable than the main branch)
* We can branch from the develop branch to test new ideas (a bit like we are doing now), but when looking at the history, it's more clear from where we started instead of a unique, huge bunch of bugfix's commit.

The cons are:  
* Probably less people will test the develop branch, since it's not called "bugfix" anymore. I don't know.
* The bugfix branch will then have to reset to the current release branch (since if I understand correctly, the release branch contains actual bugfix), so it might break workflow of those syncing with bugfix to get the last release.

What do you think ?

NezSez, 2021-02-05:
Would it be better to have two separate branches, one for development of "new" features, and one for "fixing" current release bugs? If so, would one want the dev branch based on the bugfix branch, or directly off the 2.0.x release branch?
Would doing any of these things help the current devs enough to warrant the effort/changes?
Would doing this just create new release merge nightmares for not much benefit?
Would it be beneficial for those with production print farms by allowing them to sync to just bugfixes and not have to worry about new features introducing new bugs?
Would this encourage vendors to contribute to development, rather than just leaching as seems to be the current situation?

Obviously I've over simplified; just some questions to stimulate a discussion so that any changes made can be well considered and on public record.

X-Ryl669, 2021-02-05:
> Would it be better to have two separate branches, one for development of "new" features, and one for "fixing" current release bugs? If so, would one want the dev branch based on the bugfix branch, or directly off the 2.0.x release branch?

I might be wrong, but I think the branch called `2.0.x` is the actual bugfix for last release branch. So the development branch is already the de-facto *bugfix* branch.

rhapsodyv, 2021-02-05:
> I might be wrong, but I think the branch called `2.0.x` is the actual bugfix for last release branch. So the development branch is already the de-facto _bugfix_ branch.

Not entirely true, neither false. The `2.0.x` will have very important fixes, by any Scott criteria... But it will almost never generate a patch release... 

Marlin doesn't really follow any release procedure or police... We kind of follow something like a rolling release (like arch linux)... but not entirely...

Right now, our release police is basically Scott will and time. Time to time this subject pops up, but everything depends on Scott. I guess the big part of the answer is lack of time. If we try organize it more, it will demand more time from him.

Particularly, I think we should separate bugfix from new features, like splitting bugfix in two: bugfix (that will generate patch releases only) and develop (or even master) that will generate minor/major releases.

thisiskeithb, 2021-02-06:
There used to be a "dev" branch meant for new features and not bug fixes, but then users would clone that branch & sit on it thinking they were running the latest code. The dev branch would become outdated since patches & new features were still going into the bugfix branch.

NezSez, 2021-02-06:
So currently, to test alpha/new code, do devs just pull branches from each other's forks, or is literally everything pulled into bugfix before it's available at all?

Vertabreak, 2021-02-06:
i personally do testing on my fork and try to polish it up before i pull request upstream.

NezSez, 2021-02-06:
So no way to submit for ppl to test against other platforms/ARCHs (like SCARA, DELTAS, etc) before you make a PR then?
To weed out any unintended disastrous consequences before just making it available to the masses?

ManuelMcLure, 2021-02-06:
You can point people to your fork/branch if you want them to test.

X-Ryl669, 2021-02-07:
With more than 13 thousand forks, it's very unlikely people will find your fork and you won't get a sufficient number of people to test your fork.

X-Ryl669, 2021-02-07:
> There used to be a "dev" branch meant for new features and not bug fixes, but then users would clone that branch & sit on it thinking they were running the latest code. The dev branch would become outdated since patches & new features were still going into the bugfix branch.

I'm not sure it's clear. I don't think we should have a dev branch and a bugfix branch with only the latter being updated. I was more or less thinking like this:

```
               +--> Add new feature +---------+  <--- Test ok for new feature
               |                              |
Release +----+-+----------+---------------^---v-------------------->  develop
             |            |               |
             +--------> bug fix for XXX +-+  <----- Test ok for bug fix XXX
                          |
                          +------------------->  bug fix for YYY
```
which is very close to how the bugfix branch is actually used. 

In fact, the question boils down to *should we consider the bugfix branch to be stable to use ?*
The name `bugfix` implies it's stable, while the name `develop` does not.

Also, to answer @NezSez, @ManuelMcLure, @Vertabreak concerns, currently, there are so many fork because of the required customization of `Configuration.h` / `Configuration_adv.h`. 

If we removed those files from the main repository, and renamed them to `Configuration.h.example` for example, then people will not have to fork the main repository and they could pull directly to get the latest release without breaking their config. It would require adding a script that rename the example files to .h if the .h are missing when building so it does not break current code testing. 

Hopefully, that should reduce the number of useful forks to a lot less and make it easier to identify fork with features to test.

X-Ryl669, 2021-02-07:
@rhapsodyv Maybe we should set up some **RFT** branches (request for testing). I don't think it's possible for a developer to test all existing configuration (even building them is hard). 

So having a dedicated communication channel for such tests would be better to ensure we're not introducing breaking changes.

What I want to avoid is:
1. fork from outdated code, so that, when the RFT is done, the main "branch" (whatever it's called) has diverged so far, it's impossible to merge
2. discourage developers with RFT that never progress (so the RFT must have a deadline and must be merged, unless there are reported issues with it)
3. undiscoverable pending changes (people using the main "branch" must adhere to the *contract* that they should test some RFT if they want to profit from the latest changes)

This would save some time for Scott. Once the RFT are validated, he can merge them with more confidence. It'll be cleared what's hot, what's currently boiling and what's cold/frozen. So making releases will also be easier, as he can set up a kind of "merge window" after which no more RFT are accepted (a bit like linux), and only release when all the tagged RFT are done or abandoned.

rhapsodyv, 2021-02-07:
This is entirely with @thinkyhead. We already started some discussions about that on discord, but it didn't go further.
We even tried work more like a team, but it didn't go well, either...

NezSez, 2021-02-07:
I think that most of the "Forks" are only made because new users have read dozens of webpages, also made by new users, that say that is what you have to do. It's not often stated that you only need to do that if you plan to submit code in Pull Requests for inclusion in the mainline Marlin.

NezSez, 2021-02-07:
@rhapsodyv :
Any insights into why the team effort didn't pan out ?
Is it that there just aren't enough consistent developers participating to make it worth the effort/hassle?
Or more like there are enough, but the diverse and hectic schedules make any kind of reliable deadlines practically impossible?
I looked thru the "Insights" for a couple of days and there are some patterns that stand out.

qwewer0, 2021-02-07:
A side not to branches.
Is there any particular reason that [revert-20729-bugfix-2.0.x](https://github.com/MarlinFirmware/Marlin/tree/revert-20729-bugfix-2.0.x) branch isn't removed?

rhapsodyv, 2021-02-07:
> @rhapsodyv :
> Any insights into why the team effort didn't pan out ?
> Is it that there just aren't enough consistent developers participating to make it worth the effort/hassle?
> Or more like there are [...]

Yes, I surely have lots of insights about that. But one of the biggest obstacles are: management tasks. Someone with the “power of the pen 🖊” will need to do some level of management tasks, like project organization, people management, etc. Scott doesn’t like it. And we don’t have enough people with time that like that, and even some of the people that are on the team now are very recent to get full trust from Scott for such crucial task.
It’s hard to change the way of things works.... worsts when we are talking of people working for free...

NezSez, 2021-02-08:
@rhapsodyv : Coders wanna code, not manage or even document for that matter :)

@X-Ryl669 : In most of the projects I've been involved with neither bugfix nor dev were "stable", only "releases" were, which generally merged bugfix-2.0.N into a new "minor release" periodically, and you'd create a new bugfix-2.0.N+ for example. The .x in Marlin's bugfix-2.0.x sorta puzzled me the first time I saw it, but makes sense for a rolling release kinda scheme. Also, I didn't mean to hijack your thread! Just wanted to make sure your question got some attention.

@qwewer0 : I've wondered about that, and Marlin-RTOS which cites a "Marlin 2.1 Bugfix Branch" in the README.

Vertabreak, 2021-02-08:
There use to be other other branches they where phased out i assume it was to cumbersome to maintain extra branches.

BirkBinnard, 2021-02-08:
I've been dealing with tweaking and compiling Marlin for only about 2 months now, a but I think there might be a fairly simple resolution to the "stable" VS "bug fix" versions.

Most people perceive stable as being tested and ready for "production" use. Bug fix, on the other hand, is perceived as having updates that are not yet in the stable version, but that have been tested on an individual basis. Apparently though, the Marlin bug fix versions have code that is still preliminary and/or not fully tested, and therefore may cause compile failures that did not occur with working Configuration.h and Configuration_adv.h modules. (At least that has been my experience so far.)

My suggestion is to include with each bug fix release a list of the modules that have been changed from the stable version. This list would include the module name, a brief description of the change(s) that it contains, and some info about how they were tested. With that information someone could decide if they wanted to compile with all the updated modules, or maybe leave some out and use the corresponding modules from the stable release.

X-Ryl669, 2021-02-08:
Seems like you thought like me, the semantic of the branch name is confusing. I'm pretty sure it wouldn't have happened with a branch called "develop".

Your idea sounds good, but it does not apply 100% of the time since Marlin is not only made of modules but also on core feature. 
When improving them, we sometime break them. There are interaction between any of them, and it's not possible to have a true modular build without impacting each other. So it's not possible to say "I just want to test the shiny new module X, let's only fetch that".

BirkBinnard, 2021-02-08:
I totally agree that Marlin is a collection of features, targeted at specific hardware configurations. It was, in fact, 2 of these features (sensorless homing and probing and support for a color-mixing hotend)  that convinced me it was time to become involved. But as a software developer from a different timeI also knew that these features were implemented via code organized in modules. And I recently discovered that one of these modules, G28.cpp, was coded without any support for delta style printers, which is what I have. (At least this is true in the most current BugFix version - I don't know about earlier versions,) 

If I had known this I could have saved at least a week of time trying to figure out why my printer's hotend crashed into one of its towers every time I attempted a Home function. Needless to say all of my numerous parameter tweakings were in vain simply because the necessary delta code is just not there. (I don't mean to suggest coding for deltas is simple; I know it's not.) 

My basic point was simply an attempt to make Marlin less of a black box and more transparent to people who are trying to get it to work for their specific situation. For someone who has not seen this stuff before, the combined complexity (and length) of Marlin's configuration files, PlatformIO, VSCode, and the lack of detailed motherboard documentation presents a daunting challenge indeed.

Roxy-3D, 2021-02-11:
TLDR.

> There used to be a "dev" branch meant for new features and not bug fixes, but then users would clone that branch & sit on it thinking they were running the latest code. The dev branch would become outdated since patches & new features were still going into the bugfix branch.

Yes.   There have been a fair number of attempts to better organize and police the various trees.   The big problem is there are not enough development man hours to do everything that is desired.   We could cut the speed at which new features are developed and merged and use those resources to better manage the various trees.

But the problem is the developers want to develop.  And running both a development branch, bugfix branch has been tried in the past several times.   We simply do not have the resources to do that successfully.

With that said...   We have successfully run experimental branches and made eventually made those the main release branch.   The 32-bit work with the hierarchical file organization is an example of that.   

As for renaming the BugFix branch....    My vote is against it just from a historical perspective.   Marlin has (almost) always had a BugFix branch that reflects the most current state of the firmware.   While the name might not be entirely accurate, the user base knows that is where the most current code and feature set it.   My vote is we don't change it unless there is overwhelming consent from the entire user base.

X-Ryl669, 2021-02-11:
Ok, that makes sense. So, to sum up what was said above:

1. Having a separate *bugfix* and *develop* branch was already tried and it wasn't working that well
2. There is some issue with the insanely huge number of forks, and it can probably be solved by removing `Configuration.h` and `Configuration_adv.h` from the source tree and have a script creating them if missing.
3. Documentation of the existing features and their shortcomings is lacking => more or better documentation is required
4. Testing new features is not working that well because of the zillion of different configurations, yet developers don't have enough time to do "testing marathon, release candidate, *whatever* method", so we end up with many issues raised after some feature has landed

Did I miss anything ? @thinkyhead do you agree ?

X-Ryl669, 2021-02-25:
I'm closing this. 

(2) was rejected because it's easier for users to have their config file at a fixed place and have the developers just launch a git diff to see the changes.

For (4) and your reference, I've started computing the entropy of the possible configuration options of Marlin, and it's close to 2300 bits (see [here](https://github.com/X-Ryl669/MarlinForMKS/commit/7835558a9479319862b79ca59703a1838b83a997)). So it's not humanly possible to test all configurations. I don't see any solution to 4 except a kind of genetic/evolution process, like it's currently the case: let the most used configuration be tested and user reports issue. Less used configuration that breaks will not be reported. 

For (3), I'm still trying to improve the documentation wherever I can. I think we can ask developers of new feature to add documentation before merging them.

closed 2021-02-25 by X-Ryl669

thinkyhead, 2021-05-02:
Sorry, I did not see this issue until today. That is because the volume of PRs and general tasks has been so enormous that I seldom have a chance to visit the issue queue itself. Even if you @ mention me on some issue, there is hardly a point, because there is already so much noise here and my current notifications count from this repository is over 2500.

If you want to call my attention to important topics and propose some better tools for project management and triage, you will have to contact me directly on Discord.