MarlinFirmware/Marlin #21013: [FR] add clear notes about probe offsets and G35 - Tramming Assistant

BlackDragon2020, 2021-02-06:
Really simple, I was trying to use G35 - Tramming Assistant. I keep getting abort. I did some googling founds tons of people with same issues. Then I stumbled across a post talking about margins and disabling them. It got me thinking.

If your build plate is defined as (x,y) and your probe has negative offsets it can not reach some of the probe spots if you define them as at your nozzle like the docs say. 

You can only really get to ((x - x offset), (y - y offset))

A simple note about taking into account probe offsets would save many people lots of headaches.

Thanks for your time.

NezSez, 2021-02-06:
It is a bit confusing, and there is some interplay between the following settings, among others, and some question as to what gets applied exactly when. It is not so obvious how/when some of these interact.

Configuration.h :
#define X_MIN_POS (Y_MIN/MAX, Z_MIN/MAX, etc)
NOZZLE_TO_PROBE_OFFSET
PROBING_MARGIN
LEVEL_CORNERS_INSET_LFRB
MESH_INSET (et. al.)
LEVEL_BED_CORNERS
LEVEL_CORNERS_INSET_LFRB
LEVEL_CORNERS_USE_PROBE

Configuration_adv.h :
ASSISTED_TRAMMING
TRAMMING_POINT_XY
PROBING_MARGIN_LEFT (_RIGHT, _FRONT, _BACK)

There are a few more. There are some pages which help, but they become inaccurate/outdated pretty quick.
FTR they can use volunteers for such things: [MarlinDocumentation](https://github.com/MarlinFirmware/MarlinDocumentation)

ManuelMcLure, 2021-02-06:
I’m in the process of writing a document that tries to explain how to set up machine limits.  Right now it only covers `X_MIN_POS`/`Y_MIN_POS` but I want to cover many of the items you list above. The idea is to allow users to set just the settings that are absolutely necessary.
https://manuelmclure.github.io/ConfiguringLeveling.html

NezSez, 2021-02-06:
Excellent job, just skimmed it. Thanks for puttin' the time n' effort into it. Plan to submit it to the Documentation project?

ManuelMcLure, 2021-02-06:
When it's done. Writing the text is the easy part, making clear images to explain things is the hard part.

ManuelMcLure, 2021-02-06:
And explaining all of the corner cases, like printers where the end stops don't actually allow the nozzle to reach the whole bed surface.

NezSez, 2021-02-06:
https://3dprinting.stackexchange.com/questions/8153/how-to-set-z-probe-boundary-limits-in-firmware-when-using-automatic-bed-leveling

BlackDragon2020, 2021-02-07:
> I’m in the process of writing a document that tries to explain how to set up machine limits. Right now it only covers `X_MIN_POS`/`Y_MIN_POS` but I want to cover many of the items you list above. [...]

It's looking good. I did not know how to explain it, but it took me a minute to figure out it could not reach a spot so it aborted. Maybe it would help to send a reason for abort when doing gcode like this.

Good job thanks for the hard work. I suck at writing, math is my thing ;)

sjasonsmith, 2021-02-07:
When using the `bugfi-2.0.x` branch some checks are done at build time to make sure the provided points are reachable. As long as you have the correct NOZZLE_TO_PROBE offset in the header file that should identify a lot of this type of misconfiguration.

iStruggle, 2022-08-09:
Going to add on here instead of opening a new issue, despite the age of the post.

I still see current posts where people are struggling to get this working, so I will add my findings here in the hope it helps others following.

1) First and foremost: the coordinates in config_adv.h, for
`#define TRAMMING_POINT_XY`
**are the probe coordinates NOT the nozzle coordinates**.   
This may be obvious to some, but it wasn't to me.

2) When compiling,
`NOZZLE_TO_PROBE_OFFSET` and
`PROBING_MARGIN` are not cumulative.

For example, working with one axis only.
Probe Y offset: -5  
Probing Margin: 10
Bed size: 215 x 215
X & Y Max Position: Bed Size

If an Assisted Tramming coordinate is set to (x, 9) or (x, 206), it will fail as this would try to put the probe within 10mm of the edge of the bed, ie. within the MARGIN. (x, 10) and (x, 205) would compile fine. Y offset has no impact here.

If the Probe Y offset: -15 (so, further left of the nozzle), (x, 10) would still compile without error, however, (x, 205) would no longer compile.  For the probe to reach this point, the nozzle would have to be at 205 + 15 (offset) = 220.  As the Nozzle limit is 215, this will return out of bounds.

The same evidently applies for the x axis, and the reverse is true is the offset is positive, though in this case, X & Y Min Position (MIN_POS in config.h) is often negative to allow reaching endstops and this must be accounted for.  If MAX_POS is greater than bed size, this must also be considered in conjunction with Probe offsets.

In short, for ASSISTED_TRAMMING coordinates:

**For X and Y axes**
**_If your probe offset is negative_:**
Min = Probing Margin

Max = Lower of: 
Max_Pos + Probe Offset (remember this is negative so "+(-n)"); and
Bed Size - Probing Margin

**_If your probe offset is positive_:**
Min = Higher of: 
Probing Margin
Min_Pos + Probe Offset

Max = Bed size - Probing Margin

In shorter short:

1. Move the nozzle so that the probe is where you want it to... probe during assisted tramming.

2. Take the reported coordinate and add your probe offset to each axis value respectively (remember to "add" a negative offset: "+(-n)")  

3. This number must be greater than your probing margin and less than your bed size - probing margin.  In our above example, 10<n<205.

4. If the resultant value from 2. is not within the bounds in 3., the compile will fail.

Hope this helps someone else.