MarlinFirmware/Marlin #21017: Hotend and Bed turned off shortly after filament change with M600

abbrowna, 2021-02-07:
### Bug Description

After an M600 command, the printer successfully parks the print head, and unloads the filament. After changing it out and clicking resume, the printer does resume the print at the correct temperature as previous however after a short amount of time, say 1 minute or 2. It completely shuts down the hotend, heated bed and even the part cooling fan while still "printing". It will then keep trying to print until the hotend temperature falls below the threshold for cold extrusions at which point it will stop extruding but with the gantry still moving about. 

The printer is a Wanhao Duplicator 6 running Marlin bugfix 2.0.x at 1c19af2 (Downloaded on 6 Feb 2021)

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5938352/Configuration.zip)

If you've made any other modifications describe them in detail here.

My Wanhao d6 has a custom filament runout sensor located about 400mm from the direct drive print head and plugged into a spare connection on the main board.  I usually unload the filament all the way until it is out of the filament runout sensor during a change.

### Steps to Reproduce

1. Start print from SD card
2. Wait for filament change
3. Remove filament in the process triggering runout sensor
4. Load new filament in the process releasing runout sensor
5. Click on wheel to load filament
6. Click again to resume print after purge.
7. Print resumes then after a while bug appears. 

Gcode used to test the behavior is attached:
[Filament change test.zip](https://github.com/MarlinFirmware/Marlin/files/5938348/Filament.change.test.zip)

**Expected behavior:**

After the filament change, the print is expected to continue at the previous bed and hotend temperature until set otherwise by Gcode or LCD

**Actual behavior:**

The hotend, bed and part cooling fan turn off about a minute after the print resumes without being directly instructed to do so.

#### Additional Information

* [Issue video](https://drive.google.com/file/d/1liHQhAx50GHq3LXhLE50ha6vA2LE_WAj/view?usp=sharing).
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

sarvenn, 2021-02-16:
Change this parameter as you wish (configuration_adv under advanced_pause)
#define PAUSE_PARK_NOZZLE_TIMEOUT           90
It is exactly 90 seconds in your configuration. So it is quite normal that it will wait for 1.5 minute for your response. After that timeout period, it will cooldown and screen will prompt you to click and heat the nozzle.

abbrowna, 2021-02-16:
That is true @sarvenn. However that is not where I was experiencing the
issue. In the case where the hotend timed out after 90 seconds, I click the
button to heat the nozzle which it does and then click the button again to
resume the print. The printer will then start printing again but after
about 1 minute after the resume, It will shut down the hotend, bed and
cooling fan. You can refer to the issue video I posted along with the issue.

sarvenn, 2021-02-17:
Oh sorry. I would try to configure a new firmware and try it. It sounds like a serious bug.

abbrowna, 2021-02-18:
so I just configured the main marlin branch 2.0.x (2.0.7.2 ) for the
printer porting over my configurations to the older version with the help
of winmerge. The problem still occurs.

qwewer0, 2021-02-18:
@abbrowna Try the latest bugfix-2.0.x

abbrowna, 2021-02-18:
@qwewer0  Okay, so I've just flashed bugfix at a4c7386 and copied over the [same configuration](https://github.com/MarlinFirmware/Marlin/files/5938352/Configuration.zip) of the original issue report. Still the problem persists

thisiskeithb, 2021-04-07:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

abbrowna, 2021-04-07:
@thisiskeithb Okay so I have done this with bugfix at 5ab220f  used WinMerge to implement the same configuration I had and flashed the firmware. The problem still appeared. 

But! I decided to try something. Though throughout the tests I have been printing from SD card, I decided to unplug my raspberry pi (running Octoprint) from the printer USB and re-try. And Lo and and behold it worked. The filament change happened and the printer resumed and finished the part. 

Now in an attempt to try and salvage all the time I have wasted for you kind devs... I dug deeper: Turns out my Octoprint has a script that runs "After print job is Cancelled" that disables the fan, hotend and bed. so something is triggering the cancel event for Octoprint. I noticed once the M600 is hit. octoprint changes state from "printing from SD" to "Cancelling" . Kindly see [this issue](https://github.com/OctoPrint/OctoPrint/issues/3444) which bares some similarities.

I have attached the serial log. Notice how towards the end, there's a loop where Octoprint keeps on changing states each time executing the cancel script: then repeats. Which explains why I couldn't set the temp after the "bug" appeared.
[serial.log](https://github.com/MarlinFirmware/Marlin/files/6273522/serial.log)

So I know this is the not the correct place for Octoprint issues so if the experts feel convinced this is no longer a marlin issue, you can close and I will shift this bug over to Octoprint issues.

abbrowna, 2021-04-07:
Other strange behavior I noticed is that when I do eventually stop the print after the "bug" from the LCD, It does not park the print head as it usually does on any other ordinary abort.

closed 2021-06-18 by github-actions[bot]