MarlinFirmware/Marlin #21018: [FR] Return more decimal places on G30

soxhi8, 2021-02-07:
### Description

G30 currently (Marlin 2.0.7.2 2021-01-25) returns an accuracy of two decimal places like `Z: 0.00`. It is a very useful function to correct an incorrect level, without running a complete G29. It would be nice to get more decimal places, at least three like `Z: 0.000`, but I do not see why adding up to five would do any bad.

### Feature Workflow

G30 returns (up to) five decimal places like so `Z: 0.00000`, so the user can correct the bed level with the same accuracy as with G29, to achieve a good level faster than running G29 every time a single edge level changes.

#### Additional Information

I searched for similar requests, but was not able to find matching ones. If there already is one, please do not hesitate to point it out, so I can close this one.

qwewer0, 2021-02-07:
Increase `SERIAL_FLOAT_PRECISION`.

abdebadie, 2021-02-07:
https://youtube.com/channel/UCbeBouSPj6owK3OY-9G_5XA

soxhi8, 2021-02-07:
> 
> 
> Increase `SERIAL_FLOAT_PRECISION`.

Oh boi. Thank you so very much! I thought I looked everywhere, but could not find this option.

closed 2021-02-07 by soxhi8