MarlinFirmware/Marlin #21024: [BUG] (short description)

dekesone, 2021-02-08:
### Bug Description

Board = BigTreeTech SKR 1.4 Turbo with four BigTreeTech TMC2209 drivers used in UART mode. The UART jumper (MS3) is set and the machine works properly.

Any new driver current values set in the Trinamic config area of Configuration_adv.h in Marlin do not apply when the firmware is flashed to the board and a default value that is always reported is 800.

All drivers  = _TYPE  TMC2209, TMC_DEBUG is enabled,  MONITOR_DRIVER_STATUS is enabled and REPORT_CURRENT_CHANGE is enabled.

Sending M122 shows debug data for all steppers.  The stepper current can be successfully modified with M906 and it updates properly and reports back from M122. Editing the stepper current from the LCD also updates properly.  Only when changed in Marlin code that is doesn't update.

For example for y-axis, no matter what number is used for Y_CURRENT, the driver always default to 800.
 #if AXIS_IS_TMC(Y)
    **#define Y_CURRENT       200**
    #define Y_CURRENT_HOME  Y_CURRENT
    #define Y_MICROSTEPS     16
    #define Y_RSENSE          0.11
    #define Y_CHAIN_POS      -1
  #endif
M122 READ: 
                                                          X      Y
                                     Set current	800	800

### Configuration Files
CONFIGURATION_H_VERSION 020006
CONFIGURATION_ADV_H_VERSION 020006
**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5946662/Marlin.zip)

### Steps to Reproduce

Full re-compile of firmware while changing Trinamic settings in Configuration_adv.h @section tmc_smart
Uploaded to machine via USB and also via SD card
Reboot machine via M997 and also via power off hard boot

**Expected behavior:**

values changed for *_CURRENT expected to be reflected in driver debug data returned from M122 and also expected to be shown as default values in LCD under Configuration->Advanced Settings->TMC Drivers->Driver Current

**Actual behavior:**

Once values are changed in code, and new firmware is uploaded to machine, the changed values are not reported back from M122 (the values all show 800 regardless of what is in the code)
Driver Current values in the LCD menu also show 800 regardless of what is in the code

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

rhapsodyv, 2021-02-08:
Changing config values will change defaults, not the stored values on eeprom, until you tell marlin to reload defaults to eeprom. Try it.

dekesone, 2021-02-08:
> Changing config values will change defaults, not the stored values on eeprom, until you tell marlin to reload defaults to eeprom. Try it.

That's it!  Worked as expected.  Thank you for the response!
Is it considered best practice to send an M502 after every Marlin rebuild and upload?

Issue can be closed -

closed 2021-02-08 by dekesone