MarlinFirmware/Marlin #21025: [BUG] Concerning warning in latest bugfix compile for SKR v1.3 board

bill-orange, 2021-02-09:
### Bug Description

This is only a warning.  I felt that I should report it because it may reveal a deeper issue, possibly caused by an update to Platformio.  The following waring occurs on the latest Bugfix when compiled with the included config files.

```
Marlin\src\HAL\shared\Delay.cpp:54:23: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
   54 |     register uint32_t start = HW_REG(_DWT_CYCCNT) - ASM_CYCLES_PER_ITERATION, elapsed;
      |                       ^~~~~
Marlin\src\HAL\shared\Delay.cpp:54:79: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
   54 |     register uint32_t start = HW_REG(_DWT_CYCCNT) - ASM_CYCLES_PER_ITERATION, elapsed;
      |                                                                               ^~~~~~~
```

### Configuration Files

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5948072/Marlin.zip)

### Steps to Reproduce

1. Compile
2. Observe warning
3. Note, compile is otherwise successful

**Expected behavior:**

No scary warning

**Actual behavior:**

Concerning warning

#### Additional Information

The reference to `ISO C++17 does not allow 'register' storage class specifier` suggests to me that this may be the result of a change to the compiler.

thisiskeithb, 2021-02-09:
@sjasonsmith: What do you think? I tend to ignore warnings since none of my personal builds will build cleanly.

bugfixin, 2021-02-09:
The register keyword is explicitly meaningless according to http://www.open-std.org/jtc1/sc22/wg21/docs/papers/2015/n4340 so I think it should be (and is safe to be) removed here

sjasonsmith, 2021-02-09:
@thisiskeithb I prefer to enable all warnings and treat them as errors. Marlin’s not there, but that’s what I expect of code I maintain professionally.

sjasonsmith, 2021-02-09:
Most Marlin warnings come from external libraries. We should at least try to have the Marlin code itself warning free.

ellensp, 2021-02-09:
FYI MEATPACK also uses register and produces warning on LPC176x

bill-orange, 2021-02-09:
I am also getting a new warning on a Melzi build involving how variables are declared.  I will post it as a separate issue later.

rhapsodyv, 2021-02-26:
I think this was just fixed with #21178. Bugfix should be ok now.

nanotuxi, 2021-02-26:
> I think this was just fixed with #21178. Bugfix should be ok now.

Can confirm that.

thisiskeithb, 2021-02-26:
Fix has been merged. Closing.

closed 2021-02-26 by thisiskeithb