MarlinFirmware/Marlin #21027: [BUG] Marlin 2.0.8 Delta BTT SKR 1.4 Homing problem ( Error:Printer halted. kill() called!)

jweikamp, 2021-02-09:
Setup:

FLSun Delta
BTT SKR 1.4

Problem: 

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/5950536/Configuration.txt)

After homing following message is showing:
Error:Printer halted. kill() called!

BirkBinnard, 2021-02-09:
Did you specify sensorless homing in Configuration_adv.h?

When you started the Home function, where were the sliders on the X< Y, and Z columns?

When you started the Home function, did only one slider move at a time?  Or did all 3 move at one time?

See my version of a similar problem here: https://github.com/MarlinFirmware/Marlin/issues/20986

jweikamp, 2021-02-10:
No, i haven't specified sensorless homing. Today my new drivers arrive (TMC2130's). I will try again with these and let you know the results.

ellensp, 2021-02-11:
This is not likely to be a firmware issues, but a configuration issue of some sort.

I recommend joining the discord server as listed below.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-02-11 by ellensp