MarlinFirmware/Marlin #21028: [BUG] (SKR 1.3 turbo with TMC2130 V.1 results in a "bad response")

Hannes0O0, 2021-02-09:
### Bug Description

I tried to use my new SKR 1.3 turbo mainboard with the TMC2130 drivers. The problem is, that they just dont work. I set the jumpers under the drivers correctly to work in spi mode. I also eneabled tmc debug. If I use that, it says at the end "bad response". I also tried to eneable just one motor driver, wich also didn't work. with another non spi driver it worked fine (ofcourse with another jumper and sofware configuration).
### Configuration Files

I changed the processor in the platformio.ini file to LPC1768 and ofcourse the config files.

### Steps to Reproduce

1. Use a SKR 1.3 turbo mainboard with KOOKYE TMC2130 V 1.0 drivers, with the jumpers on the mainboard covering all the red pins for the spi connection.
2. Use my Marlin config.
3. Compile it and put the .bin file a sd, wich should be included with the mainboard.
4. Reset the board.
5. Try to move a stepper.

**Expected behavior:**

The motor should move.

**Actual behavior:**

Nothing. the motor dont moves

#### Additional Information

The Software is compiled in a vm running kali, copied over to windows and from there put on the sd.

Sorry, if my english isn't to good. I hope it will get better.

[config.zip](https://github.com/MarlinFirmware/Marlin/files/5951214/config.zip)

ellensp, 2021-02-09:
Your config says

#define X_DRIVER_TYPE  A4988              <--- wrong
//#define Y_DRIVER_TYPE  TMC2130      <--- wrong, commented out defaults to A4988
//#define Z_DRIVER_TYPE  TMC2130      <--- wrong, commented out defaults to A4988
//#define Z2_DRIVER_TYPE TMC2130     <--- wrong, commented out defaults to A4988
//#define E0_DRIVER_TYPE TMC2130     <--- wrong, commented out defaults to A4988

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-02-09 by ellensp

Hannes0O0, 2021-02-09:
Sorry, but I took the wrong config files. That one was only for testing. I commented the other ones out because I wanted to see if it works with a normal driver and it did. Before it was looking like this:

#define X_DRIVER_TYPE TMC2130
#define Y_DRIVER_TYPE TMC2130
#define Z_DRIVER_TYPE TMC2130
#define Z2_DRIVER_TYPE TMC2130
#define E0_DRIVER_TYPE TMC2130

I also spent hours of searching and found + tried a lot of things. And as it turned out nothing worked. The reason I decided to make this bug report is because there only are a few things left wich could cause that error.

1. A fail of the drivers. (pretty unrealistic because it would be strange if all of them would fail.)
2. A fail of the mainboard. (it should work because it also worked with the normal drivers. The only thing wich eventually couldnt work is the connection (wich I checked as good as possible) or something in the processor. thats also unrealistic because thats the second one)
3. Marlin. Maybe the wrong configuration or something with the software it self. The reason I think so, is because that are pretty rare drivers (I mean a rare manufacturer from the drivers). Even the website where I bought it is gone. The last site on that I found it, is a french one.

Hannes0O0, 2021-02-09:
> Your config says
> 
> #define X_DRIVER_TYPE A4988 <--- wrong type?
> //#define Y_DRIVER_TYPE TMC2130 <--- wrong, commented out defaults to A4988
> //#define Z_DRIVER_TYPE TMC2130 <--- wrong, commented out defaults to A4988
> //#define Z2_DRIVER_TYPE TMC2130 <--- wrong, commented out defaults to A4988
> //#define E0_DRIVER_TYPE TMC2130 <--- wrong, commented out defaults to A4988
> 
> This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:
> 
> * MarlinFW.org [Marlin Documentation](http://marlinfw.org)
> * RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
> * [Tom's 3D Forums](https://discuss.toms3d.org/)
> * Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
> * Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
> * [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
> * Marlin Discord server. Join link: https://discord.gg/n5NJ59y
> 
> After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

Sorry, but I took the wrong config files. That one was only for testing. I commented the other ones out because I wanted to see if it works with a normal driver and it did. Before it was looking like this:

#define X_DRIVER_TYPE TMC2130
#define Y_DRIVER_TYPE TMC2130
#define Z_DRIVER_TYPE TMC2130
#define Z2_DRIVER_TYPE TMC2130
#define E0_DRIVER_TYPE TMC2130

I also spent hours of searching and found + tried a lot of things. And as it turned out nothing worked. The reason I decided to make this bug report is because there only are a few things left wich could cause that error.

A fail of the drivers. (pretty unrealistic because it would be strange if all of them would fail.)
A fail of the mainboard. (it should work because it also worked with the normal drivers. The only thing wich eventually couldnt work is the connection (wich I checked as good as possible) or something in the processor. thats also unrealistic because thats the second one)
Marlin. Maybe the wrong configuration or something with the software it self. The reason I think so, is because that are pretty rare drivers (I mean a rare manufacturer from the drivers). Even the website where I bought it is gone. The last site on that I found it, is a french one.

ellensp, 2021-02-09:
The stepper drivers are just mks tmc2130's  KOOKYE  is just a reseller.
https://github.com/makerbase-mks/MKS-StepStick-Driver/tree/master/MKS%20TMC2130/MKS%20TMC2130%20V1.0_001

ellensp, 2021-02-10:
Reading through the circuit diagram there is a PIN 10 on the tmc2130 chip. "SPI_MODE" this sets standalone mode or SPI mode.

From the datasheet: 
"When tied low, the chip is in standalone mode and pins have their CFG functions.
 When tied high, the SPI interface is available for control.

The circuit diagram shows these stepper driver are always in standalone mode.
Pin 10 is connected via a r3 (zero ohm resistor)  to GND

You would have to remove R3 to enable SPI mode.

ellensp, 2021-02-10:
The images on one amazon advert for these shows the R3 missing... check yours

Hannes0O0, 2021-02-10:
On my one at R3 there wasn't a connection / a resistor. 
After a bit of measuring i found out that as you said, one pad goes to GND. But the other one didn't have any connection to VDD. As far as I know there should be a pullup resistor but it wasnt connected to any resistor near. Thats strange.

The french amazon site where I found it is there:
https://www.amazon.fr/KOOKYE-TMC2130-Stepper-Printer-Control/dp/B07Y16BWXF
That site says that it should be already in spi mode.

ellensp, 2021-02-10:
Enable TMC_USE_SW_SPI in Configuration_adv.h

Hannes0O0, 2021-02-10:
That did it! The motor dont moves and the driver got really hot, the processor also warmed up. After removing the driver it seems to work just fine. I dont know how that could happen. I placed it correctly, set the jumpers right. I think that  was it for the drivers.

I will proberly get a whole new set with the drivers included.

Finally: I dont think it happen because of that setting. To be exact thats the second time that happend.