MarlinFirmware/Marlin #21030: [BUG] preheat for probing/levelling ignore pending heating operation

t3chguy, 2021-02-09:
### Bug Description

When using `PREHEAT_BEFORE_LEVELING` with `LEVELING_NOZZLE_TEMP` I observed Marlin overwriting my nozzle temperature with a lower one.

My start gcode has my nozzle preheat to 175 whilst heat soaking the bed for levelling. After homing, when levelling starts the nozzle temperature is reset to 120 which is my `LEVELING_NOZZLE_TEMP`.

It looks like the code isn't taking into account the temperature target and instead only cares about the current active temperature wrongly.

### Configuration Files

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/5953092/Archive.zip)

### Steps to Reproduce

1. Enable `PREHEAT_BEFORE_LEVELING`
1. Set `LEVELING_NOZZLE_TEMP` to 120
1. Run `M104 S175`
1. Run `G28`
1. Run `G29`

**Expected behavior:**

Temperature keeps rising to 175 as per the instruction.

**Actual behavior:**

Assuming temperature at time of `G29` is <120:
Temperature target is changed to 120.

ellensp, 2021-02-11:
I think your miss interpreting things
M104 S175, this sets hotend temp to 175, correct
but
G29 with LEVELING_NOZZLE_TEMP resets the hotend temperature to 120, as it is meant to do.

t3chguy, 2021-02-11:
But then the code is wrong for that too. As if the temperature was already 175 then it would not set it to 120. It only sets it to 120 if its currently less than 120, ignoring the pending heat up to 175.

thisiskeithb, 2021-02-11:
Perhaps `LEVELING_NOZZLE_TEMP` should have been named `LEVELING_NOZZLE_MIN_TEMP`.

t3chguy, 2021-02-11:
Sure, but it's implied by being a sub-option of PREHEAT_BEFORE_LEVELING. It's not SET_TEMPS_BEFORE_LEVELING

closed 2021-02-12 by ellensp