MarlinFirmware/Marlin #21036: [BUG] SHOW_REMAINING_TIME shows undefined value at boot instead of zero

sfjuocekr, 2021-02-10:
With SHOW_REMAINING_TIME enabled on a fresh boot the display reports an undefined value instead of zero if you have USE_M73_REMAINING_TIME enabled as well.

get_remaining_time is returning the value of _calculated_remaining_time, which gets its value from print_job_timer.duration() which is not running at that moment so it "should" return zero?

ellensp, 2021-02-10:
```
Please follow the instructions below. Failure to do so may result in your issue being closed.

### Before Reporting a Bug

1. Test with the `bugfix-2.0.x` branch to see whether the issue still exists.

2. Get troubleshooting help from the Marlin community to confirm it's a bug and not just a configuration error. Links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

### Instructions

1. Fill out every section of the template below.

2. Always attach configuration files, regardless of whether you think they are involved.

3. Read and understand Marlin's Code of Conduct. By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

<THE TEMPLATE>

### Bug Description

<!-- Describe the bug in this section. (You can remove this invisible comment.) -->

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

<!-- Describe the steps needed to reproduce the issue. (You can remove this invisible comment.) -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

<!-- Describe what you expected to happen here. (You can remove this invisible comment.) -->

**Actual behavior:**

<!-- Describe what actually happens here. (You can remove this invisible comment.) -->

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

sfjuocekr, 2021-02-15:
### Bug Description

With SHOW_REMAINING_TIME enabled on a fresh boot the display reports an undefined value instead of zero if you have USE_M73_REMAINING_TIME enabled as well.

### Configuration Files

[config.zip](https://github.com/MarlinFirmware/Marlin/files/5979126/config.zip)

### Steps to Reproduce

Enable SHOW_REMAINING_TIME and USE_M73_REMAINING_TIME.

**Expected behavior:**

Display to show "00:00" after startup.

**Actual behavior:**

Display shows: "4xxxx days remaining"

#### Additional Information

I have tested this against the latest changes:

"Your branch is up to date with 'origin/bugfix-2.0.x'."

ellensp, 2021-02-15:
Looks to be a division by zero error.. but im not sure how to test it beyond "Display to show "00:00" after startup." which it now  does.

in marlinui.h line 251 is

`return elapsed.value * (100 * (PROGRESS_SCALE) - progress) / progress;`

change it to

`return progress>0 ? elapsed.value * (100 * (PROGRESS_SCALE) - progress) / progress : 0;`

ellensp, 2021-02-15:
I have done more testing now with printing from sd card, This fix makes remaining time work as expected. 
I have Created a PR to add this fix into Marlin.

ellensp, 2021-02-15:
This is now fixed it latest bugfix

closed 2021-02-15 by ellensp

sfjuocekr, 2021-02-15:
Ah ok, that is how the ternary operator works here.

I tried to rewrite it to:

return progress == 0 ? 0 : elapsed.value * (100 * (PROGRESS_SCALE) - progress) / progress;

But then it would always return zero and wouldn't update when M73 was received.

I'll test this later today.