MarlinFirmware/Marlin #21045: [BUG] Reporting Bed Topography cause printer reset

jwchen119, 2021-02-10:
### Bug Description

Device: Delta printer
MB: Bigtreetech Skr v1.3
Driver: TMC2208x4

The bug is reporting Bed Topography cause printer reset.
This is odd because I can't perform any G29 command with T
I found this bug when I want to calibrate my bed manually by perform G29 P2 B T. When it go down the bed and shows the point 1 on Pronterface, my printer reset without any verbose.

### Configuration Files

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5962374/Marlin.zip)

### Steps to Reproduce

Method A.
  1. Press button on LCD
  2. Press Motion
  3. Press Unified Bed Leveling
  4. Press Output Mesh Map
  5. Press Output for Host
  6. Printer reset after step 5

Method B.
  1. Using Pronterface to connect to printer.
  2. Send command "G29 T"
  3. Printer reset after step 5

Method C.
  Any G29 command combined with T

**Expected behavior:**

My printer should not reset.

**Actual behavior:**

My printer reset.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

Roxy-3D, 2021-02-11:
Attaching Configuration.h files would help.    But this kind of 'feels' like a previous GCC optimization problem where the stack space  was being used to freely by the compiler.   (More details can be provided if the topic is interesting to anybody.)

With that said, the 'T' parameter is so simple I kind of doubt that would blow things up if the other options are working properly.

A very simple way to disprove the GCC optimization theory would be to move the 'T' option into its own .CPP file and see if it keeps misbehaving.

ManuelMcLure, 2021-02-11:
Could this be a timing thing with a too long loop causing the watchdog to trigger?

jwchen119, 2021-02-11:
> Attaching Configuration.h files would help. But this kind of 'feels' like a previous GCC optimization problem where the stack space was being used to freely by the compiler. (More details can be [...]

I've update the txt file while the Marlin.zip contained Configuration.h.

GMagician, 2021-02-11:
I had this same issue, in the same place, but that has been fixed by #20985, sure your Marlin is up to date?

jwchen119, 2021-02-11:
> I had this same issue, in the same place, but that has been fixed by #20985, sure your Marlin is up to date?

Oh, I thought 2021-1-31 should be fresh enough.
Will download the latest one and report if it fixed.

jwchen119, 2021-02-11:
OK, looks like https://github.com/MarlinFirmware/Marlin/pull/20985 help this issue out.
Case close now, thanks for helping!

closed 2021-02-11 by jwchen119