MarlinFirmware/Marlin #21046: [BUG] PSU PIN won't turn on automatically (SKR PRO)

Fighter777, 2021-02-11:
Hello after a new marlin bugfix download, I did a basic configuration to run a BTT 1.2 relay and the PSU pin refuses to go high on startup.

If I change the printer power state from the display, the PSU PIN goes high.

moreover, if I activate the AUTO_POWER_CONTROL option, I lose the display on the screen

The configuration was working on a marlin release from June 2020

Step to reproduce :
1) start motherboard
1) select turn off printer on screen (psu pin dont change)
2) select turn on printer on screen (psu pin change to HIGH state)

Hardware:
BigTreeTech SKR pro 1.1 (STM32f4)
CR10 screen (reprap configuration) 

CONFIG :
Configuration.h :
#define MOTHERBOARD BOARD_BTT_SKR_PRO_V1_1
#define SERIAL_PORT -1
#define SERIAL_PORT_2 1
#define PSU_CONTROL
#define PSU_NAME "Power Supply"
#define PSU_ACTIVE_STATE HIGH      // Set 'LOW' for ATX, 'HIGH' for X-Box

#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
#define Z2_DRIVER_TYPE TMC2209
#define E0_DRIVER_TYPE LV8729
#define LCD_LANGUAGE fr
#define SDSUPPORT
#define SD_CHECK_AND_RETRY
#define ENCODER_PULSES_PER_STEP 4
#define ENCODER_STEPS_PER_MENU_ITEM 1
#define SPEAKER
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

pins_BTT_SKR_PRO_common.h:
// #define TEMP_2_PIN                          PF6   // T3 <-> E2
#define PS_ON_PIN                           PF6 

Regards

Fighter777, 2021-02-24:
Hello, I found the offending commit: https://github.com/MarlinFirmware/Marlin/commit/b9ed139546d4d35cf925926e93f76d80245db7dc

after this commit, the pin no longer turns ON

FanDjango, 2021-02-24:
Same for me. I confirm PSU not being controlled. Also BTT SKR pro 1.1.

Fighter777, 2021-02-24:
@FanDjango reverses the changes in the commit to make it work again

FanDjango, 2021-02-24:
Yes, I am doing that right now. I was searching through BLAME on PSU_ON in MarlinCore.cpp and immediately found your issue and the mention in the Jan 25 PR. Thanks for saving me lots of time 👍

FanDjango, 2021-02-24:
@Fighter777 Instead of a revert, here is the better fix:

```
  TERN_(DYNAMIC_VECTORTABLE, hook_cpu_exceptions()); // If supported, install Marlin exception handlers at runtime

  SETUP_RUN(HAL_init());

  #if EITHER(DISABLE_DEBUG, DISABLE_JTAG)
    // Disable any hardware debug to free up pins for IO
    #if ENABLED(DISABLE_DEBUG) && defined(JTAGSWD_DISABLE)
      JTAGSWD_DISABLE();
    #elif defined(JTAG_DISABLE)
      JTAG_DISABLE();
    #else
      #error "DISABLE_(DEBUG|JTAG) is not supported for the selected MCU/Board."
    #endif
  #endif
```

Move that block IN FRONT of the KILL/SUICIDE/PSU setup. Our board needs that.

The original PR was to have those pins setup BEFORE the rather lengthy WiFi setup took place. I cannot understand why it was moved up so far that HAL init was bypassed, for example.

Fighter777, 2021-02-24:
yes, i don't know too ....

Evg33, 2021-02-25:
SKR14T starting with blank screen and no communication via USB-Serial,

thinkyhead, 2021-02-27:
If the applied patch fixes this issue then it can be closed!

Fighter777, 2021-02-27:
hello, i will test

Fighter777, 2021-02-27:
work for me

closed 2021-02-27 by Fighter777