MarlinFirmware/Marlin #21050: Controller/Extruder fan not working on Creality CR-10S with Marlin 2.0.7 bugfix

McClinten, 2021-02-11:
I have the Creality CR-10S with V2.1 board (which have a single controllable fan header)

I have setup a basic configuration from Configurations-bugfix-2.0.x\config\examples\Creality\CR-10S\CrealityV1

Compile and upload to printer. When printing from Cura, extruder fan is not working. Looking in the printers menu I can find "Fan speed 1" and "Fan speed 2". "Fan speed 2" reflects what Cura sends to control the extruder fan. If I change "Fan speed 1" in the printers menu the fan is controlled.

Setting #define NUM_M106_FANS 1 in configuration.h does not help. "Fan speed 2" is gone in the menu but Cura cannot control fan. "Fan speed 1" can still be controlled through the printers menu.

ellensp, 2021-02-12:
I compiled your Configuration on current bugfix on a ramps. I did a M43. Here are the fan pins.

```
PIN:   7   Port: H4        FAN1_PIN                               protected 
                           MOSFET_D_PIN                           protected 
PIN:   9   Port: H6        FAN_PIN                                protected 
                           RAMPS_D9_PIN                           protected 
```
FAN_PIN is on D9 works as expected  (Tested from LCD Fan speed 1) This is part cooling fan
FAN1_PIN is on D7, also works as expected.   (Tested from LCD Fan speed 2) This is what ever you want it to be Controlled via gcode M106 P1 S{0-255}

What I can't confirm is that it is actually D7, as I can find no circuit diagrams for the creality ramps, and I don't have one.(just  using a real ramps 1.4 for testing)   

Also If you have FAN1_PIN for the extruder cooling fins you should not be trying to control it.
Instead enable #define E0_AUTO_FAN_PIN FAN1_PIN in Configuration_adv.h

McClinten, 2021-02-12:
Embarrassing, I have redone my settings in Cura in the same time as I upgraded to Marlin 2, and in the machine settings -> Extruder 1 there is an attribute named "Cooling Fan number", I misinterpreted than setting and set it to 1 (as in one fan). It is the index number of the fan to be used with extruder 1

closed 2021-02-12 by McClinten