MarlinFirmware/Marlin #21060: [BUG] Issueing G29 T causes exception and restart

kpishere, 2021-02-12:
### Bug Description

Issuing a G29T command to show a report of probed points causes exception and board restarts.

However, issuing G29T1 is OK and produces the following :  

>>> g29t1
SENDING:G29T1
Bed Topography Report for CSV:
NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN
NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN
NAN	NAN	2.905	2.940	2.975	3.025	3.055	3.082	NAN	NAN
NAN	NAN	2.920	2.970	3.017	3.067	3.095	3.125	NAN	NAN
NAN	NAN	2.940	2.982	3.037	3.100	3.137	3.177	NAN	NAN
NAN	NAN	2.952	2.995	3.060	3.115	3.167	3.217	NAN	NAN
NAN	NAN	2.962	3.027	3.077	3.132	3.187	3.235	NAN	NAN
NAN	NAN	2.975	3.035	3.095	3.147	3.190	3.250	NAN	NAN
NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN
NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN

### Configuration Files

This repo contains configuration used (config files also attached).

https://github.com/kpishere/Marlin/tree/itopie400+
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/5969725/Archive.zip)

### Steps to Reproduce

M502          ; Reset settings to configuration defaults...
M500          ; ...and Save to EEPROM. Use this on a new install.
M501          ; Read back in the saved EEPROM.

M190 S65      ; Not required, but having the printer at temperature helps accuracy
M104 S210     ; Not required, but having the printer at temperature helps accuracy

G28           ; Home XYZ.
G29 P1        ; Do automated probing of the bed.
G29 P2 B T    ; Manual probing of locations. USUALLY NOT NEEDED!
G29 P3 T      ; Repeat until all mesh points are filled in.

**G29 T         ; View the Z compensation values.**

**Expected behavior:**

To output something and not crash.

**Actual behavior:**

It crashes and I must re-do procedure.

#### Additional Information

My bed probing size is restricted in order to avoid points with screw holes in corners.  It had happened that probing got stuck on one of those screws and would not continue.

ellensp, 2021-02-12:
Make sure you code has https://github.com/MarlinFirmware/Marlin/pull/20985

jwchen119, 2021-02-12:
You should update your Marlin, check this out https://github.com/MarlinFirmware/Marlin/issues/21045

kpishere, 2021-02-12:
@ellensp Yes, I'm on a fork of the following branch when this issue is reported.  I don't think it is a streaming issue.  Likely due to null values for points?

commit 92b4c050906dbde4cca6d811e5f4804fb11d1438 (origin/bugfix-2.0.x, kpishere/Marlin/bugfix-2.0.x)
Author: thinkyhead <thinkyhead@users.noreply.github.com>
Date:   Sun Jan 31 00:16:38 2021 +0000

    [cron] Bump distribution date (2021-01-31)

tablatronix, 2021-02-14:
hmm I just flashed today, and this gave me abl probing 1 single position over and over, and beeping after each one. No x.y movement.

thinkyhead, 2021-02-14:
> this gave me abl probing 1 single position over and over

That is a different bug in a different feature, so you should not hijack this issue but post a new bug report.

thinkyhead, 2021-02-15:
@kpishere — I'm unable to reproduce the issue using your configuration (with a RAMPS rather than a SmartRAMPS board). All my tests of `G29 T1` have reliably printed the mesh.

If you would like to narrow down the issue, first make sure to download the latest Marlin code. Then disable every optional feature in your configuration except UBL and the LCD. If the problem goes away then it might indicate a conflict with another one of your enabled features.

kpishere, 2021-02-15:
Yes, T1 works, it is T0 that fails. The output version that is supposed to be more human readable.  I'd shown T1 output for possible more insight into values behind the T0 crash ... Nan values.

tablatronix, 2021-02-15:
I accidently did not flash bugfix2.x I will test again, and see what mine does.

thinkyhead, 2021-03-03:
Still reliably failing with `bugfix-2.0.x`?

kpishere, 2021-03-04:
Hi, for myself, I've avoided this issue by switching two three point levelling.  However, I can give it a try tomorrow and will report back regardless.

kpishere, 2021-03-05:
@thinkyhead I've rebased on latest bugfix-2.0.x, noted that G29 command had some changes.  Also, due to increase in program size, I've had to reduce my mesh from the default 10x10 to 5x5.   The command appears to be working now.  The bug can be closed.

I think the test is still fair as you'll note below there is still an area on the print bed that the probe can't get to and hence there are some null (NAN) values -- this had changed because I've changed my hot end mount since the bug was reported.

This is the commit i've tested against [refcommit](https://github.com/kpishere/Marlin/commit/a8fa6dd107b215e2aa08aa800cff4ae1fc546d32)

>>> G29T1
SENDING:G29T1
Bed Topography Report for CSV:
0.753	0.855	0.795	0.652	NAN
0.843	0.935	0.878	0.738	NAN
0.847	0.925	0.895	0.785	NAN
0.750	0.857	0.870	0.775	NAN
0.663	0.782	0.788	0.730	NAN
>>> G29T
SENDING:G29T
Bed Topography Report:
    ( 10,225)                      (222,225)
        0       1       2       3       4
 4 | +0.753  +0.855  +0.795  +0.652    .
   |
 3 | +0.843  +0.935  +0.878  +0.738    .
   |
 2 |[+0.847] +0.925  +0.895  +0.785    .
   |
 1 | +0.750  +0.857  +0.870  +0.775    .
   |
 0 | +0.663  +0.782  +0.788  +0.730    .
        0       1       2       3       4
    ( 10, 10)                      (222, 10)

closed 2021-03-05 by kpishere