MarlinFirmware/Marlin #21063: [BUG] conversion from 'uint8_t'

JohnnyTheOne, 2021-02-12:
### Bug Description

Does anyone know why i get this error when i want to activate mmu2?
I suspect that # 20985 has to do with this change?

### Configuration Files
[Configuration.h](https://github.com/JohnnyTheOne/MarlinJTO/blob/JTO-CoreXY/Marlin/Configuration.h)
[Configuration_adv.h](https://github.com/JohnnyTheOne/MarlinJTO/blob/JTO-CoreXY/Marlin/Configuration_adv.h)

### Steps to Reproduce

1. [First Step] activate mmu2

**Expected behavior:**

**Actual behavior:**

`
Marlin\src\feature\mmu\mmu2.cpp: In static member function 'static void MMU2::mmu_loop()':
Marlin\src\feature\mmu\mmu2.cpp:291:36: error: conversion from 'uint8_t' {aka 'unsigned char'} to 'EnsureDouble' is ambiguous
  291 |           DEBUG_ECHO_F(cmd_arg, DEC);
      |                                    ^
In file included from Marlin\src\feature\mmu\../../inc/../HAL/./STM32/../../core/serial_hook.h:25,
                 from Marlin\src\feature\mmu\../../inc/../HAL/./STM32/MarlinSerial.h:27,
                 from Marlin\src\feature\mmu\../../inc/../HAL/./STM32/HAL.h:34,
                 from Marlin\src\feature\mmu\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\feature\mmu\../../inc/MarlinConfig.h:31,
                 from Marlin\src\feature\mmu\mmu2.cpp:23:
Marlin\src\feature\mmu\../../inc/../HAL/./STM32/../../core/serial_base.h:54:16: note: candidate: 'EnsureDouble::EnsureDouble(float)'
   54 |   FORCE_INLINE EnsureDouble(float a) : a(a) {}
      |                ^~~~~~~~~~~~
Marlin\src\feature\mmu\../../inc/../HAL/./STM32/../../core/serial_base.h:53:16: note: candidate: 'EnsureDouble::EnsureDouble(double)'
   53 |   FORCE_INLINE EnsureDouble(double a) : a(a) {}
      |                ^~~~~~~~~~~~
In file included from Marlin\src\feature\mmu\../../inc/MarlinConfig.h:55,
                 from Marlin\src\feature\mmu\mmu2.cpp:23:
Marlin\src\feature\mmu\../../inc/../core/serial.h:107:40: note:   initializing argument 1 of 'void SERIAL_ECHO_F(EnsureDouble, int)'
  107 | inline void SERIAL_ECHO_F(EnsureDouble x, int digit = 2) { SERIAL_IMPL.print(x, digit); }
      |                           ~~~~~~~~~~~~~^
Compiling .pio\build\BIGTREE_GTR_V1_0\src\src\gcode\bedlevel\M420.cpp.o
*** [.pio\build\BIGTREE_GTR_V1_0\src\src\feature\mmu\mmu2.cpp.o] Error 1`

GMagician, 2021-02-12:
I think issue is on SERIAL_ECHO_F using an integer variable.
Coming from code:

>   // If the compiler breaks on ambiguity here, it's likely because you're calling print(X, base) with X not a double or a float, and a
  // base that's not one of PrintBase's value. This exact code is made to detect such error, you NEED to set a base explicitely like this:
  // SERIAL_PRINT(v, PrintBase::Hex)

GMagician, 2021-02-12:
Try to replace in mmu2.cpp line 291
`DEBUG_ECHO_F(cmd_arg, DEC);`
with
`DEBUG_ECHO(cmd_arg, DEC);`

JohnnyTheOne, 2021-02-13:
replace with `DEBUG_ECHO(cmd_arg, DEC);` get a new error

`Marlin\src\feature\mmu\mmu2.cpp: In static member function 'static void MMU2::mmu_loop()':
Marlin\src\feature\mmu\mmu2.cpp:291:34: error: no matching function for call to 'SERIAL_ECHO(uint8_t&, int)'
  291 |           DEBUG_ECHO(cmd_arg, DEC);
      |                                  ^
In file included from Marlin\src\feature\mmu\../../inc/MarlinConfig.h:55,
                 from Marlin\src\feature\mmu\mmu2.cpp:23:
Marlin\src\feature\mmu\../../inc/../core/serial.h:99:6: note: candidate: 'template<class T> void SERIAL_ECHO(T)'
   99 | void SERIAL_ECHO(T x) { SERIAL_IMPL.print(x); }
      |      ^~~~~~~~~~~
Marlin\src\feature\mmu\../../inc/../core/serial.h:99:6: note:   template argument deduction/substitution failed:
Marlin\src\feature\mmu\mmu2.cpp:291:34: note:   candidate expects 1 argument, 2 provided
  291 |           DEBUG_ECHO(cmd_arg, DEC);
      |                                  ^
In file included from Marlin\src\feature\mmu\../../inc/MarlinConfig.h:55,
                 from Marlin\src\feature\mmu\mmu2.cpp:23:
Marlin\src\feature\mmu\../../inc/../core/serial.h:103:13: note: candidate: 'void SERIAL_ECHO(serial_char_t)'
  103 | inline void SERIAL_ECHO(serial_char_t x) { SERIAL_IMPL.write(x.c); }
      |             ^~~~~~~~~~~
Marlin\src\feature\mmu\../../inc/../core/serial.h:103:13: note:   candidate expects 1 argument, 2 provided
*** [.pio\build\BIGTREE_GTR_V1_0\src\src\feature\mmu\mmu2.cpp.o] Error 1`

if i disable #define MMU2_DEBUG it works fine,
thanks

ellensp, 2021-02-13:
The PR says replace  DEBUG_ECHO_F(cmd_arg, DEC); with DEBUG_ECHO(cmd_arg);
try that, I think the description above was a little confused

JohnnyTheOne, 2021-02-13:
`DEBUG_ECHO(cmd_arg);` works thanks

GMagician, 2021-02-14:
> The PR says replace DEBUG_ECHO_F(cmd_arg, DEC); with DEBUG_ECHO(cmd_arg);
> try that, I think the description above was a little confused

my fault, sorry

ellensp, 2021-02-15:
This is now in the latest bugfix.

PR #21065 (2021-02-15): Fix MMU2 compile error
Should fix #21063
closed 2021-02-15 by ellensp