MarlinFirmware/Marlin #21064: [BUG] Homing to Z-MAX is broken

kpishere, 2021-02-12:
### Bug Description

As of the following commit (below) on bugfix-2.0.x, G28 is broken for the case where one has two lead screws on one driver for z-axis and wants to home to Z-MAX with TMC Sensorless homing feature.  The gantry bangs into hard stops a few times (is only only one bang on 2.0.x branch) then drops 5mm from there.  The display shows Z absolute at 5 (on 2.0.x branch it showed 259 for Z at this point and did not drop 5mm).

'''
commit e0ca244623ea106c3bfdb9b576dfd52487d43abd (upstream/bugfix-2.0.x, kpishere/Marlin/bugfix-2.0.x, bugfix-2.0.x)
Author: kpishere <kevindpeck@gmail.com>
Date:   Fri Feb 12 06:00:41 2021 -0500

    Fix AZSMZ_12864 on SMART RAMPS (#21056)
'''

### Configuration Files

Configuration files are attached.
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/5973903/Archive.zip)

### Steps to Reproduce

* Restart printer
* Enter following commands via pronterface: M502 M500 M501 M503 G28 M114

Results of output are shown here: 

>>> m503
SENDING:M503
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:; Filament settings: Disabled
echo:  M200 S0 D1.75
echo:; Steps per unit:
echo: M92 X80.00 Y80.00 Z400.00 E398.00
echo:; Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:; Home offset:
echo:  M206 X-21.00 Y-12.00 Z0.00
echo:; Unified Bed Leveling:
echo:  M420 S0 Z10.00
Unified Bed Leveling System v1.01 inactive
echo:; Active Mesh Slot: -1
echo:; EEPROM can hold 7 meshes.
echo:; Material heatup parameters:
echo:  M145 S0 H180 B70 F0
echo:  M145 S1 H240 B110 F0
echo:; PID settings:
echo:  M301 P22.52 I1.91 D66.42
echo:  M304 P52.54 I9.72 D189.38
echo:; LCD Contrast:
echo:  M250 C190
echo:; Z-Probe Offset (mm):
echo:  M851 X55.00 Y19.00 Z-1.11
echo:; Stepper driver current:
echo:  M906 X650 Y650 Z800
echo:  M906 T0 E550
echo:; StallGuard threshold:
echo:  M914 X80 Y70 Z50
echo:; Driver stepping mode:
echo:  M569 S1 X Y
echo:  M569 S1 T0 E
echo:; Filament load/unload lengths:
echo:  M603 L0.00 U100.00
>>> m111 s247
SENDING:M111 S247
echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
echo:M105
echo:M105
>>> g28
SENDING:G28
echo:G28
>>> G28  X-42.00 Y-24.00 Z0.00
Machine Type: Cartesian
Probe: FIX_MOUNTED_PROBE
Probe Offset X55.00 Y19.00 Z-1.11 (Right-Back & Below Nozzle)
Auto Bed Leveling: UBL (disabled)
>>> homeaxis(90)
Home Fast: 388.50mm
>>> do_homing_move  X-42.00 Y-24.00 Z0.00
...(Z, 388.50, [4.00])
<<< do_homing_move  X-42.00 Y-24.00 Z0.00
Move Away: -2.00mm
>>> do_homing_move  X-42.00 Y-24.00 Z0.00
...(Z, -2.00, [4.00])
<<< do_homing_move  X-42.00 Y-24.00 Z0.00
Re-bump: 4.00mm
>>> do_homing_move  X-42.00 Y-24.00 Z0.00
...(Z, 4.00, 1.00)
<<< do_homing_move  X-42.00 Y-24.00 Z0.00
Endstop 90 hit at Phase:8 Delta:888 Distance:-0.14
>>> do_homing_move  X-42.00 Y-24.00 Z0.00
...(Z, -0.14, 1.00)
<<< do_homing_move  X-42.00 Y-24.00 Z0.00
>>> set_axis_is_at_home(Z)
Axis 90 home_offset = 0.00 position_shift = 0.00
> home_offset[Z] = 0.00
  current_position= X-42.00 Y-24.00 Z0.00 :
<<< set_axis_is_at_home(90)
  current_position= X-42.00 Y-24.00 Z0.00 : sync_plan_position
  current_position= X-42.00 Y-24.00 Z0.00 : > AFTER set_axis_is_at_home
<<< homeaxis(90)
Raise Z (before homing) by 5.00
>>> do_blocking_move_to  X-42.00 Y-24.00 Z0.00
>  X-42.00 Y-24.00 Z5.00
<<< do_blocking_move_to  X-42.00 Y-24.00 Z5.00
>>> homeaxis(88)
Home Fast: -384.00mm
>>> do_homing_move  X-42.00 Y-24.00 Z5.00
...(X, -384.00, [50.00])
<<< do_homing_move  X-42.00 Y-24.00 Z5.00
Endstop 88 hit at Phase:360 Delta:488 Distance:0.37
>>> do_homing_move  X-42.00 Y-24.00 Z5.00
...(X, 0.37, 25.00)
<<< do_homing_move  X-42.00 Y-24.00 Z5.00
>>> set_axis_is_at_home(X)
Axis 88 home_offset = -21.00 position_shift = 0.00
> home_offset[X] = -21.00
  current_position= X-21.00 Y-24.00 Z5.00 :
<<< set_axis_is_at_home(88)
  current_position= X-21.00 Y-24.00 Z5.00 : sync_plan_position
  current_position= X-21.00 Y-24.00 Z5.00 : > AFTER set_axis_is_at_home
<<< homeaxis(88)
>>> homeaxis(89)
Home Fast: -370.50mm
>>> do_homing_move  X-21.00 Y-24.00 Z5.00
...(Y, -370.50, [50.00])
<<< do_homing_move  X-21.00 Y-24.00 Z5.00
Endstop 89 hit at Phase:440 Delta:456 Distance:0.35
>>> do_homing_move  X-21.00 Y-24.00 Z5.00
...(Y, 0.35, 25.00)
<<< do_homing_move  X-21.00 Y-24.00 Z5.00
>>> set_axis_is_at_home(Y)
Axis 89 home_offset = -12.00 position_shift = 0.00
> home_offset[Y] = -12.00
  current_position= X-21.00 Y-12.00 Z5.00 :
<<< set_axis_is_at_home(89)
  current_position= X-21.00 Y-12.00 Z5.00 : sync_plan_position
  current_position= X-21.00 Y-12.00 Z5.00 : > AFTER set_axis_is_at_home
<<< homeaxis(89)
  current_position= X-21.00 Y-12.00 Z5.00 : sync_plan_position
<<< G28  X-21.00 Y-12.00 Z5.00
echo:M105
echo:M105
>>> M114
SENDING:M114
echo:M114
X:-42.00 Y:-24.00 Z:5.00 E:0.00 Count X:-1680 Y:-960 Z:2000

**Expected behavior:**

I'm expecting gantry to hit limits, one or two bangs to ensure both steppers are at limit, stay there, and then the current absolute Z position is set to the Z-MAX value.

**Actual behavior:**

Gantry hits limits a little more than desired, it then drops down 5mm and sets the current absolute Z position to Zero.

#### Additional Information

In src/gcode/calibrate/G28.cpp file, I'm seeing code changes between bugfix branch and 2.0.x that include this (below).  It looks as if there has been an oversight as to the significance of Z_HOME_DIR which is (was) also used to indicate Z-MAX homing for initial rough calibration and followed by finer adjustments with Z-Probe in bed level adjustments.   

     // Home Z last if homing towards the bed
-    #if Z_HOME_DIR < 0
-
+    #if DISABLED(HOME_Z_FIRST)

kpishere, 2021-02-12:
Did some more playing with this and yeah, something is not flipped that should be.  

In configuration.h I set 

#define Z_HOMING_HEIGHT  0 
define Z_AFTER_HOMING  0  

and that gets rid of the grinding or hammering nicely.  Still, Z = 0 at this point.  If I try to move positive, it grinds.  If I try to move negative, it does not move.  I can move to Z=0.1 and then back to Z=0.

What needs to happen is this print zone needs to be shifted down by Z-MAX.  I'd tried editing motion.cpp to have position_shift[axis] set to z-max instead of zero but when I do G30 after that, this value is not respected.

So yeah, still learning about the rational of coordinate system here and what is global etc.

ellensp, 2021-02-12:
This report is nonsensical, the commit in questions doesn't change anything anywhere near x-max or z-probe. 
It add one comment  and a extra check around ad error message.

ellensp, 2021-02-12:
Your config shows you have enabled #define SERIAL_PORT_2 1 !! which by you own comments you added in the commit "Serial1 -- Pins D18 and D19 are used for Z-MIN and Z-MAX"  will interfere with Z-MIN and Z-MAX!

closed 2021-02-12 by ellensp

kpishere, 2021-02-13:
Ouch. That is harsh ellensp.  It can all be very confusing, no argument there.  However, I'm beyond having issues with serial ports or end stops. Those are all working well. Arduino Serial is used for zmax and zmin pins. I have second hardware serial port configured for TMC drivers, that uses Arduino Serial2.  I'm trying to find what has changed and my reference may not be of any help but the problem remains.  When I home to zmax, the position gets set to zero when it should be zmax.

What I'm looking for is same behaviour as Prusia Mk3.

kpishere, 2021-02-13:
@ellensp Ah!  I've figured it out.  In my Configuration.h file, I should have the following value un-defined 

#define MANUAL_Z_HOME_POS 0

So that in Conditionals_post.h the following is then set 

#ifdef MANUAL_Z_HOME_POS
  #define Z_HOME_POS MANUAL_Z_HOME_POS
#else
  #define Z_HOME_POS (Z_HOME_DIR < 0 ? Z_MIN_POS : Z_MAX_POS)
#endif

And ultimately in motion.h the following function is defined 

#define XYZ_DEFS(T, NAME, OPT) \
  inline T NAME(const AxisEnum axis) { \
    static const XYZval<T> NAME##_P DEFS_PROGMEM = { X_##OPT, Y_##OPT, Z_##OPT }; \
    return pgm_read_any(&NAME##_P[axis]); \
  }
XYZ_DEFS(float, base_home_pos,  HOME_POS);

The ticket can be closed.