MarlinFirmware/Marlin #21067: [BUG] Fan speed not working in new Ender 3 V2 status area

Leo06ba, 2021-02-13:
Yesterday i flashed the new marlin bugfix with the upgraded status area, and i noticed that the fan speed doesn't work, both with manual setting from the control menu or during prints. Am i doing something wrong? Or it's just not fully integrated yet?

CRCinAU, 2021-02-13:
Took me a while to figure out what this bug report meant..... The fan does actually work - but the fan icon remains at 0.

See pic:
[image]

Leo06ba, 2021-02-13:
> Took me a while to figure out what this bug report meant..... The fan does actually work - but the fan icon remains at 0.
> 
> See pic:
> [image]
> 

Exactly. Sorry for my poor explaining skills.

ellensp, 2021-02-15:
This fix has been merged. Closing

closed 2021-02-15 by ellensp

hfjim-coder, 2021-02-16:
I have a question. How to fix it if it isn't fixed already?

ellensp, 2021-02-16:
you look at the PR's files that where changed https://github.com/MarlinFirmware/Marlin/pull/21072/files and you edit your copy of marlin with this change. Or you download current bugfix,