MarlinFirmware/Marlin #21069: [BUG] MKS Robin nano V3, wifi module won't compile

Malderin, 2021-02-13:
### Bug Description

Compile error with option  
`  #define MKS_WIFI_MODULE  // MKS WiFi module`

In file included from Marlin\src\lcd/extui/lib/mks_ui/draw_ui.h:85,
                 from Marlin\src\MarlinCore.cpp:71:
Marlin\src\lcd/extui/lib/mks_ui/wifiSerial.h:38:10: fatal error: libmaple/libmaple_types.h: No such file or directory
   38 | #include <libmaple/libmaple_types.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio\build\mks_robin_nano_v3\src\src\core\utility.cpp.o
*** [.pio\build\mks_robin_nano_v3\src\src\MarlinCore.cpp.o] Error 1
========================= [FAILED] Took 38.52 seconds =========================

### Configuration Files

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/5976221/Configs.zip)

snakesrules94, 2021-02-13:
Hello , not sure it can solve your problem , but serial_port_0 should be -1 , not 3 . Port 3 is for HK43 screen , on dedicated uart .

rhapsodyv, 2021-02-26:
The wifi module code wasn't ported to Nano V3 on Marlin yet.

@makerbase-mks @MKS-Sean may send it to Marlin soon.

MKS-Sean, 2021-02-27:
@rhapsodyv @Malderin @snakesrules94 
I completed it in the github of MKS, I need to organize it and submit it to marlin

snakesrules94, 2021-03-02:
> @rhapsodyv @Malderin @snakesrules94
> I completed it in the github of MKS, I need to organize it and submit it to marlin

@MKS-Sean : is it already available on MKS github for test ? my nano V3 is waiting for some tests :-)

th0m4sek, 2021-03-14:
I have same problem o nano 1.2 if i'm trying to compile with mks_robin_nano35_stm32 env.
without wifi printer is working fine with lvgl.
compilation with mks_robin_nano35 is working but my screen stays black...

ps. there are plans to support mks wifi protocol for colorui ?

rhapsodyv, 2021-03-14:
@th0m4sek your screen should not stay black and using lvgl and env mks_robin_nano35... can you open an issue with more details and your configs?

As soon as mks send the stm32 code for their WiFi module, we will be able to work in porting it to COLOR UI.

th0m4sek, 2021-03-14:
ok i will confirm with one more flash ad create issue.
esp3d implemented mks protocol is it not enough ?

th0m4sek, 2021-03-14:
I did some more tests
When i try to activate mks_wifi_module I get compilation error using stm32 and blackscreen using libmaple one.
I'm using my own esp9266 with esp3d on branch 3.0 and I was thinking that it is making some problems but as a final test I removed module complitly and try to flash again with same results.
as mentioned before I'm using mks robin nano 1.2.

rhapsodyv, 2021-03-14:
You should compile with `mks_robin_nano35` and use the esp firmware from mks!!

Standard esp3d will not work with the “MKS_WIFI_MODULE” on marlin.
They are two very different things.

th0m4sek, 2021-03-14:
I will test mks firmware next week but I sill think that it should work with esp3d in version 3 https://github.com/luc-github/ESP3D/issues/523

rhapsodyv, 2021-03-14:
> I will test mks firmware next week but I sill think that it should work with esp3d in version 3 [luc-github/ESP3D#523](https://github.com/luc-github/ESP3D/issues/523)

Hum... I will test it next week. 

I just tested Nano V2 + LVGL + Wifi Module (mks firmware) and it worked 100%. I'm adding meatpack support for the cura plugin.

snakesrules94, 2021-03-18:
> 
> 
> @rhapsodyv @Malderin @snakesrules94
> I completed it in the github of MKS, I need to organize it and submit it to marlin

Hi there ! @MKS-Sean , any news on this subject ?

th0m4sek, 2021-03-21:
there is some problem with m configuration on fresh bugfix i was able to complie working image i'm still searching for breaking change

Liriel-666, 2021-04-13:
and what is now the solution because i get that error?

ellensp, 2021-04-13:
there is none yet, thus why this is still open.

Liriel-666, 2021-04-13:
ok because igot a closed symbol of that bug report

ellensp, 2021-04-13:
your bug report was closed because it  because it was a duplication of this issue. There is no point having two issues open over the same thing. This issue is still open and ongoing.

th0m4sek, 2021-04-18:
I've finally pinpointed not booting MKS Robin nano 1.2 with wifi module.
ucommented line i configuration_adv.conf i stoping board from booting.
#define RX_BUFFER_SIZE 1024
Should I create new issue?

stream2me, 2021-04-18:
I made some progess...
To implement the nano v3 MKS made a lot of changes.
After changing some files I was able to compile without any error but the Wifi-Modul is not working.
Files i changed:

- Marlin/src/lcd/extui/lib/mks_ui/irq_overrid.cpp
- Marlin/src/pins/stm32f4/pins_MKS_ROBIN_NANO_V3.h
- Marlin/src/lcd/extui/lib/mks_ui/wifi_module.cpp
- Marlin/src/lcd/extui/lib/mks_ui/wifi_module.h
- Marlin/src/lcd/extui/lib/mks_ui/wifiSerial.cpp
- Marlin/src/lcd/extui/lib/mks_ui/wifiSerial.h

But there are more things to change.
The according commit in MKS-marlin is [this one](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/commit/c5e33578681be1aad19fc1d577d984af9fe99ce0#diff-c805367a85c4aa4a385822b7152920c41d62921fe3638e7cf573063aab5d8d95).

Maybe someone with more programming skills can have o look on it.

th0m4sek, 2021-05-01:
@rhapsodyv did you look to esp3d ? is it enough to think about colorui implementation ?

rhapsodyv, 2021-05-01:
> @rhapsodyv did you look to esp3d ? is it enough to think about colorui implementation ?

I have a nano v3 and the WiFi module. I just didn’t have time yet to do the color ui version.

Souls-Reaver, 2021-06-16:
hola a todos.
Tras probar distintas distribuciones de marlin, incluyendo la 2.0.8.1, 2.0.8.2, 2.0.9, y las bugfix me encuentro el error de que en una robin nano v3 con el wifi activado, me da este error:
[image]

y bien, si comento el wifi, me compila sin problema alguno:
[image]

alguien sabe solucionar este error?

me gustaría también, y no se si puede ser que en Colour_ui estuviera activo el wifi y tuviera el mismo funcionamiento que con LVGL_UI.

me podéis ayudar?

ellensp, 2021-07-13:
The wifi part is fixed, but two new errors have shown up 
see https://github.com/MarlinFirmware/Marlin/pull/22348 and https://github.com/MarlinFirmware/Marlin/issues/22349

ellensp, 2021-07-21:
I'm going to close this as the wifi modules now compiles.
I will open up a new issue as other issues exists

closed 2021-07-21 by ellensp