MarlinFirmware/Marlin #21070: [BUG] host serial problem/ eeprom error on MKS Robin Nano V3

snakesrules94, 2021-02-13:
Hi there , here are the behavior i have when using Bugfix branch with board robin nano V3

### Bug Description

Unable to connect the card from host thru serial port
EEprom write error on M500

### Configuration Files

Platformio.ini : target mks_robin_nano_v3_usb_flash_drive or mks_robin_nano_v3_usb_flash_drive_msc
configuration.h : #define SERIAL_PORT -1 / #define EEPROM_SETTINGS
configuration_adv.h : #define USB_FLASH_DRIVE_SUPPORT & #define USE_OTG_USB_HOST

### Steps to Reproduce

Compile Marlin with previous config .

**Expected behavior:**

i should be able to connect the board from host ( Pronterface as example )
M500 should save settings to eeprom

**Actual behavior:**

Unable to connect to host with this config , it works when choosing mks_robin_nano_v3 as target in platformio.ini
Unable to save settings in eeprom in mks_robin_nano_v3 configuration , get some writting to eeprom error message; I can't say now for others targets as i have no usb serial connection

#### Additional Information

Regards.

rhapsodyv, 2021-02-13:
I will take a look

snakesrules94, 2021-02-13:
Obrigado :-)

jzamero, 2021-02-23:
Same problem here

rhapsodyv, 2021-02-23:
Sorry about the delay, I was a little busy. I just tested it. The serial is working very fine for me. But the EEPROM have issues.

Can you share more info about the serial issue? How are you powering the board? Is there any log from Pronterface? 

I'm using `miniterm.py` with success.

I will work on the eeprom issue.

rhapsodyv, 2021-02-24:
Just fixed the eeprom issue here: https://github.com/MarlinFirmware/Marlin/pull/21174 

But the serial is always working for me.

snakesrules94, 2021-02-24:
Hi Victor, on my side , board is powered with 12V supply. The only thing i can read from pronterface is : 
`Connecting...
[ERROR] Can't write to printer (disconnected?) (SerialException): Write timeout`
. I'm using right now the marlin dev branch from 22/02 , got mks_robin_nano_v3_usb_flash_drive_msc as target , and #define SERIAL_PORT -1

snakesrules94, 2021-02-24:
Update : just try your mod for eeprom , works OK 👍 
regarding serial , i've just noticed that the serial port is seen as STM serial port by my computer , with my nano V2 , seen as a CH340 serial port , maybe i've got an issue here , i will check this .

snakesrules94, 2021-02-24:
Update 2 : with target mks_robin_nano_v3 , serial port is exposed to host and working , i can open it with pronterface and put some commands.
With target mks_robin_nano_v3_usb_flash_drive, serial port is exposed to host and working , i can open it with pronterface and put some commands.
With target mks_robin_nano_v3_usb_flash_drive_msc , serial port is exposed to host and not working , i can't open it with pronterface, media is not exposed to host .

jzamero, 2021-02-24:
Have exactly the same behavior in my tests
When using mks_robin_nano_v3_usb_flash_drive the screen (mks st35s) is unable to read Stu files from usb nor sd  card just load a new firmware

snakesrules94, 2021-02-24:
@jzamero : reading file from usb is working fine for me , have you enable `#define USB_FLASH_DRIVE_SUPPORT` and `#define USE_OTG_USB_HOST` in configuration_adv.h ?

jzamero, 2021-02-24:
Try it Not reading USB Drive or SD-Card

rhapsodyv, 2021-02-24:
I will test serial on mks_robin_nano_v3_usb_flash_drive_msc.

rhapsodyv, 2021-02-24:
Sorry, usb worked for me in _msc env too. I think you have something missing in your computer. Maybe a cdc driver missing...
Try another serial clients...

[image]

snakesrules94, 2021-02-24:
@rhapsodyv : no way with Putty as well , i've got "error writing serial port"
Could you share your bin file to test on my card ? i don't care about config , i've just got naked board & TS35 on desktop , no drivers , no sensors.

jzamero, 2021-02-24:
Delete the driver stm virtual com port and then connect again to computer
it will solve the problem

snakesrules94, 2021-02-24:
@jzamero : i've just tried on another computer without any STM stuff , the card is seen as "Serial USB" with drivers from microsoft , it works ok on using mks_robin_nano_v3_usb_flash_drive but still no way on _msc .

rhapsodyv, 2021-02-24:
> @jzamero : i've just tried on another computer without any STM stuff , the card is seen as "Serial USB" with drivers from microsoft , it works ok on using mks_robin_nano_v3_usb_flash_drive but still no way on _msc .

I'm using a Mac and both works. I will try it in a windows computer.

snakesrules94, 2021-02-24:
Just have a quick test under Ubuntu and indeed , it's working : serial port can be opened and command sent :
`-- Miniterm on /dev/ttyACM0  9600,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
M␛[E␛[2~␛[6~
echo:Unknown command: "M␛[E␛[2~␛[6~"
ok
M503
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius

echo:; Filament settings: Disabled
echo:  M200 S0 D1.75
echo:; Steps per unit:
echo: M92 X80.00 Y80.00 Z400.00 E500.00
echo:; Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Material heatup parameters:
echo:  M145 S0 H180 B70 F0
echo:  M145 S1 H240 B110 F0
echo:; PID settings:
echo:  M301 P22.20 I1.08 D114.00
ok
`
So definitivly something wrong under windows host .

snakesrules94, 2021-02-26:
@rhapsodyv , I forget to ask : media sharing is working for you or not yet ?

jzamero, 2021-02-26:
Working good

snakesrules94, 2021-02-26:
@jzamero , i guess you're not using windows :-) I've just tried in Linux and indeed , media sharing and com access is working . In windows not. When i plug the usb cord ( under win 10 ) , i can hear the usb connect sound , but right after , the disconnect sound , and then the usb connect sound again.

jzamero, 2021-02-26:
I do use windows
As I mentioned earlier
Plug the board then go to device manager  and delete the serial com
relevant to your device

Don’t forget to mark delete also the driver files.

Reboot
Plug it back.
Let windows do the magic of installation and you have a working card

Good luck
-- 
בברכה ג׳קי זמרו

snakesrules94, 2021-02-26:
Ok , following your advice , i removed the driver with which the board was recognised ( GENERICSTM32xxx ) , after that , board was recognised as an HID device , withous any possible action , i then change the driver to composite serial usb , and as you said , now magic is working ! com port is exposed and available , media is also exposed and available !
I don't know if it's linked to some ST stuff installed on my computer ( i also use stlink device , and bluepill board ) or whatever , but now it's working . Thanks for your advice .

rhapsodyv, 2021-02-26:
Yes, media share for usb flash drive need external power supply to work 100%.

So, everything is ok here? This can be closed?

snakesrules94, 2021-02-26:
@rhapsodyv : yes , main concerns are solved .thanks again for the eeprom fix . You can mark this issue as closed.

PR #21174 (2021-02-25): Fix MKS Robin Nano V3 I2C pins
### Description

MKS Robin Nano V3 doesn't use the default variant I2C pins. This PR fix that.

### Benefits

Fix #21070
closed 2021-02-26 by rhapsodyv