MarlinFirmware/Marlin #21071: [BUG] MKS Robin Nano V2.0 - Runout Sensor not compiling

DanJunior78, 2021-02-13:
### Bug Description

compiler stops with following error:

Compiling .pio\build\mks_robin_nano35\src\src\gcode\gcode.cpp.o
In file included from Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/../../feature/../lcd/tft_io/tft_io.h:29:0,
                 from Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/../../feature/../lcd/tft_io/touch_calibration.h:22,
                 from Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/../../feature/../lcd/marlinui.h:39,
                 from Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/../../feature/tmc_util.h:25,
                 from Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/trinamic.h:35,
                 from Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/indirection.h:44,
                 from Marlin\src\gcode\feature\runout\../../../feature/../module/stepper.h:47,
                 from Marlin\src\gcode\feature\runout\../../../feature/runout.h:31,
                 from Marlin\src\gcode\feature\runout\M412.cpp:28:
Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/../../feature/../lcd/tft_io/../../HAL/STM32F1/tft/tft_spi.h:24:10: fatal error: ../../../inc/MarlinConfig.h: No such file or directory
 #include "../../../inc/MarlinConfig.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\mks_robin_nano35\src\src\gcode\feature\runout\M412.cpp.o] Error 1

The mid December bugfix version is compiling fine. If i deactivate the runout sensor. It´s compiling also without error.

### Configuration Files

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5976612/Marlin.zip)

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. Compile with my config files
2. It is compiling when deactivating the runout sensor

**Expected behavior:**

see above

**Actual behavior:**

see above

ellensp, 2021-02-14:
Cannot reproduce, tried both mks_robin_nano35_stm32 and mks_robin_nano35 build environments 
Using provided config files and current bugfix. Verified  runout sensor is enabled and same code is being compiled

```
Compiling .pio/build/mks_robin_nano35/src/src/gcode/feature/pause/M701_M702.cpp.o
Compiling .pio/build/mks_robin_nano35/src/src/gcode/feature/runout/M412.cpp.o
Compiling .pio/build/mks_robin_nano35/src/src/gcode/feature/trinamic/M122.cpp.o
```

ellensp, 2021-02-14:
delete  your **.platformio** directory in your home directory. Force platformio to download new build environments and libraries

DanJunior78, 2021-02-14:
So, i am back again. Found the problem. It was not related to that. Funny part of the story, there was the V3 implemented and with the standard "mks_robin_nano35" environment, i wasn´t able to build it. There was a lot of work going to the MKS boards the last weeks.
I´ve checked the V3 has a different processor onboard (STM...F1 <-> STM...F4). So...digging around i´ve seen there was a change there and with the environment "mks_robin_nano35_stm32" it was doing its job.
Thanks a lot for your fast answer. 
Sometimes Platformio (i am more the old school programmer), kills my poor skills ;o).

closed 2021-02-14 by DanJunior78