MarlinFirmware/Marlin #21075: [BUG]  M914 - StallGuard cannot be set for the Z axis

DrumClock, 2021-02-14:
Hi,
I use TMC 2209 drivers, other parameters can only be overwritten
M914 Z does not work and is not read from the FW settings.

configuration files:  [Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5977295/Configuration.zip)

PRONTERFACE listing:
```
>>> m122s0
SENDING:M122S0
		X	X2	Y	Z	E	E1
Address		0	2	0	0	2	2
Enabled		false	false	false	false	false	false
Set current	300	300	300	500	800	800
RMS current	275	275	275	489	795	795
MAX current	388	388	388	689	1121	1121
Run current	8/31	8/31	8/31	15/31	25/31	25/31
Hold current	4/31	4/31	4/31	7/31	12/31	12/31
CS actual	4/31	4/31	4/31	7/31	12/31	12/31
PWM scale
vsense		1=.18	1=.18	1=.18	1=.18	1=.18	1=.18
stealthChop	true	true	true	true	true	true
msteps		16	16	16	16	16	16
interp		true	true	true	true	true	true
tstep		max	max	max	max	max	max
PWM thresh.	49	49	49	4	82	82
[mm/s]		201	201	201	123	30	30
OT prewarn	false	false	false	false	false	false
triggered
 OTP		false	false	false	false	false	false
pwm scale sum	5	5	5	9	14	14
pwm scale auto	0	0	0	0	0	0
pwm offset auto	36	36	36	36	36	36
pwm grad auto	14	14	14	14	14	14
off time	4	4	4	4	4	4
blank time	24	24	24	24	24	24
hysteresis
 -end		2	2	2	2	2	2
 -start		1	1	1	1	1	1
Stallguard thrs	64	64	64	0	0	0
uStep count	8	8	8	8	8	8
DRVSTATUS	X	X2	Y	Z	E	E1
sg_result	0	0	0	0	0	0
stst
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0xC0:04:00:00
		X2	0xC0:04:00:00
		Y	0xC0:04:00:00
		Z	0xC0:07:00:00
		E	0xC0:0C:00:00
		E1	0xC0:0C:00:00
Testing X connection... OK
Testing X2 connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
Testing E1 connection... OK
```

```
>>> m914
SENDING:M914
X homing sensitivity: 64.0000000000
X2 homing sensitivity: 64.0000000000
Y homing sensitivity: 64.0000000000

```

```
>>> M914 Z64
SENDING:M914 Z64
```
```

>>> M914
SENDING:M914
X homing sensitivity: 64.0000000000
X2 homing sensitivity: 64.0000000000
Y homing sensitivity: 64.0000000000
```

ellensp, 2021-02-14:
Z_SENSORLESS is being disabled.

The code shows
```
#if Z_HOME_DIR < 0 && (!HAS_CUSTOM_PROBE_PIN || ENABLED(USE_PROBE_FOR_Z_HOMING))
    #define HOMING_Z_WITH_PROBE 1
  #endif

```
This is true and so it defines HOMING_Z_WITH_PROBE

Later in/Conditionals_post.h
```
  // Disable Z axis sensorless homing if a probe is used to home the Z axis
  #if HOMING_Z_WITH_PROBE
    #undef Z_STALL_SENSITIVITY
    #undef Z2_STALL_SENSITIVITY
    #undef Z3_STALL_SENSITIVITY
    #undef Z4_STALL_SENSITIVITY
  #endif

```
So this disabled Z_STALL_SENSITIVITY on Z if you are probing 

Basically you cannot adjust Z stall sensitivity if your using HOMING_Z_WITH_PROBE

This is the what is going on, but not the why is this like this...  that I don't know.

ellensp, 2021-02-14:
you could try this in Conditionals_post.h
```
  #if !defined(BLTOUCH) && HOMING_Z_WITH_PROBE
    #undef Z_STALL_SENSITIVITY
    #undef Z2_STALL_SENSITIVITY
    #undef Z3_STALL_SENSITIVITY
    #undef Z4_STALL_SENSITIVITY
  #endif
```
So if your using a bltouch It wont disable Z_SENSORLESS

DrumClock, 2021-02-14:
Hi, @ellensp 
I'm trying to use the StallGuard function for the Z-max endstop.
I connected the ZDIAG pin TMC 2209 to the Z-max pin and thought
that it will work the same as with the X2 stepper motor which I use for IDEX mod (E2DIAG connected to Ymax endstop).

But maybe I want Marlin something he can't do yet ....

ellensp, 2021-02-14:
but you have #define Z_HOME_DIR -1 
If you had #define Z_HOME_DIR 1 then HOMING_Z_WITH_PROBE wouldn't be set

DrumClock, 2021-02-14:
Otherwise.....
It is possible to set #define USE_ZMAX_PLUG in Marlin to
used the StallGuard function from the ZDIAG - TMC 2209 pin?

closed 2021-03-29 by DrumClock