MarlinFirmware/Marlin #21078: [BUG] Ender 3 v2 Z-offset not working

HoneypotFX, 2021-02-14:
### Bug Description
Z-offset does not change.

### Steps to Reproduce

1. Go to the Prepare menu, select Auto-home then move Z to 0
2. Go to the Z-offset and adjust it
3. Press button to commit

**Actual behavior:**

It doesn't respond to any changes. You can go down to -10 or up to 10 it doesn't move.

**Expected behavior:**

When you adjust the Z-offset the z-offset should change.

### Configuration Files
[configuration_files.zip](https://github.com/MarlinFirmware/Marlin/files/5978464/configuration_files.zip)

#### Additional Information
I tried 2.0.7, 2.0.7.1, 2.0.7.2, bugfix-2.0.x, this bug appears in these branches.

CRCinAU, 2021-02-14:
This is not a bug. The Z Offset is only applied on the next Z move...

steljwagh, 2021-02-15:
It's indeed weird why it's not live in this firmware, it makes setting the Z offset a hassle compare to getting live feedback.

Leo06ba, 2021-02-15:
I can confirm this. Very annoying.

trahan625, 2021-02-15:
I am going to confirm this is annoying as heck.

CJfoks, 2021-02-18:
Also this problem exist during a printing, any changes in Z offset is ignored until next homing.

strayr, 2021-02-18:
I can't set a negative Z-offset on 2.0.7.2, the Z offset Wizard just won't do it and everywhere else the number goes in but the head doesn't move. So can confirm is happening for me on latest release, will try to compile against bugfix and see what goes wrong.

I don't have the configs for stock ender and I'm running a CR10 screen on a 4.2.2 board amongst other things so pre-compiled doesn't help me and doesn't fix the problem.

In 2.0.6.1 bugfix from https://github.com/teachingtechYT/Marlin/tree/Ender3-V2-BLtouch, i have it working, but the config is different in a few places, it's on CONFIGURATION_H_VERSION 020008, so copypasta on configs doesn't just work, so i i'm trying to track down whether this is a regression in 2.0.7 or a fix in -bugfix or both. Was actually tracking looking for relevant commits when I saw this.

Attached is my mess of a configuration. It was working fine with everything else, not sure if this is helpful as what I'm doing is quite a way from non-standard.
[config.zip](https://github.com/MarlinFirmware/Marlin/files/6005171/config.zip)

HoneypotFX, 2021-02-22:
> In 2.0.6.1 bugfix from https://github.com/teachingtechYT/Marlin/tree/Ender3-V2-BLtouch, i have it working

Live z-offset configuration doesn't work on that either.

tiagofreire-pt, 2021-02-25:
> 
> 
> > In 2.0.6.1 bugfix from https://github.com/teachingtechYT/Marlin/tree/Ender3-V2-BLtouch, i have it working
> 
> Live z-offset configuration doesn't work on that either.

Try this one: https://github.com/Jyers/Marlin/releases/tag/v1.1.1

strayr, 2021-02-26:
> > In 2.0.6.1 bugfix from https://github.com/teachingtechYT/Marlin/tree/Ender3-V2-BLtouch, i have it working
> Live z-offset configuration doesn't work on that either.
Confirmed, can't set Z offset when my first layer is going down, but the Z-offset wizard and other methods of setting z-offset when not printing ARE working and i can print ok, and they are NOT in 2.0.7.2

strayr, 2021-02-26:
> Try this one: https://github.com/Jyers/Marlin/releases/tag/v1.1.1
Why is the source code zipped so you can't browse or compare it against other versions using git? A binary might be an immediate fix, but it's not going to help find a regression or get a fix back into mainline

tiagofreire-pt, 2021-02-27:
> Why is the source code zipped so you can't browse or compare it against other versions using git? A binary might be an immediate fix, but it's not going to help find a regression or get a fix back into mainline

Source code: https://github.com/Jyers/Marlin

stephenwheet, 2021-02-28:
Any update on this. I’m also having an issue with z-offset.

tfaz099, 2021-03-24:
> > This is not a bug. The Z Offset is only applied on the next Z move...
> 
> I don't think so. It's working on 2.0.1 with Creality's firmware.
Agreed, it even moves in babystepping and click to commit.  I found that if i change z-offset and click then perform an autohome and then move z to 0, is the only way to test.  on every other firmware moving z offset and clicking moves the z axis in the direction requested.

strayr, 2021-03-28:
This appears fixed in bugfix-2.0.x as of 2021-03-27?

I appear to be getting results live-tuning the calibration file generated by superslicer.

Importantly my babysteps are tied to my zprobe offset, i don;t know if that makes a difference for anyone else.

#define BABYSTEPPING
#define BABYSTEP_ZPROBE_OFFSET

HoneypotFX, 2021-03-28:
> This appears fixed in bugfix-2.0.x as of 2021-03-27?
>
> I appear to be getting results live-tuning the calibration file generated by superslicer.
>
> Importantly my babysteps are tied to my zprobe [...]

Not working for me. Go to the Prepare menu, select Z offset, tune it. Is it move?

strayr, 2021-03-28:
Ok, orginal problem steps to reproduce not fixed, it does _appear_ to be adjusting a live print though where it wasn't before. Sorry, I apear to be commenting on a parallel issue.

I'm wondering whether the offset gets called every time a UBL mesh adjust happens though, or I just got lucky with the mesh behaviour?

HoneypotFX, 2021-03-28:
This is the solution:
#define BABYSTEP_WITHOUT_HOMING
#define BABYSTEP_ALWAYS_AVAILABLE

closed 2021-03-28 by HoneypotFX