MarlinFirmware/Marlin #21082: [FR] NUCLEO-G474RE - Nucleo-64 board with STM32G4 170Mhz Cortex-M4 MCU and 512KB flash memory

Gamester17, 2021-02-15:
### Description

Feature request of HAL support in Marlin for NUCLEO-G474RE low-cost development board with STM32G4 (STM32 G4) 32-bit MCU? 

https://docs.platformio.org/en/latest/boards/ststm32/nucleo_g474re.html#board-ststm32-nucleo-g474re

NUCLEO-G474RE is based on STMicro's STM32 Nucleo-64 MB1367 reference board and STM32G474RE has ARM Cortex-M4 running at 170Mhz, 512KB flash memory, 128KB RAM memory, ARDUINO Uno V3 connectivity and ST morpho headers, plus integrated STLINK debugger/programmer interface. STM32G4 MCUs features "Hardware Math Accelerator, High-resolution Timer v2, and High Analog level integration".

https://www.st.com/en/evaluation-tools/nucleo-g474re.html

https://www.st.com/en/microcontrollers-microprocessors/stm32g474re.html

https://www.st.com/resource/en/user_manual/dm00556337-stm32g4-nucleo-64-boards-mb1367-stmicroelectronics.pdf

NUCLEO-G474RE devboard with STM32G474RE MCU cost ~ $15 from STMicro direct and looks to be available at global distributors:

https://estore.st.com/en/nucleo-g474re-cpn.html

[image]

[image]

#### Additional Information

ST released above NUCLEO-G474RE with STM32G474RE MCU about 18 months ago as its first STM32G4 Nucleo-64 devboard.

NUCLEO-G491RE and NUCLEO-G431RB are two another STM32G4 Nucleo-64 devboards with slightly different specifications:

https://www.st.com/en/evaluation-tools/nucleo-g491re.html

https://www.st.com/en/evaluation-tools/nucleo-g431rb.html

They have the same Nucleo-64 form factor but comes with STM32G491RE and STM32G431RB MCU chips respecivly:

https://www.st.com/en/microcontrollers-microprocessors/stm32g491re.html

https://www.st.com/en/microcontrollers-microprocessors/stm32g431rb.html

The STM32G4 series of MCUs as a whole have three lines of products:

- The STM32G4x1 Access line – general-purpose microcontrollers with an entry-level set of analog peripherals
- The STM32G4x3 Performance line – general-purpose microcontrollers with the maximum number of analog peripherals
- The STM32G4x4 Hi-resolution line with high-resolution timer and complex waveform builder plus event handler (HRTIM) for digital power conversion, such as digital switched-mode power supplies, lighting, welding, solar and wireless charging

STMicroelectronics claims that the STM32G4 series should have a high degree of compatibility with the STM32F3 (STM32 F3) series: 

https://www.st.com/content/st_com/en/about/media-center/press-item.html/p4159.html

https://www.st.com/en/microcontrollers-microprocessors/stm32g4-series.html

[image]

X-Ryl669, 2021-02-16:
Is it supported in [STSTM32](https://github.com/platformio/platform-ststm32) yet ? If not, please open a FR there first.

I am asking because it has to be supported here first so that Marlin could use it.

Gamester17, 2021-02-16:
> Is it supported in [STSTM32](https://github.com/platformio/platform-ststm32) yet ? If not, please open a FR there first.
> 
> I am asking because it has to be supported here first so that Marlin could use it.

NUCLEO-G474RE and NUCLEO-G431RB is already supported by [STSTM32](https://github.com/platformio/platform-ststm32) (but NUCLEO-G491RE is not supported there yet):

https://docs.platformio.org/en/latest/boards/ststm32/nucleo_g474re.html

https://docs.platformio.org/en/latest/boards/ststm32/nucleo_g431rb.html

https://github.com/platformio/platform-ststm32/blob/252752d7c2745e387862d11f26f35cf2180f5ead/boards/nucleo_g474re.json

https://github.com/platformio/platform-ststm32/blob/252752d7c2745e387862d11f26f35cf2180f5ead/boards/nucleo_g431rb.json