MarlinFirmware/Marlin #21084: LEVEL_CORNERS_USE_PROBE Didnt Deploy Neddle

ils15, 2021-02-15:
Hi
I installed Latest Bugfix and activate LEVEL_CORNERS_USE_PROBE 

When I try to use the feature, Neddle Didn't Deploy

I'm using BLTOUCH HS Mode (i don't know if this configuration is causing the issue)

Steps to Reproduce

1- Needs a BLTOUCH
2- Activate LEVEL_CORNERS_USE_PROBE  in config settings
3- Go do Menu>Leveling>Level Corners
4- Go to the First point and saw the Neddle procedure

ellensp, 2021-02-15:
Please attach your configuration files.

ils15, 2021-02-15:
> Please attach your configuration files.

Edited

galluccib, 2021-02-17:
Running into this issue as well after flash 2/14 bugfix... configs attached.. 
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5994043/Configuration.zip)

galluccib, 2021-02-17:
Auto Home works (G28) but when I using bed visualizer it starts but the probe doesn't come down..  It's pretty much doing the same (G28 then G29)

thisiskeithb, 2021-02-18:
> I'm using BLTOUCH HS Mode (i don't know if this configuration is causing the issue)

If you disable `BLTOUCH_HS_MODE`, does it work as expected?

Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

LearnIIBurn, 2021-02-20:
Just wanted to chime in and say I am experiencing the same behavior with my BLTouch while leveling corners using probe. Also running my BLTouch in high speed mode.

I will follow your instructions thisiskeithb and add my logs here soon as possible.

qwewer0, 2021-02-20:
I too was able to reproduce the issue.

SKR Mini E3 v1.2
STM32F103RC_btt
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6014633/Configuration.zip)

<details>
  <summary>Log output (M502, M500, M111 S247, G28, LEVEL_BED_CORNERS)</summary>
  
```
Connecting...
Printer is now online.
>>> M502
SENDING:M502
echo:Hardcoded Default Settings Loaded
>>> M500
SENDING:M500
echo:Settings Stored (658 bytes; crc 13584)
//action:notification Settings Stored
>>> M111 S247
SENDING:M111 S247
echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
echo:G28
>>> G28  X-1.7000 Y-1.7000 Z0.0000
Machine Type: Cartesian
Probe: BLTOUCH
Probe Offset X-45.2000 Y-4.5000 Z-1.1550 (Left-Front & Below Nozzle)
Raise Z (before homing) by 5.0000
>>> do_blocking_move_to  X-1.7000 Y-1.7000 Z0.0000
>  X-1.7000 Y-1.7000 Z5.0000
<<< do_blocking_move_to  X-1.7000 Y-1.7000 Z5.0000
  current_position= X0.0000 Y0.0000 Z5.0000 : sync_plan_position
>>> do_blocking_move_to  X0.0000 Y0.0000 Z5.0000
>  X-407.5500 Y-347.5500 Z5.0000
<<< do_blocking_move_to  X-407.5500 Y-347.5500 Z5.0000
>>> homeaxis(88)
Home Fast: -407.5500mm
>>> do_homing_move  X0.0000 Y0.0000 Z5.0000
...(X, -407.5500, [50.0000])
<<< do_homing_move  X0.0000 Y0.0000 Z5.0000
Move Away: 5.0000mm
>>> do_homing_move  X0.0000 Y0.0000 Z5.0000
...(X, 5.0000, [50.0000])
<<< do_homing_move  X0.0000 Y0.0000 Z5.0000
Re-bump: -10.0000mm
>>> do_homing_move  X0.0000 Y0.0000 Z5.0000
...(X, -10.0000, 16.67)
<<< do_homing_move  X0.0000 Y0.0000 Z5.0000
Endstop 88 hit at Phase:184 Delta:200 Distance:0.1500
>>> do_homing_move  X0.0000 Y0.0000 Z5.0000
...(X, 0.1500, 16.67)
<<< do_homing_move  X0.0000 Y0.0000 Z5.0000
>>> set_axis_is_at_home(X)
Axis 88 home_offset = 0.0000 position_shift = 0.0000
> home_offset[X] = 0.0000
  current_position= X-1.7000 Y0.0000 Z5.0000 :
<<< set_axis_is_at_home(88)
  current_position= X-1.7000 Y0.0000 Z5.0000 : sync_plan_position
  current_position= X-1.7000 Y0.0000 Z5.0000 : > AFTER set_axis_is_at_home
<<< homeaxis(88)
>>> homeaxis(89)
Home Fast: -347.5500mm
>>> do_homing_move  X-1.7000 Y0.0000 Z5.0000
...(Y, -347.5500, [50.0000])
<<< do_homing_move  X-1.7000 Y0.0000 Z5.0000
Move Away: 5.0000mm
>>> do_homing_move  X-1.7000 Y0.0000 Z5.0000
...(Y, 5.0000, [50.0000])
<<< do_homing_move  X-1.7000 Y0.0000 Z5.0000
Re-bump: -10.0000mm
[... 261 lines cut ...]
<<< set_axis_is_at_home(90)
  current_position= X162.2000 Y119.5000 Z1.1550 : sync_plan_position
  current_position= X162.2000 Y119.5000 Z1.1550 : > AFTER set_axis_is_at_home
  current_position= X162.2000 Y119.5000 Z1.1550 : Probe::set_deployed
deploy: 0
Probe::do_z_raise(6.0000)
>>> do_blocking_move_to  X162.2000 Y119.5000 Z1.1550
>  X162.2000 Y119.5000 Z7.1550
<<< do_blocking_move_to  X162.2000 Y119.5000 Z7.1550
BLTouch STOW requested
BLTouch Command :90
bltouch.stow_proc() end
>>> do_blocking_move_to  X162.2000 Y119.5000 Z7.1550
>  X162.2000 Y119.5000 Z7.1550
<<< do_blocking_move_to  X162.2000 Y119.5000 Z7.1550
<<< homeaxis(90)
<<< home_z_safely  X162.2000 Y119.5000 Z7.1550
  current_position= X162.2000 Y119.5000 Z7.1550 : sync_plan_position
<<< G28  X162.2000 Y119.5000 Z7.1550
Disconnected.
```

</details>

[Log output (M502, M500, M111 S247, G28, LEVEL_BED_CORNERS).txt](https://github.com/MarlinFirmware/Marlin/files/6014622/Log.output.M502.M500.M111.S247.G28.LEVEL_BED_CORNERS.txt)

thisiskeithb, 2021-02-20:
> I too was able to reproduce the issue.

If you disable `BLTOUCH_HS_MODE`, does it work as expected?

qwewer0, 2021-02-20:
> If you disable `BLTOUCH_HS_MODE`, does it work as expected?

Thanks for mentioning it.

Yes, it is working with `BLTOUCH_HS_MODE` disabled. Probe pin deploys and senses the bed corners automatically.

ils15, 2021-02-20:
When i disable BLTOUCH HS mode, function works as expected

qwewer0, 2021-02-22:
A test on https://github.com/MarlinFirmware/Marlin/pull/21161 would be appreciated.

qwewer0, 2021-02-24:
@ils15, @galluccib, @LearnIIBurn
Could you test https://github.com/MarlinFirmware/Marlin/pull/21161 and compare the `LEVEL_CORNERS_USE_PROBE` behavior without `BLTOUCH_HS_MODE`?

Thanks in advance.

thisiskeithb, 2021-02-25:
Fix has been merged. Closing.

closed 2021-02-25 by thisiskeithb