MarlinFirmware/Marlin #21085: [FR] (feature request title)

BROSVORN, 2021-02-15:
### Description

### Feature Workflow

1. [First Action]
2. [Second Action]
3. [and so on...]

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

BROSVORN, 2021-02-15:
> 
> 
> ### Description
> 
> 
> 
> ### Feature Workflow
> 
> 
> 
> 1. [First Action]
> 2. [Second Action]
> 3. [and so on...]
> 
> #### Additional Information
> 
> * Provide pictures or links that demonstrate a similar feature or concept.
> * See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2021-02-15:
Please don't play with the marlin github repository.

closed 2021-02-15 by ellensp