MarlinFirmware/Marlin #21093: [FR] Support Mini12864 (FYSETC, MKS, Etc..) on BigTreeTech BTT002

kylix-rd, 2021-02-16:
### Description

Connecting a FYSETC Mini12864 display to a BigTreeTech BTT002 doesn't currently build. The board does already support REPRAP_DISCOUNT_SMART_CONTROLLER and REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER. All the requisite pins are available on EXP1 and EXP2.

### Feature Workflow

Currently trying to get FYSETC_MINI_12864_2_1 working:

Enable FYSETC_MINI_12864_2_1 in Configuration.h.

Doesn't compile.

#### Additional Information

Applying the attached patch to Marlin/src/pins/stm32F4/pins_BTT_BTT002_V1_0.h does get the code to compile and enables the associated menus.

The problem now is that the display doesn't display anything on boot. Once an SD card is inserted into the reader on the display, the display will begin to function and will remain perfectly functional until the next reboot. On each boot, an SD card needs to be inserted/removed from the reader before the display will begin to function. Normal screen updates do not work to get the initial display functioning. Weird.

A pull request with the following patch will be forthcoming.

--------------------------------
diff --git a/Marlin/src/pins/stm32f4/pins_BTT_BTT002_V1_0.h b/Marlin/src/pins/stm32f4/pins_BTT_BTT002_V1_0.h
index 939bc1eccd..b65898e49e 100644
--- a/Marlin/src/pins/stm32f4/pins_BTT_BTT002_V1_0.h
+++ b/Marlin/src/pins/stm32f4/pins_BTT_BTT002_V1_0.h
@@ -225,20 +225,49 @@
 
     #define LCD_SDSS                        PA4
 
-    #define LCD_PINS_ENABLE                 PE9
-    #define LCD_PINS_D4                     PE10
+    #if ENABLED(FYSETC_MINI_12864)
+      #define DOGLCD_CS                     PE9
+      #define DOGLCD_A0                     PE8
+      #define DOGLCD_SCK                    PA5
+      #define DOGLCD_MOSI                   PA7
+
+      #define LCD_BACKLIGHT_PIN             -1
+
+      //#define FORCE_SOFT_SPI                      // Use this if default of hardware SPI causes display problems
+                                                  //   results in LCD soft SPI mode 3, SD soft SPI mode 0
+
+      #define LCD_RESET_PIN                 PE10  // Must be high or open for LCD to operate normally.
+
+      #if EITHER(FYSETC_MINI_12864_1_2, FYSETC_MINI_12864_2_0)
+        #ifndef RGB_LED_R_PIN
+          #define RGB_LED_R_PIN             PE11
+        #endif
+        #ifndef RGB_LED_G_PIN
+          #define RGB_LED_G_PIN             PE12
+        #endif
+        #ifndef RGB_LED_B_PIN
+          #define RGB_LED_B_PIN             PE13
+        #endif
+      #elif ENABLED(FYSETC_MINI_12864_2_1)
+        #define NEOPIXEL_PIN                PE11
+      #endif
+
+    #else                                         // !FYSETC_MINI_12864
+
+      #define LCD_PINS_ENABLE               PE9
+      #define LCD_PINS_D4                   PE10
 
-    #if IS_ULTIPANEL
-      #define LCD_PINS_D5                   PE11
-      #define LCD_PINS_D6                   PE12
-      #define LCD_PINS_D7                   PE13
+      #if IS_ULTIPANEL
+        #define LCD_PINS_D5                 PE11
+        #define LCD_PINS_D6                 PE12
+        #define LCD_PINS_D7                 PE13
+      #endif
 
       #if ENABLED(REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER)
         #define BTN_ENC_EN           LCD_PINS_D7  // Detect the presence of the encoder
       #endif
 
-    #endif
-
+    #endif // !FYSETC_MINI_12864
   #endif
 
   // Alter timing for graphical display
------------------------------------

ellensp, 2021-02-16:
'we' are working on unified lcd's Ie defined in one place with a standardised pin naming across all boards. This will eventually remove the need to add all the lcd's to all the boards pins.h files...
but this is a long way off.

kylix-rd, 2021-02-16:
Until then, it would be nice to have some more eyes on the issue embedded in this FR. Namely the mentioned display initialization issue. It's almost working, but could use some added expertise from someone more familiar with the intricacies of LCD initialization.

kylix-rd, 2021-02-16:
It appears that the u8g library is initialized prior to the sd card subsystem and uses a different code path to initialize the SPI. The u8g code path ends up initializing the default HW SPI2 port rather than the HW SPI1 port. When an SD card is inserted, the SD card subsystem then sets up the correct SPI port, which is then subsequently used by the u8g library. Until then, the display output is being set to the wrong SPI port. Forcing the use of HW SPI1 should fix the issue.

kylix-rd, 2021-02-16:
The latest patch in the PR now enables the display to now function properly as expected. While my above analysis is correct, the use of FORCE_SOFT_SPI works. I suspect I had an out-of-date u8g library which had no-op'ed the software SPI for the stm32F4.

thisiskeithb, 2021-02-16:
Closing since your PR was merged.

closed 2021-02-16 by thisiskeithb