MarlinFirmware/Marlin #21104: [BUG] (RFC) Wait message did not trigger

X-Ryl669, 2021-02-16:
### Bug Description

I'm trying to fix #21010. In this bug report, the old release code used to have this behavior: 
```
2021-02-15 15:00:28,479 - Send: M21
2021-02-15 15:00:32,503 - Recv: echo:No SD card
2021-02-15 15:00:32,511 - Recv: ok P63 B31

2021-02-15 15:00:32,515 - Send: M155 S2
2021-02-15 15:00:32,575 - Recv:  T:29.37 /0.00 (3870.0000) B:35.16 /0.00 (3824.0000) @:0 B@:0
2021-02-15 15:00:32,578 - Recv: Not SD printing
2021-02-15 15:00:32,594 - Recv: ok P63 B31
```
If you look closely to the timestamp, you'll see that the command M21 takes 4 seconds to answer. Over commands are answered almost instantly.

Yet, if I understand the code correctly, Marlin's idle() function calls `manage_inactivity` which, in turns, calls `Queue::get_available_commands()` which then calls this: 
```
    static millis_t last_command_time = 0;
    const millis_t ms = millis();
    if (length == 0 && !serial_data_available() && ELAPSED(ms, last_command_time + NO_TIMEOUTS)) {
      SERIAL_ECHOLNPGM(STR_WAIT);
      last_command_time = ms;
    }
  [... parsing here ...]
  last_command_time = ms;
  _enqueue_command(cmd with send ok) // that does length ++
```
Then `idle()` calls `queue.advance()` that consume a command that was received, parse it and, if required, send an ok, decrease length. 

So to sum up the behavior observed, knowning `BUF_SIZE = 32` and `NO_TIMEOUTS = 1000`,

1. Marlin received "M21", it had length=0 when it happened, `last_command_time` was close to ms (which is set to 15:00:28.479)
2. It parsed the command, then saved `last_command_time` to 15:00:28.479
3. It enqueued the command in the buffer, so `length == 1`
4. It returned from the function
5. Then, following `idle()` code, queue is advanced, meaning that M21 is executed, and answer `ok P63 B31` is sent (meaning `length == 1` as expected) and length decremented to 0
6. Then `queue.advance()` is returning and `idle()` is called later again
7. Back in `queue.get_serial_command()` we find the above snippet running again. `length` is 0 and `last_command_time - ms` is greater than `NO_TIMEOUTS`. There is no `serial_data_available()`, yet, no `STR_WAIT` message is emitted.

The current bugfix branch for the same behavior emits a `wait` message like this:
```
2021-02-15 12:53:56,803 - Send: M21
2021-02-15 12:54:00,820 - Recv: echo:No SD card
2021-02-15 12:54:00,826 - Recv: ok P63 B31
2021-02-15 12:54:00,828 - Recv: wait
```
which is what I'd have expected from my understanding of the code above.

Yet, the OP in #21010 says the old code works but the new one does not. 

What is your interpretation here ? 

How come the previous code did not emit wait message ?

Would it be better to update the `last_command_time` when the command is actually executed ?

rhapsodyv, 2021-02-16:
What do you mean by new and old code? Do you have a good and a bad commit?

X-Ryl669, 2021-02-16:
It depends on what you call good. The old is c929fb5 and last bugfix is the new one. There was no change to that part of the code, AFAIK.

X-Ryl669, 2021-02-17:
Ok, I conclude that the old code misbehaved. I've found many subtle bugs in the queue code, so I'll close this one and open another issue.

rhapsodyv, 2021-02-17:
> Ok, I conclude that the old code misbehaved. I've found many subtle bugs in the queue code, so I'll close this one and open another issue.

Did #21010 happens before the serial refactoring (pr: #20783)?

X-Ryl669, 2021-02-17:
No, or not that I'm aware of. The serial refactoring is the reason for another issue (I'm currently correcting) but it's not related this. This "wait" difference probably happened somewhere else. I can't bisect because I don't have the printer to test upon. I've opened #21119 for all the bugs I've found so far in the queue code. If you want to have a look and confirm, please do.

rhapsodyv, 2021-02-17:
I point is: SKR turbo is a pretty common board, maybe the most common LPC board out there, so it's unlike that the cause of that issue is something old. I'm not saying that it cannot be. It really can. But is more likely that some recent change broken some old behaviour. Maybe it is not even related with the serial changes, but another recent one.
It would be good to try some old versions to have a better idea (even if not precise) where/when that bug was inserted.

Snargaloo, 2021-02-17:
It seems the bug only gets out of control when an internal serial port of SKR 1.4 turbo (such as tft) is used for communication instead of USB port.

X-Ryl669, 2021-02-17:
Do you want me to reopen this issue ?

X-Ryl669, 2021-02-17:
@rhapsodyv : I don't think it's an issue. As far as I know, Octoprint completely ignore the "wait" message so this "bug" could have been present for month/years and nobody cared about it. I don't know of other host software acting differently, so if you do, please tell me so I can test their behavior w.r.t. wait messages. 

In all cases, now I'm 100% sure the wait message are generated when the printer is waiting for more serial data, so I think it's good, that's why I closed the issue.

X-Ryl669, 2021-02-18:
It seems that the wait messages are not ignored by Octoprint finally, see [here](https://github.com/MarlinFirmware/Marlin/issues/21010#issuecomment-781095435). So this issue is still relevant. If any of you have an idea why wait message were not generated, I'm all for it.

X-Ryl669, 2021-02-25:
PR #21178 fixed the issue.

closed 2021-02-25 by X-Ryl669