MarlinFirmware/Marlin #21105: Babysteps not working with Z-Offset

Steve021171, 2021-02-16:
Looks like the Z-Offset gets messed up if I use babysteps to adjust the nozzle height. Next print the nozzle is way to high, and homing engraves the bed....

M851 works fine, but after using babysteps this value is not working anymore.
That means if I adjust the Z-Offset for the probe just with M851 command, it works. 
As soon I used babysteps it does not work anymore.

Somebody else has this problem ???????

Software is Marlin 2.0.7.2 
Printer is Artillery Sidewinder X1 and Genius
Mainboard is MKS SGEN L (32Bit) with TMC2209 in UART
Probe is Z-endstop prox switch from the printer.

ellensp, 2021-02-16:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

Steve021171, 2021-02-18:
Tried with the Bugfix version on my X1, and somebody else from my Facebook group on his Genius. Still same thing.
Looks like it is not saving the offset correctly if I change it with babysteps, because if I leave the menu and go back in it shows the old value, not the changed one. But I can see the Z-axis moving if I push the buttons.

Next time when I start another print the nozzle is way to high and nothing sticks to the bed. So I have to start over with adjusting the nozzle height.

DistortedDesigns, 2021-02-18:
I am having issues with baby stepping. You set your negative value and the next time you home it adds the negative value to your Z height. Basically you home with bl touch. Your Z0 is 2mm off the bed. You babystep down -2mm. Save settings m500. Print something, it homes, and it is 4mm off the bed now. I still have z height adjustment as well but it won't go below 0 now. I am running bugfix from 2 weeks ago. I have an ender 3 v1, BTT 1.4 turbo and BTT tft 35 v2 screen.

strayr, 2021-02-18:
I think this regression is in 2.0.7, seem to have 2.0.6.1-bugfix working well and 2.0.7.2 release the Z 0ffsets and Z-offset wizard are confusitngly broken to the point where i'm doubting my own sanity

digant73, 2021-02-21:
@DistortedDesigns it seems such kind of problems happen editing a mesh point. If BL is enabled it seems that to the z height you want to save an offset is added so you will have a wrong z height applied

Doormat1, 2021-02-23:
Im having a similar issue with an ender 3 using an SKR mini E3 v2.0. I cannot change the Z-offset when the printer is running despite having babystepping Z enabled.

stephenwheet, 2021-02-28:
Ok I thought I was going crazy. I have this problem. I just upgrade to the micro Swiss direct drive kit. You need to change the offset because of the new bracket. That I did fine but then z-offset is crazy. I messed with it for like an hour and couldn’t figure out what was going on. It’s like it adding it rather than offset. Super weird behavior than what I’m used it. Normally it autohome then adjust down the z-offset and rehome again.. rinse repeat. I was trying to use the on screen menu on my ender 3 v2.  Then switched to octoprint and was doing it there also. It was super confusing.

closed 2021-06-11 by github-actions[bot]