MarlinFirmware/Marlin #21106: [BUG] SD_ABORT_NO_COOLDOWN disabled, but heaters remain on after Stop Print

Arogen, 2021-02-16:
Using Marlin 2.0.7, if I stop a print, the heaters both remain on even though the configuration_adv.h file has:
  //#define SD_ABORT_NO_COOLDOWN

### Steps to Reproduce

1. Start printing something
2. Stop the print before it finishes
3. Observe the bed and nozzle do not cool down

**Expected behavior:**
When SD_ABORT_NO_COOLDOWN is commented out and you stop a print, the heaters turn off.

**Actual behavior:**
Heaters stay on until manually turned off

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/5990685/configs.zip)

ellensp, 2021-02-16:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

Arogen, 2021-02-17:
The issue exists in the bugfix branch as well.

closed 2021-06-20 by github-actions[bot]