MarlinFirmware/Marlin #21107: [BUG] DETECT_BROKEN_ENDSTOP causes halt in IDEX mode

DrumClock, 2021-02-16:
Hi,
I confirm when the X-min endstop is closed (X1 carriage - parkinkg)
after sending the g-code G28, the printer stops with the error "parking failed".

If enstop X-max or Y-min are closed, the G28 command runs without problems.

This error is only when` ENABLED (DUAL_X_CARRIAGE)` is enabled

_Originally posted by @DrumClock in 
https://github.com/MarlinFirmware/Marlin/issues/20667#issuecomment-756165907_

ellensp, 2021-02-16:
no even vaguely following template. Closing.

thinkyhead, 2021-02-16:
The issue seems pretty obvious. The test is using `X_ENDSTOP` to check the triggered endstop. It is defined as:
```cpp
#define X_ENDSTOP (X_HOME_DIR < 0 ? X_MIN : X_MAX)
```
But on an IDEX machine we check the X homing direction using `x_home_dir(active_extruder)`. So, for this broken endstop testing logic maybe this would work…
```cpp
#define X_ENDSTOP (x_home_dir(active_extruder) < 0 ? X_MIN : X_MAX)
```
…but that's the general idea. I'll do some tests and post a PR to test shortly.

thinkyhead, 2021-02-16:
Ok, once it is passing testing, please give the above-linked PR a try. So far it just applies the fix as described here.

DrumClock, 2021-02-17:
Hi, yes it's okay now. Thank you very much @thinkyhead .
That's the way it should be, quickly and correctly.

Some users accuse me of coming up with something that Marlin is not ready for. 

Isn't that called development and progress?

Marlin does not have 1000 functions that do not work but 100 that work perfectly.

I have offered FW testing several times but no one is interested in it .... so why beg ....

DrumClock, 2021-03-03:
Hi,
..... with the new release of Bugfix 2.0.x does not work again in IDEX mode .....

DrumClock, 2021-03-05:
Hi @thinkyhead 
someone uploaded an old file there .... \ src \ module \ endstops.h

ellensp, 2021-03-05:
last change to https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/endstops.h was 17 days ago.

DrumClock, 2021-03-06:
Hi @ellensp 
Hi, today it's there so why did I download an old file from upstrem bugfix 2.0.x yesterday to my Githup desktop?

closed 2021-03-06 by DrumClock