MarlinFirmware/Marlin #21108: [BUG] LCD Menu Bed Leveling Z-Offset Adjustment is Too Fine

borland1, 2021-02-16:
### Bug Description
When adjusting the `Z-Offset` using LCD menu with rotary encoder knob, the adjustment is too fine.  Adjustment is in displayed millimeters, but increment is in thousandth of a millimeter ( 0.001 mm).

There is an incremental setting in Configuration.h file for  MESH_EDIT_Z_STEP, but no increment feature for the Z-Offset.   A Z-Offset increment of 0.05 mm would be more reasonable, however some may consider that still too fine.

I saw a similar change request reported previously on Closed issue #20630.

After mesh bed leveling has been setup, the Z-Offset is used to adjust first layer squish and accommodate for variations in bed adhesion.  The LCD menu currently displays Z-Offset as “Bed Z”.   Z-Offset was apparently inadvertently changed to “Bed Z” nomenclature.  Term “Z-Offset” is widely used. 

Recommended: 
1.	Add configurable variable to set a Z-Offset menu increment in `Configuration.h` file, such as:  
              `#define Z_OFFSET_INCREMENT 0.1  // units in mm`.
2.	Replace “`Bed Z`” nomenclature in language files to “`Z-Offset`”.

[image]
[image]

### Configuration Files
[ConfigFiles.zip](https://github.com/MarlinFirmware/Marlin/files/5991925/ConfigFiles.zip)

xeddog, 2021-02-17:
It sounds like you are using babystepping, and there is a  babystep multiplicator (or something like that) value that you can tweak.  I have three printers, and on two of them the babystep increment is .025, and the third is .012.  I also seem to remember that there is a value in the LCD definitions, but it escapes me at the moment.

borland1, 2021-02-17:
Thanks for the help.  No, I have not enabled `BABYSTEPPING` in Configuration_adv.h.  My configuration is for Manual Mesh Med Leveling.

This `BUG` report is an issue with standard stepping, being a nuisance because the adjustment is too fine and since there is no available Z-Offset increment multiplier in the bed mesh leveling configuration.  The user is required to rotate encoder knob many times to arrive at a reasonable Z-Offset value.

It appears that `BABYSTEPPING` is a workaround to this `BUG` since it does offer configuring a multiplier.  I will try configuring `BABYSTEPPING` for Manual Mesh Bed Leveling and report any results.

ManuelMcLure, 2021-02-17:
I just checked on my printer and my Z offset adjusts in 0.01mm increments. I'm using the following configuration on a Re-ARM/RAMPS combination with a RepRapDiscount Full Graphics display:
https://github.com/ManuelMcLure/Marlin/commit/edc1a7819d444fc45b59d588aaca27a15af02f57

It looks like what determines whether you get 2 or 3 digits past the decimal is the setting for `Z_PROBE_OFFSET_RANGE_MIN` and `Z_PROBE_OFFSET_RANGE_MAX` - if both of these have absolute values 9 or less then you get 0.001mm adjustment, if either or both are more than 9 in absolute value you get 0.01mm adjustment.

borland1, 2021-02-17:
That's interesting, but My configuration uses the default values:
```
#define Z_PROBE_OFFSET_RANGE_MIN -20
#define Z_PROBE_OFFSET_RANGE_MAX 20
```
which are greater than "9 or less". Your configuration file only specifies:

`#define Z_PROBE_OFFSET_RANGE_MAX 20`

ManuelMcLure, 2021-02-17:
Ah... the code I was referring to only works if you have a bed probe. You're doing manual leveling, and there it's hard coded to be N.NNN format:
```
 277   //
 278   // Mesh Bed Leveling Z-Offset
 279   //
 280   #if ENABLED(MESH_BED_LEVELING)
 281     EDIT_ITEM(float43, MSG_BED_Z, &mbl.z_offset, -1, 1);
 282   #endif
 ```

borland1, 2021-02-17:
Thanks,   then, it looks to me like this might be a fix, in file menu_bed_leveling.cpp..   But not clear to others editing the config file just why it might work?

```
 277   //
 278   // Mesh Bed Leveling Z-Offset
 279   //
 280   #if ENABLED(MESH_BED_LEVELING)
 281     EDIT_ITEM(float43, MSG_BED_Z, &mbl.z_offset, Z_PROBE_OFFSET_RANGE_MIN, Z_PROBE_OFFSET_RANGE_MAX);
 282   #endif
```

ManuelMcLure, 2021-02-17:
No - the important bit is the `float43` - that's "4 digit float, 3 of which are to the right of the decimal". Changing that to `float42_52` will change it to use 0.01mm precision, I think.

borland1, 2021-02-18:
OK, maybe something like this then ...

in file `configuration.h`
```
1038 #define Z_PROBE_OFFSET_RANGE_MIN -20
1039 #define Z_PROBE_OFFSET_RANGE_MAX 20
         #if Z_PROBE_OFFSET_RANGE_MIN >= -9 && Z_PROBE_OFFSET_RANGE_MAX <= 9
            #define LCD_Z_OFFSET_TYPE float43    // Values from -9.000 to +9.000
         #else
            #define LCD_Z_OFFSET_TYPE float42_52 // Values from -99.99 to 99.99
         #endif
```
in file `menu_bed_leveling.cpp`
```
 277   //
 278   // Mesh Bed Leveling Z-Offset
 279   //
 280   #if ENABLED(MESH_BED_LEVELING)
 281     EDIT_ITEM(LCD_Z_OFFSET_TYPE, MSG_BED_Z, &mbl.z_offset, Z_PROBE_OFFSET_RANGE_MIN,  
            Z_PROBE_OFFSET_RANGE_MAX);
 282   #endif
```

borland1, 2021-02-18:
OK, that seems to work fine.  

How about the adding of a multiplier in file `Configuration.h`?  Such as here..

```
1346  #if ENABLED(LCD_BED_LEVELING)
1347    #define MESH_EDIT_Z_STEP  0.025 // (mm) Step size while manually probing Z axis.
1348    #define LCD_PROBE_Z_RANGE 4     // (mm) Z Range centered on Z_MIN_POS for LCD Z adjustment
1349    //#define MESH_EDIT_MENU         // Add a menu to edit mesh points
        //#define Z_OFFSET_MULTIPLIER  1   // Increase value to increase Z-Offset (Bed Z) step size  <==
1350 #endif
```
then, where to reconfigure the incremental value???
```
        #if ENABLED(LCD_BED_LEVELING && Z_OFFSET_MULTIPLIER)
           // ???  = ?? * Z_OFFSET_MULTIPLIER;
        #endif
```

strayr, 2021-02-18:
in config_adv.h as of CONFIGURATION_ADV_H_VERSION 020007 maybe earlier

//#define BABYSTEP_MILLIMETER_UNITS       // Specify BABYSTEP_MULTIPLICATOR_(XY|Z) in mm instead of micro-steps
  #define BABYSTEP_MULTIPLICATOR_Z  1
  
  gives me 1 step of the Z stepper per nudge, becasue multiplicator is in STEPS not MM and i have 0.025mm steps which is 1/4 to 1/8 of a layer height. 
  
  If that comment goes away it babysteps in 1mm increments. IFF someone misunderstood and set the multiplicator to 0.025 or suchlike for mm stepping but its doing 1/40 seeps per click or something stupid then that could be the problem? UNit assumptions gine sideways?

borland1, 2021-06-29:
kPATm,  I have not tried above workaround with the latest Marlin (2.09.1), but since `LCD_Z_OFFSET_TYPE` in file `menu_bed_leveling.cpp` is conditional, you also need to specify `Z_PROBE_OFFSET_RANGE_MIN` and  `Z_PROBE_OFFSET_RANGE_MAX` in `Configuration.h` file, as stated above, such that values are in range -99.99 to 99.99.

kPATm, 2021-06-29:
Thanks for the fast reply. So do I actually have to enter values in the range of -99.99 to 99.99 in the menu_bed_leveling.cpp section?

Configuration.h already has values of -20 to 20

so menu_bed_leveling.cpp would look like this

 277   //
 278   // Mesh Bed Leveling Z-Offset
 279   //
 280   #if ENABLED(MESH_BED_LEVELING)
 281     #if ( Z_PROBE_OFFSET_RANGE_MIN >= -9 && Z_PROBE_OFFSET_RANGE_MAX <= 9 )
              #define LCD_Z_OFFSET_TYPE float43    // Values from -20 to + 20
            #else
              #define LCD_Z_OFFSET_TYPE float42_52 // Values from -20 to 20
         #endif
         EDIT_ITEM(LCD_Z_OFFSET_TYPE, MSG_BED_Z, &mbl.z_offset, Z_PROBE_OFFSET_RANGE_MIN,  
             Z_PROBE_OFFSET_RANGE_MAX);
 282   #endif

Thanks again

kPATm, 2021-06-29:
Well I just tried that with no change again

This is what I have in menu_bed_leveling.cpp

  // Mesh Bed Leveling Z-Offset
  //
  #if ENABLED(MESH_BED_LEVELING)
  #if ( Z_PROBE_OFFSET_RANGE_MIN >= -9 && Z_PROBE_OFFSET_RANGE_MAX <= 9 )
  #define LCD_Z_OFFSET_TYPE float43    // Values from -20 to + 20
  #else
  #define LCD_Z_OFFSET_TYPE float42_52 // Values from -20 to 20
  #endif
  EDIT_ITEM(LCD_Z_OFFSET_TYPE, MSG_BED_Z, &mbl.z_offset, Z_PROBE_OFFSET_RANGE_MIN,  
  Z_PROBE_OFFSET_RANGE_MAX);
  #endif

and this is in configuration.h

// For M851 give a range for adjusting the Z probe offset
#define Z_PROBE_OFFSET_RANGE_MIN -20
#define Z_PROBE_OFFSET_RANGE_MAX 20

borland1, 2021-06-29:
Sorry you have problems, but this is not tech support.

borland1, 2021-07-01:
I looked at the source code for latest release versions, i.e., 2.0.9.1 and bugfix 2.0.x.  For `Mesh Bed Leveling`, the 0.001mm resolution is still hard coded with `float43` in file `menu_bed_leveling.cpp`, line 281.   
```
 278   // Mesh Bed Leveling Z-Offset
 279   //
 280   #if ENABLED(MESH_BED_LEVELING)
 281     EDIT_ITEM(float43, MSG_BED_Z, &mbl.z_offset, Z_PROBE_OFFSET_RANGE_MIN, Z_PROBE_OFFSET_RANGE_MAX);
 282   #endif
```
Alternatively, for 0.01mm resolution, line 281 could be hard to `float42_52`, regardless of set values of `Z_PROBE_OFFSET_RANGE_MIN` and `Z_PROBE_OFFSET_RANGE_MAX`.

thisiskeithb, 2021-07-01:
@borland1: Since you have a working example, it's best if you submitted a [Pull Request](https://marlinfw.org/docs/development/getting_started_pull_requests.html) so it can be reviewed.

Sonicboom247, 2022-04-01:
Was this somehow backed out recently? I am now having the same issue on my hornet with the latest bugfix but not on my sidewinder which has a touch screen. Been driving me crazy. Now it seems baby stepping does nothing but bed Z lowers the nozzle...

closed 2022-04-01 by thisiskeithb

thisiskeithb, 2022-04-01:
@Sonicboom247: Nope. Still there:

https://github.com/MarlinFirmware/Marlin/blob/72a9a02c17e59fa91e8284ca70f8d675498e9f0f/Marlin/src/lcd/menu/menu_bed_leveling.cpp#L275-L285